realsenseai/realsense-ros #503: no image data published with Realsense d435 on intel nuc machine

taochenshh, 2018-10-26:
Hi, I was trying to use the realsense d435 on an intel nuc machine. I can successfully install the driver. However, when I launched the node like `roslaunch realsense2_camera rs_camera.launch`, I didn't see and image data being published. What could be the problem?

The launching script output:
```bash
[ INFO] [1540576309.709325496]: Initializing nodelet with 4 worker threads.
[ INFO] [1540576309.732897999]: RealSense ROS v2.1.0
[ INFO] [1540576309.732932734]: Running with LibRealSense v2.16.1
[ INFO] [1540576309.874786764]: getParameters...
[ INFO] [1540576309.905868617]: setupDevice...
[ INFO] [1540576309.905900972]: JSON file is not provided
[ INFO] [1540576309.905926901]: ROS Node Namespace: camera
[ INFO] [1540576309.905942011]: Device Name: Intel RealSense D435
[ INFO] [1540576309.905955332]: Device Serial No: 815412070383
[ INFO] [1540576309.905968140]: Device FW version: 05.08.15.00
[ INFO] [1540576309.905980642]: Device Product ID: 0x0B07
[ INFO] [1540576309.905992775]: Enable PointCloud: Off
[ INFO] [1540576309.906004815]: Align Depth: Off
[ INFO] [1540576309.906016564]: Sync Mode: Off
[ INFO] [1540576309.906048929]: Device Sensors: 
[ INFO] [1540576309.906072982]: Stereo Module was found.
[ INFO] [1540576309.906088157]: RGB Camera was found.
[ INFO] [1540576309.906118621]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1540576309.906131298]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1540576309.906143881]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1540576309.906160488]: setupPublishers...
[ INFO] [1540576310.318581886]: Expected frequency for depth = 30.00000
[ INFO] [1540576312.374378247]: Expected frequency for infra1 = 30.00000
[ INFO] [1540576312.434720269]: Expected frequency for infra2 = 30.00000
[ INFO] [1540576312.456050096]: Expected frequency for color = 30.00000
[ INFO] [1540576312.478567152]: setupStreams...
 26/10 13:51:52,478 WARNING [140651352803200] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ INFO] [1540576312.695452191]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1540576312.696366161]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1540576312.697234000]: infra2 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1540576312.705179481]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1540576312.721593754]: num_filters: 0
[ INFO] [1540576312.721773130]: publishStaticTransforms...
[ INFO] [1540576312.722440776]: RealSense Node Is Up!
 26/10 13:51:52,774 WARNING [140651352803200] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
[ WARN] [1540576312.898398112]: Reconfigure callback failed with exception start_streaming(...) failed. UVC device is already streaming!: 
 26/10 13:51:57,716 WARNING [140650624304896] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
 26/10 13:51:57,720 WARNING [140650598512384] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
 26/10 13:51:57,724 WARNING [140650515265280] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
 26/10 13:52:02,718 WARNING [140650624304896] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
 26/10 13:52:02,726 WARNING [140650598512384] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
```

bringBackm, 2018-10-27:
l meet the same problem under Ubuntu 16.04 and ROS Kinetic within my PC.
It may seems to be related with the issus #502

bringBackm, 2018-10-27:
l meet the same issue under Ubuntu 16.04 and ROS Kinetic within my PC.
It seems to be related with the issue #502 
After l solve the catkin_make problem using the methods suggested in issue #502, l cannot receive any image messages in ROS

doronhi, 2018-10-28:
Usually I find that it disconnecting and reconnecting the device fix it. Something annoying that shouldn’t happen and should be addressed to. Does it help or the problem persists?

taochenshh, 2018-10-29:
@bringBackm  not yet

closed 2018-10-29 by taochenshh

bringBackm, 2018-10-30:
Thank you , l solve it too.
@taochenshh 
@doronhi