realsenseai/realsense-ros #506: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory]

Kaesarlch, 2018-11-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.10.6 |
| librealsense & realsense | 2.16.1 & 2.1.0 |
| Operating System & Version |  Ubuntu 18.04.1 LTS  | 
| Kernel Version (Linux Only)    |  4.15.0.38-generic                                        | 
| Platform                                 | Intel NUC |
| Language                            |  Python 2.7 / C++                         | 
| ROS                                     | Melodic   |
| Segment			|  Robot                          | 

### Issue Description

Hello everyone,
I am working on a 2 wheeled robot and we want to use the D435 for security / mapping purpose.
Right now I am trying to get the libraries to work with the ROS wrapper.

I have been checking many post in here and other forums but could not find a solution to my problem:
After `catkin_make_isolted --install` I can start a few .launch files like `view_d435_model.launch` but all the `rs_*.launch` will return the same error like for example with `roslaunch realsense2_camera rs_camera.launch`:

... logging to /home/superrobot/.ros/log/901850f6-dd6b-11e8-b627-d4258b378539/roslaunch-superrobot-NUC7i5DNKE-6249.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://superrobot-NUC7i5DNKE:35853/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 1000
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [6261]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 901850f6-dd6b-11e8-b627-d4258b378539
process[rosout-1]: started with pid [6272]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [6278]
process[camera/realsense2_camera-3]: started with pid [6280]
[ INFO] [1541031469.092084646]: Initializing nodelet with 4 worker threads.
[ERROR] [1541031469.111536836]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1541031469.111568268]: The error before refreshing the cache was: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[FATAL] [1541031469.111672617]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera-3] process has died [pid 6280, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/superrobot/.ros/log/901850f6-dd6b-11e8-b627-d4258b378539/camera-realsense2_camera-3.log].
log file: /home/superrobot/.ros/log/901850f6-dd6b-11e8-b627-d4258b378539/camera-realsense2_camera-3*.log

What I have checked:
-updated D435 firmware to 5.10.6
-under `lsusb` I can see my D435, with a speed 5000M which should confirm USB3.0
-with `realsense-viewer` and `pyrealsense2` I can correctly stream from the camera
-I trying upgrading my linux kernel from 4.15 to 4.16 and 4.19 with no good effect.

Any help would be appreciated!

Kaesarlch, 2018-11-01:
nevermind! it is actually working.

I had no idea I needed to started rviz by myself!

So in the end what I did is build LibRealSense from source and build realsense2_camera in a catkin workspace with a simple `catkin_make`

closed 2018-11-04 by doronhi

Horizonzhou, 2020-04-29:
I have met this problem too, could you please share your experience about this problem?

qinhongtju, 2021-02-20:
I want to know if your problem has been solved. I also face the same problem, can I ask you something？

sazzath, 2021-06-12:
I have the same problem too ... Help me please ....

sazzath, 2021-06-12:
> I want to know if your problem has been solved. I also face the same problem, can I ask you something？

[image]

HappySamuel, 2021-11-12:
Hi @Kaesarlch , @sazzath 

I also faced same issue, is this issue cannot be solved even with installing the latest ros-noetic-realsense2-camera ? Only catkin_make realsense2_camera from source can get it working?

kinzess, 2021-11-25:
no, same issue

> Hi @Kaesarlch , @sazzath
> 
> I also faced same issue, is this issue cannot be solved even with installing the latest ros-noetic-realsense2-camera ? Only catkin_make realsense2_camera from source can get it working?
> 
> Best, Samuel