realsenseai/realsense-ros #507: Pointcloud unexpected interpolation along view direction?

taochenshh, 2018-11-02:
I am using ubuntu 16.04, realsense d435 and launched the` rs_rgbd.launch` file.  THe pointcloud seems to be interpolating in the viewing direction. And because of this, there seems to be elongated tails of points where there should be empty space.

Example:
RGB Image:
[image]

The point cloud looks like these:
**Front view**
[image]
**Side view**
[image]
The red circle shows the issue. How to correct it? Thanks.

closed 2018-11-12 by taochenshh

MrYelameli, 2019-08-05:
How you did the calibration? what is the solution? Would you kindly let me know? I am also facing the similar problem.

mahdipo, 2020-09-21:
step 1: due to sensor size for D435  840X480 is the standard resolution and generates less noise in this mode.
step 2: change preset in realsense viewer to HighAccuracy or HighDensity instead of default or custom, its give you much better result. also you can make this changes in your code too.