realsenseai/realsense-ros #510: realsense d435 + turtlebot2

ELDEMIRY, 2018-11-05:
Hi, 
I am an absolute beginner in ROS
I have a turtlebot2 mounted by velodyne laser scanner 
this turtlebot can built a map and can also perform autonomous navigation. (gmapping used) 
Now my objective is replacing the laser scanner with realsense  d435 and do the same job.
I can start the camera node and make it publish the RGBD point cloud using aligned depth topic
by this command 
roslaunch realsense2_camera rs_rgbd.launch
so my question is how to replace the velodyne laser scanner by the d435 to be used in gmapping to do the same job of mapping  environment.

ELDEMIRY, 2018-11-21:
Thank you so much. I tried it before but i failed. Attached the content of the lunch file which i already use for velodyne laser scanner.
I know that i should replace  pointcloud_to_laser scan  by depthimage_to_laserscan and to include the realsense driver. Also to replace tf2 of velodyne to realsense. 
So, if possible can you give me an example of how to change the following launch file contents.?
Many thanks in advance.

[DEFAULT.txt](https://github.com/intel-ros/realsense/files/2603639/DEFAULT.txt)

SteveMacenski, 2018-11-21:
@ELDEMIRY your description is exactly correct. You just need to swap your node to depth images (and the appropriate parameters for your sensor), make sure your scan topic is the same, launch the realsense driver (which I assume you can find an example here for), and update TF.

I dont have an example for you unfortunately, I use a lidar on my robots.

ELDEMIRY, 2018-11-21:
Many thanks.

SteveMacenski, 2018-11-21:
No problem. Do you mind closing the ticket? This isnt really in the scope of ros wrapped realsense driver anymore

closed 2018-11-21 by ELDEMIRY

jameszw001, 2021-02-05:
@SteveMacenski  did you create the map successfully at the end?