realsenseai/realsense-ros #511: No longer possible to publish registered cloud with full depth field of view

JosephBowkett, 2018-11-08:
Good evening,

I've recently shifted to using realsense2_camera 2.1.1 from 2.0.4, but noticed that it appears to be no longer possible to publish a color registered point cloud with the full depth field of view, as BaseRealSenseNode::publishPointCloud now seems to drop all points outside of the RGB FOV.

The previous behavior of registering all points within the RGB FOV and leaving all points outside of that blue suited my needs perfectly, as it gave me depth data on as much of the workspace as possible (the depth FOV on the D435 was a key selling point for me), while providing color for context in the center.

I've hacked my own modification to publish points outside the RGB FOV as blue, but it would be nice to be able to select this behavior with the pointcloud_texture_stream parameter. If it is in fact possible to achieve this behavior with the existing arguments then I'd be happy to be proven wrong.

As an aside, 2.0.4 -> 2.1.1 also saw the pointcloud shift from structured (width/height from depth) to unstructured (width = num_points, height = 1), which makes sense when discarding points outside the RGB FOV, but complicates my post-processing which assumes a structured cloud. As a further feature request, it would be great if the pointcloud could remain structured if an option to use the full depth FOV was selected.

Cheers

doronhi, 2018-11-09:
Thank you for your comment.
I think it’s very reasonable and will seriously consider modifying it back, as soon as time permits.

neomanic, 2018-11-22:
I'll second this request to keep the full FOV of the depth camera in the colorised point cloud. I only came across this issue because I was looking for something else, as we are still using the older version. I will hold off on upgrading for now.

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

closed 2019-04-15 by RealSenseCustomerSupport

eyalu, 2019-06-16:
I would also like to second that this change is needed - I'm trying to get a 3d space point from the rgb pixel. Used to be very easy with the structured pointcloud, now it's a mess with having to read the intrinsic data from the camerainfo topic and assigning the data one element at a time... very frustrating for a relatively common use case.

doronhi, 2019-06-17:
The original issue, as far as I understand it, was solved by setting "allow_no_texture_points:=true" (description in README.md file).
@eyalu , not sure I understand what is the functionality that you need. 
How would you do it before and how do you do it now?

eyalu, 2019-07-11:
Sorry for replying after such a long time, I didn't monitor the thread by accident...
I used a different depth camera that provided me with a structured cloud. Used to take the u, v values along with the steps and find physical locations of points based on the rgb pixels.
I thought this functionality (rgb pixel to point in physical space) is a basic feature of the ros node, but only when I came to implement did I find out that it's not a simple (or even possible) task using the existing node, without rewriting everything to use the sdk itself.
Am I wrong? Is there another approach to it?
Many thanks @doronhi , will monitor closely now :-)

doronhi, 2019-07-11:
Not sure I understand. Are you trying to achieve the same goal as @npd26 at #807 ?