realsenseai/realsense-ros #513: catkin_make for realsense failing because of compilation issues

brookdpg, 2018-11-10:
I assumed this was similar to https://github.com/intel-ros/realsense/issues/484,
 but, I do have the latest of the dev branch

:~/catkin_ws$ `sudo apt-get install librealsense2-dev`
Reading package lists... Done
Building dependency tree       
Reading state information... Done
librealsense2-dev is already the newest version (2.16.4-0~realsense0.127).

 and not sure why I am getting compilation errors for running the
` catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release `
step:

....
[ 97%] Building CXX object realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/realsense_node_factory.cpp.o
[ 97%] Building CXX object realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o
[ 98%] Building CXX object realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/rs435_node.cpp.o
In file included from /home/ops/catkin_ws/src/realsense/realsense2_camera/src/../include/rs435_node.h:6:0,
                 from /home/ops/catkin_ws/src/realsense/realsense2_camera/src/rs435_node.cpp:1:
/home/ops/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h: In member function ‘void realsense2_camera::PipelineSyncer::operator()(rs2::frame) const’:
/home/ops/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h:72:23: error: ‘invoke’ was not declared in this scope
    invoke(std::move(f));
                       ^
/home/ops/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h:72:23: note: suggested alternative:
In file included from /usr/include/boost/thread/detail/thread.hpp:23:0,
                 from /usr/include/boost/thread/thread_only.hpp:22,
                 from /usr/include/boost/thread/thread.hpp:12,
                 from /usr/include/boost/thread.hpp:13,
                 from /opt/ros/kinetic/include/diagnostic_updater/diagnostic_updater.h:48,
                 from /home/ops/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h:12,
                 from /home/ops/catkin_ws/src/realsense/realsense2_camera/src/../include/rs435_node.h:6,
                 from /home/ops/catkin_ws/src/realsense/realsense2_camera/src/rs435_node.cpp:1:
/usr/include/boost/thread/detail/invoke.hpp:131:17: note:   ‘boost::detail::invoke’
     inline auto invoke(BOOST_THREAD_RV_REF(Fp) f, BOOST_THREAD_RV_REF(Args) ...args)
                 ^
In file included from /home/ops/catkin_ws/src/realsense/realsense2_camera/src/../include/sr300_node.h:6:0,
                 from /home/ops/catkin_ws/src/realsense/realsense2_camera/src/realsense_node_factory.cpp:5:
/home/ops/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h: In member function ‘void realsense2_camera::PipelineSyncer::operator()(rs2::frame) const’:
/home/ops/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h:72:23: error: ‘invoke’ was not declared in this scope
    invoke(std::move(f));
                       ^
/home/ops/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h:72:23: note: suggested alternative:
In file included from /usr/include/boost/thread/detail/thread.hpp:23:0,
                 from /usr/include/boost/thread/thread_only.hpp:22,
                 from /usr/include/boost/thread/thread.hpp:12,
                 from /usr/include/boost/thread.hpp:13,
                 from /opt/ros/kinetic/include/diagnostic_updater/diagnostic_updater.h:48,
                 from /home/ops/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h:12,
                 from /home/ops/catkin_ws/src/realsense/realsense2_camera/src/../include/sr300_node.h:6,
                 from /home/ops/catkin_ws/src/realsense/realsense2_camera/src/realsense_node_factory.cpp:5:
/usr/include/boost/thread/detail/invoke.hpp:131:17: note:   ‘boost::detail::invoke’
     inline auto invoke(BOOST_THREAD_RV_REF(Fp) f, BOOST_THREAD_RV_REF(Args) ...args)
                 ^
In file included from /home/ops/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:1:0:
/home/ops/catkin_ws/src/realsense/realsense2_camera/src/../include/base_realsense_node.h: In member function ‘void realsense2_camera::PipelineSyncer::operator()(rs2::frame) const’:
/home/ops/catkin_ws/src/realsense/realsense2_camera/src/../include/base_realsense_node.h:72:23: error: ‘invoke’ was not declared in this scope
    invoke(std::move(f));
                       ^
/home/ops/catkin_ws/src/realsense/realsense2_camera/src/../include/base_realsense_node.h:72:23: note: suggested alternative:
In file included from /usr/include/boost/thread/detail/thread.hpp:23:0,
                 from /usr/include/boost/thread/thread_only.hpp:22,
                 from /usr/include/boost/thread/thread.hpp:12,
                 from /usr/include/boost/thread.hpp:13,
                 from /opt/ros/kinetic/include/diagnostic_updater/diagnostic_updater.h:48,
                 from /home/ops/catkin_ws/src/realsense/realsense2_camera/src/../include/base_realsense_node.h:12,
                 from /home/ops/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:1:
/usr/include/boost/thread/detail/invoke.hpp:131:17: note:   ‘boost::detail::invoke’
     inline auto invoke(BOOST_THREAD_RV_REF(Fp) f, BOOST_THREAD_RV_REF(Args) ...args)
                 ^
/home/ops/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp: In member function ‘void realsense2_camera::BaseRealSenseNode::publishAlignedDepthToOthers(rs2::frameset, const ros::Time&)’:
[... 177 lines cut ...]
                       ^
/home/ops/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h:72:23: note: suggested alternative:
In file included from /usr/include/boost/thread/detail/thread.hpp:23:0,
                 from /usr/include/boost/thread/thread_only.hpp:22,
                 from /usr/include/boost/thread/thread.hpp:12,
                 from /usr/include/boost/thread.hpp:13,
                 from /opt/ros/kinetic/include/diagnostic_updater/diagnostic_updater.h:48,
                 from /home/ops/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h:12,
                 from /home/ops/catkin_ws/src/realsense/realsense2_camera/src/../include/rs415_node.h:6,
                 from /home/ops/catkin_ws/src/realsense/realsense2_camera/src/rs415_node.cpp:1:
/usr/include/boost/thread/detail/invoke.hpp:131:17: note:   ‘boost::detail::invoke’
     inline auto invoke(BOOST_THREAD_RV_REF(Fp) f, BOOST_THREAD_RV_REF(Args) ...args)
                 ^
realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:86: recipe for target 'realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o' failed
make[2]: *** [realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:110: recipe for target 'realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/rs415_node.cpp.o' failed
make[2]: *** [realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/rs415_node.cpp.o] Error 1
realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:158: recipe for target 'realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/sr300_node.cpp.o' failed
make[2]: *** [realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/sr300_node.cpp.o] Error 1
CMakeFiles/Makefile2:2148: recipe for target 'realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/all' failed
make[1]: *** [realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
ops@ops-XPS-15-9560:~/catkin_ws$

doronhi, 2018-11-18:
Could it be that an older version of librealsense is still lurks around?
Try checking with something like:
`
sudo find / -name librealsense2.so*
`
Do you see only version 2.16.4?

Nachmr, 2018-11-26:
Hi! someone with the same issue here. @dorodnic this is the output of `sudo find / -name librealsense2.so*`:
find: ‘/run/user/1000/gvfs’: Permission denied
/usr/lib/x86_64-linux-gnu/librealsense2.so.2.11.0
/usr/lib/x86_64-linux-gnu/librealsense2.so.2
/usr/local/lib/librealsense2.so.2.10.1
/usr/local/lib/librealsense2.so
/usr/local/lib/librealsense2.so.2
/usr/local/lib/librealsense2.so.2.10.2

piepieninja, 2018-11-28:
I also receive a failure at this stage in the `catkin_make` :

```
[ 77%] Building CXX object realsense2_camera/CMakeFiles/realsense2_camera.dir/src/rs415_node.cpp.o
[ 81%] Building CXX object realsense2_camera/CMakeFiles/realsense2_camera.dir/src/realsense_node_factory.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:110: recipe for target 'realsense2_camera/CMakeFiles/realsense2_camera.dir/src/rs415_node.cpp.o' failed
make[2]: *** [realsense2_camera/CMakeFiles/realsense2_camera.dir/src/rs415_node.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:300: recipe for target 'realsense2_camera/CMakeFiles/realsense2_camera.dir/all' failed
make[1]: *** [realsense2_camera/CMakeFiles/realsense2_camera.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j2 -l2" failed
```

I tired the simple installation instructions and got this. I then tried the source install and got some errors relating to nodelets, but that seems unrelated.

piepieninja, 2018-12-01:
I reinstalled everything from scratch on my vm and now I have this error relating to `rs2::processing_block`

```
realsense2_camera/src/base_realsense_node.cpp:643:56: error: ‘class rs2::processing_block’ has no member named ‘process’
                         frameset = filter_it->_filter->process(frameset);
                                                        ^
c++: internal compiler error: Killed (program cc1plus)
```

Guptajakala, 2018-12-01:
@piepieninja Hi, I'm meeting the same problem with you because of "rs2::processing_block". Any solutions?

closed 2019-01-06 by doronhi