realsenseai/realsense-ros #514: No pointcloud2 avaliable

CycleMark, 2018-11-13:
Hi,

I've just got a D435 sensor. I've installed all the associated s/w and upgraded the firmware.
When I try to visualise the pointcloud2 topic there's no such topic available. Even when trying to "add by topic".

Does anyone have any suggestions?

Thanks
Mark

doronhi, 2018-11-18:
Hi Mark,
Can you post the command line you use to run and the printout to the screen?

CycleMark, 2018-11-19:
Hi Doronhi,

Not quite sure what happened but the pointcloud it now showing very. But very very slow. About 1 fps.

I'm starting it with 

roslaunch realsense2_camera rs_rgbd.launch

Output is 

`mark@mxnetLinux-6:~$ roslaunch realsense2_camera rs_rgbd.launch
... logging to /home/mark/.ros/log/ef732cfe-ec3b-11e8-adbd-247703ebae54/roslaunch-mxnetLinux-6-13541.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.40:40816/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 1000
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
[... 56 lines cut ...]
[ INFO] [1542660281.473775588]: publishStaticTransforms...
[ INFO] [1542660281.473909255]: RealSense Node Is Up!
[ INFO] [1542660281.473950524]: Start updating dynamic parameters...
 19/11 20:44:41,524 WARNING [139628834191232] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 19/11 20:44:41,525 WARNING [139628834191232] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 19/11 20:44:41,525 WARNING [139628834191232] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 19/11 20:44:41,525 WARNING [139628834191232] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
[ INFO] [1542660281.583859057]: Done updating dynamic parameters...
[ WARN] [1542660315.138045969]: Messages of type 2 arrived out of order (will print only once)
[ WARN] [1542660315.138314891]: Messages of type 1 arrived out of order (will print only once)
^C[camera/points_xyzrgb_hw_registered-5] killing on exit
[camera/color_rectify_color-4] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[camera/realsense2_camera-3] killing on exit
[ INFO] [1542660951.830544008]: Interrupt Signal is received! Terminating RealSense Node...
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
`

Any idea why so slow? This is on an intel i7 box.

Thanks
Mark

doronhi, 2018-11-20:
While using rgbd, pointcloud is being built by another node listening on color and depth frames. What is their rate?

Is running:
‘
roslaunch realsense2_camra rs_camera filters:=pointcloud
‘
Behaves differently?

CycleMark, 2018-11-23:
Hi Doronhi,

I'm still trying to get consistency with my experimentation (hence the delay in replying). 

The frame rates seem to have resolved - they are now reporting around 30FPS. However the system does appear to be a little unstable.

By that I mean I am finding I have to regularly disconnect and reconnect the camera as I get the following errors reported:

 21/11 13:22:38,883 ERROR [140461992802176] (types.h:180) xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error
[ERROR] [1542806558.884178452]: An exception has been thrown: xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error
[ERROR] [1542806558.884322624]: An exception has been thrown: xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error
 21/11 13:22:38,887 WARNING [140461219841792] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
 21/11 13:22:38,888 WARNING [140461211449088] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds

As yet I don't know what is going on

Rgs

Mark

doronhi, 2018-11-23:
There is a strong suspicion of hardware synchronization issue. Try setting color_fps:=15 (half of depth_fps)

CycleMark, 2018-11-23:
Ok tried that. Unfortunately no difference. Another error I get reported (again on occasion - so not consistent) is:

 23/11 16:49:08,448 WARNING [139929085859712] (types.cpp:57) set_pu(id=4) failed! Last Error: Input/output error
[ERROR] [1542991748.448571564]: Failed. Skip initialization of parameter 1
 23/11 16:49:13,471 WARNING [139929085859712] (types.cpp:57) set_xu(id=8) failed! Last Error: Input/output error
[ERROR] [1542991753.471525335]: Failed. Skip initialization of parameter 5
 23/11 16:49:13,471 WARNING [139929085859712] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 23/11 16:49:13,471 WARNING [139929085859712] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 23/11 16:49:13,472 WARNING [139929085859712] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 23/11 16:49:13,472 WARNING [139929085859712] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 23/11 16:49:18,491 WARNING [139929085859712] (types.cpp:57) get_xu(id=11) failed! Last Error: Input/output error
[ERROR] [1542991758.491543682]: Failed. Skip initialization of parameter 9
 23/11 16:49:23,510 WARNING [139929085859712] (types.cpp:57) get_xu(id=2) failed! Last Error: Input/output error
[ERROR] [1542991763.510785097]: Failed. Skip initialization of parameter 10
 23/11 16:49:28,533 WARNING [139929085859712] (types.cpp:57) set_xu(id=2) failed! Last Error: Input/output error
[ERROR] [1542991768.533490907]: Failed. Skip initialization of parameter 11
 23/11 16:49:33,564 WARNING [139929085859712] (types.cpp:57) set_xu(id=11) failed! Last Error: Input/output error
[ERROR] [1542991773.564460716]: Failed. Skip initialization of parameter 100
[ INFO] [1542991773.564564315]: Done updating dynamic parameters...

Are there any useful logs I can send?

I don't really want to be rebooting the robot to get the sensor to come up

Thanks

Mark

CycleMark, 2018-11-25:
Still getting a lot of:

WARNING [140461211449088] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds

From both the ROS interface and now the realsense viewer. 

Have a feeling there is a hardware fault. I'm going to replace and will get back to this thread when it comes through

Mark

Haya-Alsharif, 2018-11-27:
@CycleMark Can you make sure it is not a hardware reset issue? you can do that by disconnecting the camera usb and connecting it again and launching the file.

CycleMark, 2018-12-01:
Its a problem with the hardware - Currently logged with Intel to resolve

CycleMark, 2018-12-01:
- Sorry clicked too soon...

Am I able to get a point cloud with out the colour camera?

Thanks

CycleMark, 2019-01-10:
HI @doronhi ,

Thanks for the reply. I've just upgraded to the latest version 2.1.3 and the supported version of LibRealSense v2.17.1.

It seems now I'm getting the following error - googling hasn't, as yet, reviled anything conclusive. Given the hardware setup hasn't changed. Would you have any suggestions?

Error message is:
[ERROR] [1547106257.718752321]: Hardware Notification:USB CAM overflow,1.54711e+12,Error,Hardware Error

Full start up log attached

Thanks and Regards

Mark
[D435 Startup Error.txt](https://github.com/intel-ros/realsense/files/2744180/D435.Startup.Error.txt)

doronhi, 2019-01-10:
The meaning of the message is that too much data was stuck on the way from camera to computer.
How is the camera behavior?

CycleMark, 2019-01-10:
Currently the camera is connected t a USB 2 port as previously. I'm using one of the intel Up boards.

doronhi, 2019-01-10:
I’m sorry, I meant to ask do you receive frames? what is the frame rate? Does the error message indicate an actual problem?

CycleMark, 2019-01-10:
Ah - yes it's causing a problem. All I get from camera is the the standard video feed in rviz

CycleMark, 2019-03-04:
Hi @doronhi 

Just revisiting this issue. Still not solved - I upgraded to v2.18.1. The up board should I think be able to handle the data (I've connected it to the USB3 port) - not to mention the boards were designed with these in mind.

I do get data streamed though despite the warning messages - but after a while it kills the system

Do you have any other thoughts one what I can test/try?

Thanks and Regards

Mark

[image]

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

@CycleMark What's FW version of the camera you're using? And could you please try the latest librealsense, FW and ROS version to see if you still have the issue? Thanks!

CycleMark, 2019-04-28:
Hi,

I've upgraded again to the latest FW, librealsense and still the same problem. Kinetic shouldn't be this issue. There is something odd about getting the data off the USB 3 port on the upboard:

New output

/ros_realsense$ roslaunch realsense2_camera rs_camera.launch --screen
... logging to /home/mark/.ros/log/34535626-6998-11e9-955b-dbadc11087ae/roslaunch-kobukiMxnet2-24152.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.79:33413/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters:
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
[... 66 lines cut ...]
[ INFO] [1556489588.189912069]: insert Infrared to Stereo Module
[ INFO] [1556489588.308767747]: num_filters: 0
[ INFO] [1556489588.308947950]: SELECTED BASE:Depth, 0
[ INFO] [1556489588.309713287]: RealSense Node Is Up!
[ INFO] [1556489588.309819731]: Setting Dynamic reconfig parameters.
 28/04 23:13:08,371 WARNING [139628335429376] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ WARN] [1556489588.374849497]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 28/04 23:13:08,463 WARNING [139628193117952] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ INFO] [1556489588.496130664]: Done Setting Dynamic reconfig parameters.
[ WARN] [1556489590.665822313]: Hardware Notification:USB SCP overflow,1.55649e+12,Error,Hardware Error
[ WARN] [1556489592.668674565]: Hardware Notification:USB SCP overflow,1.55649e+12,Error,Hardware Error
^C[camera/realsense2_camera-2] killing on exit
[ INFO] [1556489593.111015273]: Unloading nodelet /camera/realsense2_camera from manager realsense2_camera_manager
[camera/realsense2_camera_manager-1] killing on exit
[ INFO] [1556489593.116765356]: Interrupt Signal is received! Terminating RealSense Node...
[camera/realsense2_camera_manager-1] escalating to SIGTERM
shutting down processing monitor...
... shutting down processing monitor complete
done

Any idea where the problem could lie?

Thanks

Mark

RealSenseCustomerSupport, 2019-04-29:
----------------------------------------------

@CycleMark “USB SCP overflow” often happens when USB bandwidth limitation.Could you please verify if the USB connection type is USB3 by the tool rs-enumerate-device or RealSenseViewer? Please make sure you're using USB3 cable besides the connection to USB3 port.  Thanks!

rs-enumerate-device: https://github.com/IntelRealSense/librealsense/tree/master/tools/enumerate-devices
RealSenseViewer: https://github.com/IntelRealSense/librealsense/tree/master/tools/realsense-viewer

RealSenseCustomerSupport, 2019-04-29:
----------------------------------------------

@CycleMark Just thought of another easier way to check USB connection. Please use "lsusb -t" to check the USB connection type is USB3 or not.  Thanks!

CycleMark, 2019-04-29:
Hi,

Thanks for that. It seems as though it might be the USB 2 hub picking it up - despite the fact it's plugged into the UBS 3 port on the UP board. Any idea how to change that?

[image]

Kind Regards

Mark

RealSenseCustomerSupport, 2019-04-29:
----------------------------------------------

@CycleMark If you want to use USB3, then please change to USB3 cable and USB3 port to try again. If you want to keep using USB3, then I suggest you decrease resolution or FPS for the depth/IR/color. Or you can just enable one or two but not all of these four streamings. You can modify the "enable_depth"/"enable_infra1"/"enable_infra2"/"enable_color"  to "false" in rs_camera.launch

  <arg name="depth_width"        default="640"/>
  <arg name="depth_height"        default="480"/>
  **<arg name="enable_depth"        default="true"/>**​

  <arg name="infra_width"        default="640"/>
  <arg name="infra_height"      default="480"/>
  **<arg name="enable_infra1"      default="true"/>**

**  <arg name="enable_infra2"      default="true"/>**​

  <arg name="color_width"        default="640"/>
  <arg name="color_height"        default="480"/>
**  <arg name="enable_color"        default="true"/>**

CycleMark, 2019-04-30:
Hi,

I'm not quite sure what this the matter with the USB3 port. I've double checked the wire (and also tried another one) and can confirm that it is certainly a USB3 cable (It's thicker that standard and has blue indicators.

That said to try and get this working I have disabled infra 1 & 2 which does seem to resolve the initial issue.

However when connecting via rviz the machine with the realsense sensor just hangs.

Investigating further the dmesg dumps out the attached 
[realsensedmesgdump.txt](https://github.com/intel-ros/realsense/files/3130069/realsensedmesgdump.txt)

(Which doesn't look good)

Process output looks like:
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [6417]
process[camera/realsense2_camera-3]: started with pid [6418]
[ INFO] [1556603590.497768779]: Loading nodelet /camera/realsense2_camera of type realsense2_camera/RealSenseNodeFactory to manager realsense2_camera_manager with the following remappings:
[ INFO] [1556603590.510916694]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1556603590.566018195]: Initializing nodelet with 4 worker threads.
[ INFO] [1556603590.583068634]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] is now available.
[ INFO] [1556603590.825937118]: RealSense ROS v2.2.3
[ INFO] [1556603590.826025290]: Running with LibRealSense v2.21.0
[ INFO] [1556603591.342329264]: getParameters...
[ INFO] [1556603591.671541559]: setupDevice...
[ INFO] [1556603591.671627906]: JSON file is not provided
[ INFO] [1556603591.671680030]: ROS Node Namespace: camera
[ INFO] [1556603591.671746440]: Device Name: Intel RealSense D435
[ INFO] [1556603591.671804563]: Device Serial No: 827112071647
[ INFO] [1556603591.672040105]: Device FW version: 05.11.06.200
[ INFO] [1556603591.672102403]: Device Product ID: 0x0B07
[ INFO] [1556603591.672158164]: Enable PointCloud: Off
[ INFO] [1556603591.672370220]: Align Depth: Off
[ INFO] [1556603591.672477541]: Sync Mode: Off
[ INFO] [1556603591.672693822]: Device Sensors:
[ INFO] [1556603591.672882478]: Stereo Module was found.
[ INFO] [1556603591.672956114]: RGB Camera was found.
[ INFO] [1556603591.673132283]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1556603591.673190593]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1556603591.673241529]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1556603591.673304752]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1556603591.673370913]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1556603591.678271492]: setupPublishers...
[ INFO] [1556603591.704056124]: Expected frequency for depth = 30.00000
[ INFO] [1556603591.892630000]: Expected frequency for color = 30.00000
[ INFO] [1556603592.026284518]: setupStreams...
 30/04 06:53:12,027 WARNING [140122305628032] (sensor.cpp:365) Unregistered Media formats : [ UYVY ]; Supported: [ GREY UYVY Z16  ]
[ INFO] [1556603592.037058108]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1556603592.038690843]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1556603592.040814325]: insert Depth to Stereo Module
[ INFO] [1556603592.040951508]: insert Color to RGB Camera
[ INFO] [1556603592.159517735]: num_filters: 0
[ INFO] [1556603592.159773477]: SELECTED BASE:Depth, 0
[ INFO] [1556603592.161044748]: RealSense Node Is Up!
[ INFO] [1556603592.161231480]: Setting Dynamic reconfig parameters.
 30/04 06:53:12,218 WARNING [140121600780032] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ WARN] [1556603592.221466062]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 30/04 06:53:12,315 WARNING [140121583994624] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ INFO] [1556603592.369102027]: Done Setting Dynamic reconfig parameters.
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[ INFO] [1556603594.751864035]: Unloading nodelet /camera/realsense2_camera from manager realsense2_camera_manager
[ INFO] [1556603594.753730275]: Interrupt Signal is received! Terminating RealSense Node...
[camera/realsense2_camera_manager-2] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

Looks like there is something missing

Kind Regards

Mark

RealSenseCustomerSupport, 2019-04-30:
----------------------------------------------

@CycleMark Sorry for one typo in previous comment, I meant to say " If you want to keep using **USB2**, then I suggest you decrease resolution or FPS for the depth/IR/color. "

For the hang issue, we'll further check. Thanks!

CycleMark, 2019-04-30:
Hi - Yep I knew what you meant :) But USB 3 if failing to detect for reasons I can't yet figure (It's connected to the USB3 OGA port on the Up Board). I will send a photo later on

RealSenseCustomerSupport, 2019-05-05:
----------------------------------------------

@CycleMark Could you please connect the camera to another PC's USB3 port to see if it can be detected as USB3 mode? You can try more cables or more USB3 ports to rule out if the cable or USB port you tested is not USB3.

RealSenseCustomerSupport, 2019-05-09:
----------------------------------------------

@CycleMark Any new findings? Looking forward to your update. Thanks!

CycleMark, 2019-05-09:
HI - Sorry I meant to reply. I've tried other cables (all ensured they were USB) and had same issue. I'll check again on another PC tonight (Though from memory I think that works)

Kind Regards

Mark

CycleMark, 2019-05-11:
Hi,

Further update - the camera works okay on other PCs and I've tried multiple cables with the upboard to no avail. I've tried connecting it just to the USB 2 connection and USB3. It's only recognised as a USB 2 device - as expected in the latter case (The PC's see it as USB 3)

What i have seen is that connecting to see the image with rviz seems to crash the board. I think there is a conflict with the real sense driver and the USB WiFi dongle as with rviz connected the connection just drops. 

If I use rqt_image_viewer then is does work but the frame rate is very low

I've attached some photos as an example of the various connections and the wire for clarity
[Photos.zip](https://github.com/IntelRealSense/realsense-ros/files/3169253/Photos.zip)

Rgs

Mark

RealSenseCustomerSupport, 2019-05-13:
----------------------------------------------

@CycleMark There was some discussions about enumerate as USB2 in https://github.com/IntelRealSense/librealsense/issues/1391  Please have a try with other adapters to connect to USB3 OTG port. Thanks!

CycleMark, 2019-05-13:
Hi,

I've tried a bunch of new cables, adaptors etc and non of them work. I've run through the list of increasingly dubious suggestions from that link. Again nothing.

I can only conclude that the upboard's USB 3 interface is incorrectly designed. Does that seem reasonable?

Kind Regards

Mark

RealSenseCustomerSupport, 2019-05-15:
----------------------------------------------

@CycleMark Sorry for that. Could you please contact the technical support of Upboard to see if they have better suggestions about this?

CycleMark, 2019-05-20:
Hi -ok will do. Who/whats the best contact?

Thanks

Mark

RealSenseCustomerSupport, 2019-05-21:
----------------------------------------------

@CycleMark I think maybe you can check the official site of upboard or the vendor where you get the board. Thanks!

RealSenseCustomerSupport, 2019-06-03:
----------------------------------------------

@CycleMark Any other questions about this ticket? Looking forward to your update. Thanks!

CycleMark, 2019-06-05:
Hi - Sorry I've not had a chance to contact Intel. Shall do so soon. Kinda regards

Mark

RealSenseCustomerSupport, 2019-06-17:
----------------------------------------------

@CycleMark That's fine. Any update please let us know. Thanks!

RealSenseCustomerSupport, 2019-06-24:
----------------------------------------------

@CycleMark The remaining USB type issue is not librealsense related. Can we close the ticket at this point?

closed 2019-06-27 by RealSenseCustomerSupport