realsenseai/realsense-ros #515: start_streaming(...) failed. UVC device is already streaming!

forqed, 2018-11-15:
I have fresh Ubuntu 16.04. When I follow instructions, I encountered a problem. After type "roslaunch realsense2_camera rs_camera.launch" I got:
`16/11 00:04:35,328 WARNING [140028622067584] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 16/11 00:04:35,329 WARNING [140028622067584] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 16/11 00:04:35,329 WARNING [140028622067584] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!`
From roslaunch realsense2_camera, only "demo_pointcloud.launch" works properly. Here create visualization works fine.

alphaccw, 2018-11-17:
exat the same issue for me. Have you solve it already?

alphaccw, 2018-11-17:
exact the same issue for me. Have you solve it already?

forqed, 2018-11-17:
> exact the same issue for me. Have you solve it already?

Not yet. I still waiting for answer.

doronhi, 2018-11-18:
Besides the warning, are the topics being published?

forqed, 2018-11-18:
> Besides the warning, are the topics being published?
Yes, I checked it.

doronhi, 2018-11-18:
What version do you use? Is that the latest release?
As far as I know, this is a warning only. Annoying, I thought I already fixed it, but has no effect of anything else.

Haya-Alsharif, 2018-11-27:
@doronhi I have been recieving the same warnings since the last update to the package. This function is throwing this warning:
```
void BaseRealSenseNode::toggleSensors(bool enabled)
{
    for (auto it=_sensors.begin(); it != _sensors.end(); it++)
    {
        auto& sens = _sensors[it->first];
        try
        {
            if (enabled)
                sens.start(_syncer);
            else
                sens.stop();
        }
        catch(const rs2::wrong_api_call_sequence_error& ex)
        {
            ROS_DEBUG_STREAM("toggleSensors: " << ex.what());
        }
    }
}
```
It seems to me that it was added to solve the hardware reset issue, however the iterator somehow exceeds the count of the sensor.. Again I am no expert but I thought this might help. I commented the part that uses this function and instead, I used the HW reset approach explained [here](https://github.com/intel-ros/realsense/issues/443#issuecomment-418602124).

JianhengShi, 2018-11-28:
exact the same issue. Topics are started though

Raisintoe, 2018-12-11:
I have that warning, and no data is being published over the topics. I also get the warning: WARNING [140490720667392] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds

I know that I was receiving data at some point, but I cannot get data streams from my RealSense D435 anymore. I don't know why.

Raisintoe, 2018-12-11:
Thank you Haya-Alsharif, I will try that added line of code. But I have already tried reconnecting the USB. I am using the USB 3.0 port. If I boot my computer with the camera plugged in, the camera is simply never found. When I unplug it, and plug it back in, that is when I get the warning that frames did not arrive within 5 sec.

Raisintoe, 2018-12-11:
Yeah, I added your line of code to reset the device, but I still get the same message: "(backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds". I am running the command "roslaunch realsense2_camera rs_camera.launch" and when I use "rostopic list", I get:
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_infra1
/camera/extrinsics/depth_to_infra2
/camera/infra1/camera_info
/camera/infra1/image_rect_raw
/camera/infra1/image_rect_raw/compressed
/camera/infra1/image_rect_raw/compressed/parameter_descriptions
/camera/infra1/image_rect_raw/compressed/parameter_updates
/camera/infra2/camera_info
/camera/infra2/image_rect_raw
/camera/infra2/image_rect_raw/compressed
/camera/infra2/image_rect_raw/compressed/parameter_descriptions
/camera/infra2/image_rect_raw/compressed/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/realsense2_camera_manager/parameter_descriptions
/camera/realsense2_camera_manager/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf_static

but none of these topics are actually publishing anything that I have been able to see.

Raisintoe, 2018-12-11:
I hope I am not posting too many things here. But I thought it would be helpful for everyone if I post the roslaunch output that I have been getting:

~$ roslaunch realsense2_camera demo_pointcloud.launch 
... logging to /home/derek/.ros/log/95384538-fd81-11e8-9de4-4025c2a5b140/roslaunch-derek-G74Sx-6965.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://derek-G74Sx:34382/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_imu: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_pointcloud: True
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 1000
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)
    rviz (rviz/rviz)
[... 24 lines cut ...]
[ INFO] [1544559364.815887105]: Sync Mode: On
[ INFO] [1544559364.815960929]: Device Sensors: 
[ INFO] [1544559364.816000454]: Stereo Module was found.
[ INFO] [1544559364.816033078]: RGB Camera was found.
[ INFO] [1544559364.816077564]: setupPublishers...
[ INFO] [1544559364.819098471]: Expected frequency for depth = 30.00000
[ INFO] [1544559364.831900938]: Expected frequency for color = 30.00000
[ INFO] [1544559364.841937142]: setupStreams...
 11/12 13:16:04,842 WARNING [139786900645760] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ INFO] [1544559364.845099926]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1544559364.847190988]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1544559364.858781577]: Add Filter: pointcloud
[ INFO] [1544559364.859461322]: num_filters: 1
[ INFO] [1544559364.859502561]: publishStaticTransforms...
[ INFO] [1544559364.859643605]: RealSense Node Is Up!
[ INFO] [1544559364.859688628]: Start updating dynamic parameters...
 11/12 13:16:04,911 WARNING [139786900645760] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 11/12 13:16:04,911 WARNING [139786900645760] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 11/12 13:16:04,911 WARNING [139786900645760] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 11/12 13:16:04,911 WARNING [139786900645760] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
[ INFO] [1544559365.148471364]: Done updating dynamic parameters...
 11/12 13:16:09,861 WARNING [139785951770368] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
 11/12 13:16:09,862 WARNING [139786033157888] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
 11/12 13:16:14,863 WARNING [139785951770368] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
 11/12 13:16:14,867 WARNING [139786033157888] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds

Raisintoe, 2018-12-12:
I sorta solved my problem. I had to upgrade my RealSense D435 frimware from 5.9.2.0 to 5.10.x.x. This has my camera working on USB 2.0 now, but it still does not work on USB 3.0. The camera isn't even found on USB3.0 now. I haven't reset my computer yet since the update though, it's possible that my USB 3.0 port is still locked from attempting the firmware update on that port.

Haya-Alsharif, 2018-12-12:
@Raisintoe I just upgraded my SDK to 2.17.0 and started facing similar issues to the one your mentioned. Now I am upgrading my camera FW to 5.10.6.0 to see if this will solve the issue.
My previous SDK was 16.5 and camera FW was 5.9.2.0

Haya-Alsharif, 2018-12-13:
Update: a simple reboot after the upgrades of both SDK and camera FW solved all the issues for me  :sweat_smile:

closed 2019-01-07 by doronhi