realsenseai/realsense-ros #516: [Feature request D435] Restamp frames using hw measurements from an external board

m0oz, 2018-11-16:
Currently multiple cameras can be synced to expose simultaneously as described in this [whitepaper](https://realsense.intel.com/wp-content/uploads/sites/63/Multiple_Camera_WhitePaper_rev1.1.pdfl)
Timestamping will be done using frame metadata that was generated by the camera which was set as master device.
This is sufficient for most time-critical applications, but it will not work for any setup where many sensors such as IMU, and multiple cameras have to be time synchronized. While all realsense cameras will have synchronized timestamps, the reference clock will still be the crystal on the master realsense's ASIC.

My suggestion would be to add a class that allows to subscribes that provides timestamps that are sent by a board (e.g. a pixhawk or arduino).
This could work two ways. Either the board captures output pulses of the master realsense or it is triggering the realsenses externally and sending the time of the trigger interrupts.

Let me know about you thoughts.
We have a fork that is doing just this and have been testing it for some time to use the realsense for VIO.
If this seems like a valid feature I would go ahead and create a PR

doronhi, 2018-11-18:
Hi,
Thanks for your suggestion. It sounds reasonable and I appreciate it. We need some time to consider and will let you know as soon as I can.
Thanks again.

m0oz, 2018-11-26:
@doronhi Just let me know about your thoughts.
I opened #520 so that you see what we did so far. I don't know if it is mergeable as is but if it sounds like a good feature I am happy to adapt it

RealSenseCustomerSupport, 2019-05-13:
----------------------------------------------

There is a PR on the proposed changes and also was validated by someone outside IDC group.
Now waiting for IDC to make decision on it.

RealSenseSupport, 2021-02-19:
@m0oz doronhi has provided some comments below for this feature request. Sorry that we won't consider this feature at this moment. Can we close it accordingly? Thanks!

https://github.com/IntelRealSense/realsense-ros/pull/520#issuecomment-511661465
https://github.com/IntelRealSense/realsense-ros/pull/520#issuecomment-511689004

RealSenseSupport, 2021-03-08:
@m0oz Any other questions about this? Can we close it? Looking forward to your reply. Thanks!

RealSenseSupport, 2021-03-16:
@m0oz Sorry that we didn't get your reply for weeks. Will close it at this point. Thanks!

closed 2021-03-16 by RealSenseSupport