realsenseai/realsense-ros #517: Cannot run D435 on fresh Ubuntu 16.04 install

5yler, 2018-11-16:
I am trying to get the driver running on a fresh Ubuntu install. The D435 camera I am using and associated USB cable are both functional (I can run the driver fine on a different Ubuntu 16.04 machine). I installed `librealsense` with pre-built binaries as described [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) and am using the latest code on `development` branch for `realsense2_camera`. As I had no luch with 4.10 kernel, I tried upgrading to 4.15 - the output of both cases is pasted below. Please advise.

## `4.10.0-28` Kernel
Output of `roslaunch realsense2_camera rs_rgbd.launch`:

```
... logging to /home/syler/.ros/log/ba0813d6-e9d9-11e8-a700-c0b6f9704397/roslaunch-bronco-3296.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://bronco:37509/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 1000
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
[... 181 lines cut ...]
```
### System Configuration

Output of `lsusb` for the camera:
```
Bus 006 Device 003: ID 8086:0b07 Intel Corp. 
```
Output of `lsusb -t` for the camera:
```
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 3, If 0, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 1: Dev 3, If 1, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 1: Dev 3, If 2, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 1: Dev 3, If 3, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 1: Dev 3, If 4, Class=Video, Driver=uvcvideo, 5000M
```
Output of `uname -a`:
```
Linux bronco 4.15.0-39-generic #42~16.04.1-Ubuntu SMP Wed Oct 24 17:09:54 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```
Output of `modinfo uvcvideo | grep "version:" `:
```
version:        1.1.2.realsense-1.3.3
srcversion:     C4FB5E9B212E9C39ED6CD7C
```

alphaccw, 2018-11-17:
'Frames didn't arrived within 5 seconds', i met this problem for quite many times. just plug out and replug in the camera, it can solved for me.

alphaccw, 2018-11-17:
@5yler , this sounds weird. I solved this kind of errors all by replug, or rerun the command. i am using ubantu 16.04 and kernel 4.15

doronhi, 2018-11-18:
The problem of 'Frames didn't arrived within 5 seconds' could usually be solved by setting the IR and Color sensors to a different frequency. Try running with color_fps:=15. It seems like this is a firmware problem that has something to do with synchronization of the streams.
About the problem 'No RealSense devices were found!', it usually refers to either a kernel or permission problems. Could it be that after upgrading the kernel you didn't install the realsense kernel-patch for that kernel?
`modinfo uvcvideo | grep "version:` should include realsense string"
If that's ok, try installing Intel Realsense permission scripts located in librealsense source directory:
`
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/ 
sudo udevadm control --reload-rules && udevadm trigger 
`

5yler, 2018-11-20:
Installing the permission scripts didn't work - however I solved the problem by running
```bash
sudo usermod -aG dialout $USER
```
and restarting the computer.

closed 2018-11-20 by 5yler