realsenseai/realsense-ros #518: Support for D435i with IMU

SteveMacenski, 2018-11-20:
Is that in the plan to support it here? It would be super useful to get the IMU data out

atyshka, 2018-11-26:
Should be easy since there’s already support for imu in older models

mfehr, 2018-12-13:
We're also interested in getting the D435i to work with this ROS node. I slightly adapted the ROS node yesterday to get it to accept the new sensor and publish all the other topics like images and depth, but even though the IMU logic is there (as @atyshka pointed out) and the motion module is recognized I cannot get it to actually publish any IMU data. If anyone is working on this or has some pointers how to configure the node we'd appreciate it. Thanks!

doronhi, 2018-12-15:
I updated the development branch with a version that support the D435i. Waiting for some feedbacks and stability before releasing it as the next version.
I also added a white paper in the wiki section describing a use case.

closed 2018-12-17 by SteveMacenski