realsenseai/realsense-ros #519: Native Ubuntu 16.04 in Windows 10 using RVIZ to view video

tmichals, 2018-11-25:
Is it possible to view a ROS/RealSense video stream from a nvidia TX2 through local network switch to a Native Ubuntu 16.04 Window 10 PC using RVIZ?     I uploaded a video of the details https://youtu.be/Bd6TBd4Z8vk of what I am attempting.   

RVIZ works great on the TX2 I can see the video fine... but not on Ubuntu 16.04 Native Windows 10 PC.   Hopefully it is in my PC ROS install? 

RVIZ appears to need the D435 connected on the same computer, but I was hoping to connect remotely.  The Windows computer Ubuntu doing a lsusb returns nothing and running the roslaunch realsense2_camera rs_rgbd.launch dies because it can not see the camera(when attached to usb3).  But since I would be using ROS that is not necessary or desired for what I want to do.

tmichals, 2018-12-09:
> Hi @tmichals,
> Sorry for the long delay.
> Did you overcome the problem already?
> Like in all other ROS nodes, There is no inherent problem running the realsense2_camera on one machine and the rviz on another.
> From my experience, the problem could be with setting both ROS_IP and ROS_MASTER_URI. You have to make sure you set both of them correctly in every terminal you use. Otherwise you might get rostopic list to work but rostopic echo will show nothing and it could create exactly your problem.

Hi doronhi,
 Thank you for replying.   I am still working on the problem trying to get rviz realsense camera to work on my PC with rviz.   I am pretty sure about registering the ROS_MASTER_URI because I am able to get rviz and the realsense camera working on the TX2.   I can get the realsense camera working on windows 10 just using the realsense install application.  BUT When I open the ubuntu terminal on my windows 10 PC and rviz no luck.   I will detail this out in a video so you can see my steps, next, and post it on my youtube channel.  I will let you know when I post it...perhaps you will see something that I am doing wrong.
thanks,Tom

doronhi, 2018-12-10:
You mentioned double-checking the ROS_MASTER_URI setting so I assume you refered also to the **ROS_IP** definition in **every terminal**.
Looking forward to your video then.

doronhi, 2018-12-26:
Hope it is solved by now. If not, please free to reopen issue.

closed 2018-12-26 by doronhi

tmichals, 2018-12-26:
Hi doronhi, sorry I have been busy with the holidays...  For a sanity check for myself, I setup an ubuntu14 ROS PC and I was able to connect to the realsense camera on my TX2.  RVIZ worked perfectly on the ubuntu14 PC on the network.  So the RealSense TX2 to ubuntu14 ROS RVIZ does work on another computer as long as it is Linux.  I still ran roscore - ROS_MASTER_URI - on the windows10 PC in an ubuntu16 terminal and that worked great.    3 different computers all working together...  

I think the problem is somewhere with RealSense and (Xming and RVIZ).  Because I can run roscore on the windows10 Ubuntu16 terminal, and I can do the turtleisum demo from a Xming window controlling the turtles on the TX2 from window10 Xming terminal on the windows10 PC.   ROS networking must be working normally.      I am not sure if there is something that needs to be configured or setup in Xming that I am missing or RealSense is not working with Xming somehow or Xming and RVIZ don't work together. 

Do you know of anyone that has tested RealSense in this way?  I only want to carry around 1 computer to interface to the robot.   As it is I have to have a windows PC for control and a Linux PC for video.  

I plan to work on a video to demonstrate.