realsenseai/realsense-ros #526: What's the difference between /camera/color/image_raw and /camera/color/image_rect_color?

mshenyu, 2018-11-30:
Hello,
I've wrote a node that subscribes both /camera/color/image_raw and /camera/color/image_rect_color topics, and transfer them into cv::Mat images.

And then I do a pixel wise difference of these 2 cv::Mat. I found that their difference is a zero matrix, which means they are exactly the same thing in terms of pixel values:

```
    cv::Mat color = color_ptr->image; 
    cv::Mat depth = depth_ptr->image;
    cv::Mat colorRect = colorRect_ptr->image;    // the difference between image_raw and image_rect_color
    cv::Mat absolute_difference;
    cv::absdiff(color, colorRect, absolute_difference);
    double s = cv::sum( absolute_difference )[0];
    cout << s << endl;
```
 This prints '0'.

So I am curious what's the difference of these two topics?
Thanks!

doronhi, 2018-12-13:
Where did you get a topic named /camera/color/image_rect_color ?
It could be, indeed, a copy created by rgb_processing node.

mshenyu, 2018-12-13:
This topic is generated after I run 
```
roslaunch realsense2_camera rs_rgbd.launch
```
Yeah, I am just curious why there are two same topics published.

closed 2018-12-26 by doronhi

Algabri, 2019-12-05:
you can see [http://library.isr.ist.utl.pt/docs/roswiki/openni_launch.html#Published_Topics](here)