realsenseai/realsense-ros #528: multi time run intel-ros/realsense, uvcvideo: Non-zero status (-71) happens

FENGZHANG123, 2018-12-06:
Required Info |  
-- | --
Camera Model | { D415/ D435}
Firmware Version | (5.10.6)
Operating System & Version | { Linux (Ubuntu 16)}
Kernel Version (Linux Only) | (4.15.0-42-generic)
Platform | PC
SDK Version | { 2.16.3 }
Language | {C++ }
Segment | {Robot}
intel-ros/realsense | {2.11.1}

Issue Description

Hi, when I run realsense2_camera first time, everything is fine.But when I shutdown the program with 'Ctrl C' and restart it, error happens:
PARAMETERS

    /downwire_camera/realsense2_camera/accel_fps: 1000
    /downwire_camera/realsense2_camera/accel_optical_frame_id: downwire_camera_a...
    /downwire_camera/realsense2_camera/align_depth: False
    /downwire_camera/realsense2_camera/aligned_depth_to_color_frame_id: downwire_camera_a...
    /downwire_camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: downwire_camera_a...
    /downwire_camera/realsense2_camera/aligned_depth_to_infra1_frame_id: downwire_camera_a...
    /downwire_camera/realsense2_camera/aligned_depth_to_infra2_frame_id: downwire_camera_a...
    /downwire_camera/realsense2_camera/base_frame_id: downwire_camera_link
    /downwire_camera/realsense2_camera/color_fps: 30
    /downwire_camera/realsense2_camera/color_frame_id: downwire_camera_c...
    /downwire_camera/realsense2_camera/color_height: 480
    /downwire_camera/realsense2_camera/color_optical_frame_id: downwire_camera_c...
    /downwire_camera/realsense2_camera/color_width: 848
    /downwire_camera/realsense2_camera/depth_fps: 30
    /downwire_camera/realsense2_camera/depth_frame_id: downwire_camera_d...
    /downwire_camera/realsense2_camera/depth_height: 480
    /downwire_camera/realsense2_camera/depth_optical_frame_id: downwire_camera_d...
    /downwire_camera/realsense2_camera/depth_width: 848
    /downwire_camera/realsense2_camera/enable_color: True
    /downwire_camera/realsense2_camera/enable_depth: True
    /downwire_camera/realsense2_camera/enable_fisheye: True
    /downwire_camera/realsense2_camera/enable_imu: True
    /downwire_camera/realsense2_camera/enable_infra1: True
    /downwire_camera/realsense2_camera/enable_infra2: True
    /downwire_camera/realsense2_camera/enable_pointcloud: False
    /downwire_camera/realsense2_camera/enable_sync: False
    /downwire_camera/realsense2_camera/filters:
    /downwire_camera/realsense2_camera/fisheye_fps: 30
    /downwire_camera/realsense2_camera/fisheye_height: 480
    /downwire_camera/realsense2_camera/fisheye_optical_frame_id: downwire_camera_f...
    /downwire_camera/realsense2_camera/fisheye_width: 848
    /downwire_camera/realsense2_camera/gyro_fps: 1000
    /downwire_camera/realsense2_camera/gyro_optical_frame_id: downwire_camera_g...
    /downwire_camera/realsense2_camera/infra1_fps: 30
    /downwire_camera/realsense2_camera/infra1_frame_id: downwire_camera_i...
    /downwire_camera/realsense2_camera/infra1_height: 480
    /downwire_camera/realsense2_camera/infra1_optical_frame_id: downwire_camera_i...
    /downwire_camera/realsense2_camera/infra1_width: 848
    /downwire_camera/realsense2_camera/infra2_fps: 30
    /downwire_camera/realsense2_camera/infra2_frame_id: downwire_camera_i...
    /downwire_camera/realsense2_camera/infra2_height: 480
    /downwire_camera/realsense2_camera/infra2_optical_frame_id: downwire_camera_i...
    /downwire_camera/realsense2_camera/infra2_width: 848
    /downwire_camera/realsense2_camera/json_file_path:
    /downwire_camera/realsense2_camera/pointcloud_texture_index: 0
    /downwire_camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
    /downwire_camera/realsense2_camera/rosbag_filename:
    /downwire_camera/realsense2_camera/serial_no: 827312071232
    /rosdistro: kinetic
    /rosversion: 1.12.14

NODES
/downwire_camera/
[... 123 lines cut ...]
06/12 17:29:39,879 WARNING [139908659517312] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ INFO] [1544088579.880634647]: depth stream is enabled - width: 848, height: 480, fps: 30
[ INFO] [1544088579.880881410]: infra1 stream is enabled - width: 848, height: 480, fps: 30
[ INFO] [1544088579.881104228]: infra2 stream is enabled - width: 848, height: 480, fps: 30
06/12 17:29:39,881 WARNING [139908659517312] (backend-v4l2.cpp:1248) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
06/12 17:29:39,881 WARNING [139908659517312] (sensor.cpp:338) Unregistered Media formats : [ RW16 ]; Supported: [ ]
[ INFO] [1544088579.883141575]: color stream is enabled - width: 848, height: 480, fps: 30
06/12 17:29:44,891 WARNING [139907938899712] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
06/12 17:29:44,975 ERROR [139908659517312] (types.h:179) xioctl(VIDIOC_STREAMON) failed for buf_type=1 Last Error: Input/output error
[ERROR] [1544088590.097255257]: An exception has been thrown: xioctl(VIDIOC_STREAMON) failed for buf_type=1 Last Error: Input/output error
[ERROR] [1544088590.097558169]: An exception has been thrown: xioctl(VIDIOC_STREAMON) failed for buf_type=1 Last Error: Input/output error
[FATAL] [1544088590.245427995]: Failed to load nodelet '/downwire_camera/points_xyzrgb_sw_registeredof typedepth_image_proc/point_cloud_xyzrgbto managerrealsense2_camera_manager'
[FATAL] [1544088590.245463713]: Failed to load nodelet '/downwire_camera/realsense2_cameraof typerealsense2_camera/RealSenseNodeFactoryto managerrealsense2_camera_manager'
[FATAL] [1544088590.245472823]: Failed to load nodelet '/downwire_camera/color_rectify_colorof typeimage_proc/rectifyto managerrealsense2_camera_manager'
[FATAL] [1544088590.245516913]: Failed to load nodelet '/downwire_camera/register_depthof typedepth_image_proc/registerto managerrealsense2_camera_manager'
[downwire_camera/realsense2_camera_manager-2] process has died [pid 16466, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/zf/.ros/log/73d9914a-f939-11e8-a579-58fb84344185/downwire_camera-realsense2_camera_manager-2.log].
log file: /home/zf/.ros/log/73d9914a-f939-11e8-a579-58fb84344185/downwire_camera-realsense2_camera_manager-2*.log
[downwire_camera/realsense2_camera-3] process has died [pid 16467, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/zf/.ros/log/73d9914a-f939-11e8-a579-58fb84344185/downwire_camera-realsense2_camera-3.log].
log file: /home/zf/.ros/log/73d9914a-f939-11e8-a579-58fb84344185/downwire_camera-realsense2_camera-3*.log
[downwire_camera/color_rectify_color-4] process has died [pid 16468, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load image_proc/rectify realsense2_camera_manager --no-bond image_mono:=color/image_raw image_rect:=color/image_rect_color __name:=color_rectify_color __log:=/home/zf/.ros/log/73d9914a-f939-11e8-a579-58fb84344185/downwire_camera-color_rectify_color-4.log].
log file: /home/zf/.ros/log/73d9914a-f939-11e8-a579-58fb84344185/downwire_camera-color_rectify_color-4*.log
[downwire_camera/register_depth-5] process has died [pid 16476, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load depth_image_proc/register realsense2_camera_manager false rgb/camera_info:=color/camera_info depth/camera_info:=depth/camera_info depth/image_rect:=depth/image_rect_raw depth_registered/image_rect:=depth_registered/sw_registered/image_rect_raw __name:=register_depth __log:=/home/zf/.ros/log/73d9914a-f939-11e8-a579-58fb84344185/downwire_camera-register_depth-5.log].
log file: /home/zf/.ros/log/73d9914a-f939-11e8-a579-58fb84344185/downwire_camera-register_depth-5*.log
[downwire_camera/points_xyzrgb_sw_registered-6] process has died [pid 16485, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb realsense2_camera_manager false rgb/image_rect_color:=color/image_rect_color rgb/camera_info:=color/camera_info depth_registered/image_rect:=depth_registered/sw_registered/image_rect_raw depth_registered/points:=depth_registered/points __name:=points_xyzrgb_sw_registered __log:=/home/zf/.ros/log/73d9914a-f939-11e8-a579-58fb84344185/downwire_camera-points_xyzrgb_sw_registered-6.log].
log file: /home/zf/.ros/log/73d9914a-f939-11e8-a579-58fb84344185/downwire_camera-points_xyzrgb_sw_registered-6*.log

doronhi, 2019-01-08:
It seems to be a firmware issue, indeed related to the Ctrl-C shutdown.
Until it is fixed, I've merged a "reset hardware" option (usage: initial_reset:=true) in the last release (v2.1.3)

FENGZHANG123, 2019-01-09:
Yes, I have solved this problem also by hardware reset. Sorry for forgetting to close this problem.

closed 2019-01-09 by FENGZHANG123