realsenseai/realsense-ros #529: cannot run roslaunch realsense2_camera rs_camera.launch

HLNN, 2018-12-06:
I can`t get the frames, when I run `roslaunch realsense2_camera rs_camera.launch`. 
I tried it with `realsense-viewer`, and it was the same. However the `realsense-viewer` on win10 can works. 

 I got the output below:

```
vtol@vtol:~$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/vtol/.ros/log/f41a4d40-f943-11e8-8473-f8633f4a9bbf/roslaunch-vtol-31294.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/vtol/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://vtol:45511/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 1000
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [31304]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f41a4d40-f943-11e8-8473-f8633f4a9bbf
process[rosout-1]: started with pid [31317]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [31334]
process[camera/realsense2_camera-3]: started with pid [31335]
[ INFO] [1544093089.489847818]: Initializing nodelet with 4 worker threads.
[ INFO] [1544093089.534657002]: RealSense ROS v2.1.1
[ INFO] [1544093089.534686809]: Running with LibRealSense v2.16.3
[ INFO] [1544093089.700663846]: getParameters...
[ INFO] [1544093089.740896678]: setupDevice...
[ INFO] [1544093089.740925861]: JSON file is not provided
[ INFO] [1544093089.740954524]: ROS Node Namespace: camera
[ INFO] [1544093089.740980467]: Device Name: Intel RealSense D435
[ INFO] [1544093089.740994057]: Device Serial No: 827312073922
[ INFO] [1544093089.741010187]: Device FW version: 05.09.02.00
[ INFO] [1544093089.741024493]: Device Product ID: 0x0B07
[ INFO] [1544093089.741040234]: Enable PointCloud: Off
[ INFO] [1544093089.741061027]: Align Depth: Off
[ INFO] [1544093089.741077263]: Sync Mode: Off
[ INFO] [1544093089.741159945]: Device Sensors: 
[ INFO] [1544093089.741193004]: Stereo Module was found.
[ INFO] [1544093089.741214479]: RGB Camera was found.
[ INFO] [1544093089.741253269]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1544093089.741271435]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1544093089.741288574]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1544093089.741307209]: setupPublishers...
[ INFO] [1544093089.743488527]: Expected frequency for depth = 30.00000
[ INFO] [1544093089.769186932]: Expected frequency for infra1 = 30.00000
[ INFO] [1544093089.792588615]: Expected frequency for infra2 = 30.00000
[ INFO] [1544093089.813684158]: Expected frequency for color = 30.00000
[ INFO] [1544093089.835069448]: setupStreams...
 06/12 18:44:49,836 WARNING [140571855181696] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ INFO] [1544093089.838501282]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1544093089.838675021]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1544093089.838829980]: infra2 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1544093089.844470492]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1544093089.912097527]: num_filters: 0
[ INFO] [1544093089.912134821]: publishStaticTransforms...
[ INFO] [1544093089.912364416]: RealSense Node Is Up!
[ INFO] [1544093089.912393778]: Start updating dynamic parameters...
 06/12 18:44:49,965 WARNING [140571855181696] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 06/12 18:44:49,965 WARNING [140571855181696] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 06/12 18:44:49,965 WARNING [140571855181696] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 06/12 18:44:49,965 WARNING [140571855181696] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
[ INFO] [1544093090.143278313]: Done updating dynamic parameters...
 06/12 18:44:54,907 WARNING [140571100301056] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
 06/12 18:44:54,914 WARNING [140571085096704] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
 06/12 18:44:54,919 WARNING [140571074230016] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
 06/12 18:44:59,911 WARNING [140571100301056] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
 06/12 18:44:59,919 WARNING [140571085096704] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
 06/12 18:44:59,921 WARNING [140571074230016] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[ INFO] [1544093104.115755736]: Interrupt Signal is received! Terminating RealSense Node...
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

the output of `lsusb` and `lsusb -t` for the camera

```
vtol@vtol:~$ lsusb
Bus 002 Device 004: ID 8086:0b07 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 8087:0a2b Intel Corp. 
Bus 001 Device 004: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```
```
vtol@vtol:~$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
    |__ Port 2: Dev 4, If 0, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 2: Dev 4, If 1, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 2: Dev 4, If 2, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 2: Dev 4, If 3, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 2: Dev 4, If 4, Class=Video, Driver=uvcvideo, 5000M

```
output of `uname -a`

```
vtol@vtol:~$ uname -a
Linux vtol 4.15.0-42-generic #45~16.04.1-Ubuntu SMP Mon Nov 19 13:02:27 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

Output of `modinfo uvcvideo | grep "version:" `

```
vtol@vtol:~$ modinfo uvcvideo | grep "version:"
version:        1.1.2.realsense-1.3.4
srcversion:     C4FB5E9B212E9C39ED6CD7C

```

doronhi, 2018-12-13:
the issue of "Frames didn't arrived within 5 seconds" is known. Currently it usually means that you need to disconnect and reconnect the camera or set the color_fps to half of the infra1_fps (i.e., 15 and 30 fps).
However, since you mention that there is a problem with the realsense_viewer in linux, I would open the issue in the librealsense page: https://github.com/IntelRealSense/librealsense

closed 2018-12-13 by doronhi

HLNN, 2019-02-22:
Thanks 
I update the framework, it works now

jinnshaw, 2019-02-25:
Update the framework. Refer to https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Linux-RealSense-D400-DFU-Guide.pdf

HappySamuel, 2019-06-13:
Hi all,

Is that means if I update the framework, the issue of "Frames didn't arrived within 5 seconds" will be gone completely?

Best,
Samuel