realsenseai/realsense-ros #531: cpp:643:56: error: ‘class rs2::processing_block’ has no member named ‘process’

jinfagang, 2018-12-11:
using ros kinetic
catkin_make error:

```
catkin_ws/src/realsense-2.1.1/realsense2_camera/src/base_realsense_node.cpp:643:56: error: ‘class rs2::processing_block’ has no member named ‘process’
                         frameset = filter_it->_filter->process(frameset);
                                                        ^
realsense-2.1.1/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:86: recipe for target 'realsense-2.1.1/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o' failed
make[2]: *** [realsense-2.1.1/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:1188: recipe for target 'realsense-2.1.1/realsense2_camera/CMakeFiles/realsense2_camera.dir/all' failed
make[1]: *** [realsense-2.1.1/realsense2_camera/CMakeFiles/realsense2_camera.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

```

gabrielebaris, 2018-12-11:
Hi,
I am using ROS melodic on Ubuntu 18.04 running inside Docker.
I had the same problem cloning the repo on the development branch. I solved downloading the [build 2.1.2](https://github.com/intel-ros/realsense/releases/tag/2.1.2)
I'm still experiencing errors at runtime, but at least it compiles.
I hope this can help

gabrielebaris, 2018-12-11:
@jinfagang I don't know. I found out the problem only today (I started using this camera just a few days ago) and tried the latest release. When I discovered it solved that issue, I replied to your issue hoping it would help you.
I'm sure they will solve the bug also in the development branch as soon as they can.

rcacurs, 2018-12-12:
Hi. Are you using the latest version from development branch? This problem was introduced with realsense library version 2.17, but was solved with one of the pull requests.

jinfagang, 2018-12-13:
@rcacurs Which one? But not in master

rcacurs, 2018-12-13:
branch: development, or tag: 2.1.2 [This commit  fixed issue ](https://github.com/intel-ros/realsense/commit/96402fd4ebdcf274d165b6d76c8a1efcbfffdf93)

gabrielebaris, 2018-12-13:
Hi @rcacurs,
I cloned again from the development branch (v2.1.2) and indeed it compiles with librealsense v2.17.0 (installed via apt). Thanks a lot. Anyway, when I run any launch file dealing with the camera, it does not start. 
Here a log of what happens running `roslaunch realsense2_camera rs_camera.launch`:
```
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://127.0.0.1:34789/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 1000
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

[... 34 lines cut ...]
[ INFO] [1544691345.326192445]: setupStreams...
 13/12 08:55:45,326 WARNING [140314731857792] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ INFO] [1544691345.327383579]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1544691345.327526407]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1544691345.327667199]: infra2 stream is enabled - width: 640, height: 480, fps: 30
 13/12 08:55:45,328 WARNING [140314731857792] (backend-v4l2.cpp:1248) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 13/12 08:55:45,328 WARNING [140314731857792] (sensor.cpp:338) Unregistered Media formats : [ RW16 ]; Supported: [ ]
[ INFO] [1544691345.329471830]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1544691345.388800875]: num_filters: 0
[ INFO] [1544691345.388855805]: publishStaticTransforms...
[ INFO] [1544691345.389161970]: RealSense Node Is Up!
[ INFO] [1544691345.389212440]: Start updating dynamic parameters...
 13/12 08:55:45,440 WARNING [140314731857792] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 13/12 08:55:45,441 WARNING [140314731857792] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 13/12 08:55:45,441 WARNING [140314731857792] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 13/12 08:55:45,441 WARNING [140314731857792] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 13/12 08:55:45,502 ERROR [140314731857792] (types.h:188) xioctl(VIDIOC_S_CTRL) failed Last Error: Numerical result out of range
[ERROR] [1544691345.502622948]: Failed. Skip initialization of parameter 24
[ INFO] [1544691345.503156700]: Done updating dynamic parameters...
 13/12 08:55:45,577 WARNING [140313020970752] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ WARN] [1544691345.580695056]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 13/12 08:55:45,586 WARNING [140313031837440] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details
```
I am running ROS melodic inside a Docker container, using Ubuntu 18.04. I don't think is a problem related to drivers or the SDK since Python bindings seem to work well.
Thanks in advance.

doronhi, 2018-12-13:
I updated the development branch. The whole initialization of the dynamic parameters is replaced. Could you check if it solved the problem?

gabrielebaris, 2018-12-13:
Hi @doronhi,
I pulled your update and compiled again. It is working now
Thanks a lot

doronhi, 2018-12-13:
Thank you. I close the issue now. Please re-open if issue is not resolved.

closed 2018-12-13 by doronhi