realsenseai/realsense-ros #532: demo_pointcloud.launch cannot see pointcloud in rviz

dakejahl, 2018-12-13:
running `roslaunch realsense2_camera demo_pointcloud.launch`
[image]

I do not see anything in rviz. I've been successful in running `rs-pointcloud` in `/usr/local/bin` so I don't think there is anything wrong with my hardware. I am just not getting a point cloud visualization in rviz.

[image]

doronhi, 2018-12-13:
Where is the Error, Ignoring transform for child_frame_id "camera_color_frame" coming from?
I mean, is there anything else running that could publish a different transform for the frame_id "camera_color_frame"?

dakejahl, 2018-12-13:
I have no idea where the error is coming from. I have simply followed the instructions in the README and am trying to using one of the launch files to visualize a pointcloud. I have nothing else in ROS running. I am using firmware version 5.9.13 on the D435. I

dakejahl, 2018-12-13:
Same issue with `realsense` `2.1.2` and `librealsense` `2.17.0`
[image]

doronhi, 2018-12-16:
The values for the translation are not valid.
Could you run:
`rs-enumerate-devices -c`
and look in the output for a section starting with: "Extrinsic from "Color"	  To	  "Infrared 1"". Something like this:

```
Extrinsic from "Color"	  To	  "Infrared 1" :
Rotation Matrix:
0.99989        0.013115       -0.0062178     
-0.013114      0.99991        0.0001742      
0.0062195      -9.2639e-05    0.99998        

Translation Vector: -0.014903349801898  -0.000295773410471156  -0.000403936661314219  

```

pavloblindnology, 2018-12-18:
Confirm that pointcloud is not published. Extrinsics are ok.

doronhi, 2018-12-26:
Please reopen the issue if problem is not solved. Also, out of curiosity and if you have the time, i'd appreciate to learn what the problem was if it is solved.

closed 2018-12-26 by doronhi

mzahana, 2019-03-18:
I have the same problem. I am using realsense2 v2.2.1 and librealsense v2.19.0.
When I execute ```rs-enumerate-devices -c``` I get
```
Extrinsic from "Color"	  To	  "Infrared 1" :
Rotation Matrix:
0              0              0              
0              0              0              
0              0              0              

Translation Vector: 0  0  0 
```

What could be the issue?

nekovfra, 2019-05-04:
I am having the same issue with new D435i camera. The rotational matrices are also zero. Tried on both Ubuntu16/Kinetic and 18/Melodic, in VMware, on machine. Might be issue with the camera HW itself.

doronhi, 2019-05-05:
It seems like a known firmware bug. You can stay updated at [issue#3788](https://github.com/IntelRealSense/librealsense/issues/3788)