realsenseai/realsense-ros #540: Roslaunch realsense2_camera rs_camera.launch error

hankewei, 2018-12-17:
$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/data/.ros/log/90fd2724-01e4-11e9-8477-2cfda1be608f/roslaunch-data-System-Product-Name-17501.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://data-System-Product-Name:36161/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [17512]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 90fd2724-01e4-11e9-8477-2cfda1be608f
process[rosout-1]: started with pid [17525]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [17542]
process[camera/realsense2_camera-3]: started with pid [17543]
[ INFO] [1545041681.316954481]: Initializing nodelet with 12 worker threads.
[ INFO] [1545041681.345171734]: RealSense ROS v2.1.2
[ INFO] [1545041681.345195337]: Running with LibRealSense v2.17.0
[ INFO] [1545041681.527233192]: getParameters...
[ INFO] [1545041681.558879507]: setupDevice...
[ INFO] [1545041681.558896460]: JSON file is not provided
[ INFO] [1545041681.558908016]: ROS Node Namespace: camera
[ INFO] [1545041681.558916873]: Device Name: Intel RealSense SR300
[ INFO] [1545041681.558925199]: Device Serial No: 612203004489
[ INFO] [1545041681.558934304]: Device FW version: 3.21.0.0
[ INFO] [1545041681.558942204]: Device Product ID: 0x0AA5
[ INFO] [1545041681.558949521]: Enable PointCloud: Off
[ INFO] [1545041681.558957918]: Align Depth: Off
[ INFO] [1545041681.558965484]: Sync Mode: Off
[ INFO] [1545041681.558986352]: Device Sensors: 
[ INFO] [1545041681.559002449]: Coded-Light Depth Sensor was found.
[ INFO] [1545041681.559011931]: RGB Camera was found.
[ INFO] [1545041681.559032348]: (Infrared, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1545041681.559042181]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1545041681.559050489]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1545041681.559058015]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1545041681.559071334]: setupPublishers...
[ INFO] [1545041681.560683402]: Expected frequency for depth = 30.00000
[ INFO] [1545041681.579627957]: Expected frequency for infra1 = 30.00000
[ INFO] [1545041681.597876995]: Expected frequency for color = 30.00000
[ INFO] [1545041681.617351155]: setupStreams...
 17/12 18:14:41,617 WARNING [140202630846336] (sensor.cpp:338) Unregistered Media formats : [ INZC PAIR Y10  YUYV ]; Supported: [ ]
[ INFO] [1545041681.617974279]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1545041681.618162193]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1545041681.619391644]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1545041681.626431307]: num_filters: 0
[ INFO] [1545041681.626540457]: RealSense Node Is Up!
[ INFO] [1545041681.626561484]: Setting Dynamic reconfig parameters.
[ERROR] [1545041681.626729099]: **An exception has been thrown: Character [-] at element [5] is not valid in Graph Resource Name [Coded-Light_Depth_Sensor].  Valid characters are a-z, A-Z, 0-9, / and _.**
[FATAL] [1545041681.705673648]: **Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'**
[camera/realsense2_camera_manager-2] process has died [pid 17542, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/data/.ros/log/90fd2724-01e4-11e9-8477-2cfda1be608f/camera-realsense2_camera_manager-2.log].
log file: /home/data/.ros/log/90fd2724-01e4-11e9-8477-2cfda1be608f/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has died [pid 17543, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/data/.ros/log/90fd2724-01e4-11e9-8477-2cfda1be608f/camera-realsense2_camera-3.log].
log file: /home/data/.ros/log/90fd2724-01e4-11e9-8477-2cfda1be608f/camera-realsense2_camera-3*.log

doronhi, 2018-12-18:
Did you check out the last commit, where I merged PR #537 by @carlos-wong ?

carlos-wong, 2018-12-18:
@dorodnic I met same log before I made the PR

closed 2018-12-26 by doronhi