realsenseai/realsense-ros #543: Urdf and mesh for D415

tragolf, 2018-12-19:
Hi,

I have been using the D435 with ROS. For this model I could find correct URDF-Files and Mesh. Now I want to use the D415 but I can't find the URDF and Mesh. Therefore I can't add the camera to the endeffector of my robot in MoveIt. The transformation between the camera coordinatesystem and the robot is very off.
Does anyone know where to get these files? Especially the internal tf between cam left, cam right, depth... frames is important.
Thanks

SteveMacenski, 2018-12-19:
Adding to this a little, the mesh for the d435 is super high resolution, it actually takes more cpu to render it than the rest of my robot combined

elbartojs, 2019-01-09:
I found the 3D model here : 
https://www.intel.fr/content/www/fr/fr/support/articles/000026841/emerging-technologies/intel-realsense-technology.html

But I'm also looking for the URDF of the D415, does someone have it ?

doronhi, 2019-01-27:
Release 2.1.4 includes urdf file for D415 (add in PR #585)

closed 2019-01-27 by doronhi