realsenseai/realsense-ros #544: Are the united IMU messages wrong? [branch: development] [D435i]

mfehr, 2018-12-19:
Hi realsense team,

First of all, thanks again for pushing the update to the development branch that allows publishing IMU data for the D435i so quickly. (#518)

Upon closer inspection of the IMU messages and the code that is publishing it i noticed the following:

The logic that unites imu messages is wrong IMO. If I'm not mistaken the current implementation does the following: If there is an imu callback it distinguishes accel/gyro types, and then just updates the gyro or accel part of the imu message and publishes it with the timestamp of that frame/measurement.

This leads to IMU messages being published at accel_fps + gyro_fps, which not only contain the same measurements several times, but the timestamp in the IMU message header corresponds only to one of the samples, either gyro or accel, never both (I guess unless the samples happen to fall on the same timestamp every now and then). Only the measurement of the most recent stream update changes. 

For a gyro @ 400Hz and accel @ 250Hz it publishes messages at 650hz.

```bash
subscribed to [/camera/imu]
average rate: 649.598
```
Here is an example of three consecutive IMU messages, I removed all the irrelevant parts:

```bash
---
header: 
  seq: 2773
  stamp: 
    secs: 1545251443
    nsecs: 595507383
angular_velocity: 
  x: -0.0
  y: 0.00174532923847
  z: 0.0069813169539
linear_acceleration: 
  x: -1.01412665844
  y: 1.88056492805
  z: 9.58005046844
---
header: 
  seq: 2774
  stamp: 
    secs: 1545251443
    nsecs: 597342491
angular_velocity: 
  x: -0.0
  y: 0.00174532923847
  z: 0.0069813169539
linear_acceleration: 
  x: -0.974743068218
  y: 1.90025663376
  z: 9.56035804749
---
header: 
  seq: 2775
  stamp: 
    secs: 1545251443
    nsecs: 598009348
angular_velocity: 
  x: -0.0
  y: 0.0
  z: 0.00872664619237
linear_acceleration: 
  x: -0.974743068218
  y: 1.90025663376
  z: 9.56035804749
```

Basically what's happening is the following:
 ```bash
    # IMU message
    # [timestamp, accelerometer measurements, gyroscope measurements]
    [t_1, meas_acc_t_1, meas_gyr_t_0],
    [t_2, meas_acc_t_2, meas_gyr_t_0]
    [t_3, meas_acc_t_2, meas_gyr_t_3]
```
I tried to create a visual-inertial map with maplab, but it was very unstable. After discovering this IMU issue, I implemented a rather hacky fix that takes the slower signal and interpolates the faster signal to match it. The code is not pretty, probably drops some samples (the sliding window used to interpolate is only one measurement wide) and I'm not suggesting to merge it or anything, but with this fix I actually managed to do some visual inertial mapping using maplab. The code can be found [here](https://github.com/intel-ros/realsense/compare/development...ethz-asl:devel/d435i) if anyone is interested.

What do you think? Maybe I'm missing something or it is a D435i specific issue, but as I see it the IMU messages of this node are wrong/redundant. Also, since the FPS profiles for gyro and accel are completely different, it's not possible to set the same sampling frequency, therefore even though I haven't tested all the configurations, I think this issue will be the same.

doronhi, 2018-12-19:
Hi,
You are completely right about the mechanism of combining together the IMU components. My thoughts were that it was better to duplicate information than to drop some of it off.

Maybe I was mistaken and that’s one of the reasons this is still in development and not set as a “release” yet.

I would like to check your code so if you could pass your testing app as well, it will give me something to work with.
Also, I’d like to take the opportunity and ask for more thoughts and insights about this topic that you or anybody else may have.

mfehr, 2018-12-19:
Hi @doronhi 
Thanks for the quick reply.
The code can be found on our [fork in the devel/d435i branch](https://github.com/intel-ros/realsense/compare/development...ethz-asl:devel/d435i). The mapping framework I was refering to is [maplab on the july release candidate branch](https://github.com/ethz-asl/maplab/tree/pre_release_public/july-2018). [Here](https://gist.github.com/mfehr/3e76d5515531c4ed7028503d9be079b6) are the IMU sigmas, an example camera calibration and bash script I used for maplab.

Regarding my comment about dropping measurements: the logic I implemented of course drops data for the faster signal, but the resulting synchronized messages are stable at accel_fps. I just wanted to point out that I'm not 100% confident my code doesn't also drop an accel measurement every now and then, because it is only a very small sliding window.

From what I've seen with the estimator having a lower frequency but more accurate IMU measurements is better, which intuitively makes also more sense to me. But on the other hand, there are so many variables in an estimator like maplab, there might also be some other issues with the calibration or IMU parameters that I'm using.

Regarding how to best interpolate: interpolating based on more than two data points (polynomial or spline) could also improve the data, but also increases the delay. With the linear interpolation the messages shouldn't be delayed by more than one measurement of the slower signal. But maybe someone with more experience in signal processing and state estimation should answer this question :)

If a user is interested in the individual gyro/accel samples at the original frequency, as you know, the node can publish them separately as well.

eaglej, 2018-12-21:
@mfehr thanks for raising this!

I would suggest treating the accel and gyro as completely independent sensors and not duplicating or interpolating data.

When you have data from the accel, put -1 in element 0 of the gyro matrix and when you have data from the gyro, put -1 in element 0 of the accel matrix.

http://docs.ros.org/api/sensor_msgs/html/msg/Imu.html

mfehr, 2018-12-21:
@eaglej
This option is already available in the driver, have a look [here](https://github.com/intel-ros/realsense/blob/development/realsense2_camera/src/base_realsense_node.cpp#L366). But it's off by default and I haven't checked whether it follows the convention you mentioned. I agree it should probably be the default behavior, as it gives the user access to the raw, uninterpolated data.

I disagree about getting rid of the option to combine the messages altogether. There are many applications that by now treat the IMU sensor as combined sensor and therefore expect a combined set of measurements of gyro/accelerometer. Pushing the responsibility of writing an interface to combine the two separate streams to the end-user seems unnecessary, especially considering how easy it is to do this here, where the transmission over ROS hasn't yet added another level of difficulty in terms of dealing with delayed, dropped or reordered messages. Considering that the realsense is already providing a lot of processed data, such as point clouds and aligned images, having a stream of combined IMU messages seems like something you would expect to get as well :)

doronhi, 2018-12-27:
I add my version of interpolating the data here:
https://github.com/doronhi/realsense.git

I may merge it into the main repo sometimes next week. I didn't get around yet to check it using @mfehr setup so I didn't have a proper way of estimating its value yet. I plan to do it prior to it's release but you are more then welcome and I'd really appreciate it if you tried it out.
It has a latency of up to 1 message but as it has a frequency of almost twice the slow sensor (that is 2*250 = 500 Hz) I hope it's not too much.

trixr4kdz, 2019-01-05:
Hello,

On the topic of united IMU message, I noticed that the orientation quaternion is set to all zeroes. Will there be an update soon to provide the actual orientation angles?

trixr4kdz, 2019-01-05:
Ah I see. That works. Thanks @hvpandya!

closed 2019-01-06 by doronhi