realsenseai/realsense-ros #547: Compilation error

atyshka, 2018-12-20:
I've just pulled the latest changes to the repo (excited about the reset pr!) and I'm unable to build. Here's the error I'm getting:

`/usr/include/c++/7/bits/stl_iterator_base_types.h:205:5: error: no type named ‘iterator_category’ in ‘struct std::iterator_traits<rs2::frameset::iterator>’`

I'm using the latest realsense drivers. Was there a recent API change that broke the code? Can anyone else reproduce this issue?

doronhi, 2018-12-20:
Sorry about that. You need to take the latest development branch from librealsense.
It’s been update a couple of hours ago.

doronhi, 2018-12-21:
You are absolutely right. The next release will be coordinated with librealsense’s. Hopefully in a matter of days.

closed 2018-12-21 by doronhi

mattcorsaro1, 2019-02-15:
I'm still getting this error. I'm using librealsense-2.17.0 and the development branch of this package. ROS Kinetic on 16.04, kernel 4.4.0-142-generic.

`In file included from /usr/include/c++/5/algorithm:62:0,
                 from /usr/include/boost/smart_ptr/shared_ptr.hpp:37,
                 from /usr/include/boost/shared_ptr.hpp:17,
                 from /opt/ros/kinetic/include/class_loader/./class_loader.hpp:36,
                 from /opt/ros/kinetic/include/class_loader/class_loader.h:35,
                 from /opt/ros/kinetic/include/pluginlib/./class_list_macros.hpp:40,
                 from /opt/ros/kinetic/include/pluginlib/class_list_macros.h:35,
                 from /home/mcorsaro/kuka_ws/src/realsense/realsense2_camera/src/../include/../include/realsense_node_factory.h:6,
                 from /home/mcorsaro/kuka_ws/src/realsense/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/mcorsaro/kuka_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:1:
/usr/include/c++/5/bits/stl_algo.h: In instantiation of ‘_Iterator std::__find_if(_Iterator, _Iterator, _Predicate) [with _Iterator = rs2::frameset::iterator; _Predicate = __gnu_cxx::__ops::_Iter_pred<realsense2_camera::BaseRealSenseNode::publishPointCloud(rs2::points, const ros::Time&, const rs2::frameset&)::<lambda(rs2::frame)> >]’:
/usr/include/c++/5/bits/stl_algo.h:3815:28:   required from ‘_IIter std::find_if(_IIter, _IIter, _Predicate) [with _IIter = rs2::frameset::iterator; _Predicate = realsense2_camera::BaseRealSenseNode::publishPointCloud(rs2::points, const ros::Time&, const rs2::frameset&)::<lambda(rs2::frame)>]’
/home/mcorsaro/kuka_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:1717:125:   required from here
/usr/include/c++/5/bits/stl_algo.h:162:34: error: no matching function for call to ‘__iterator_category(rs2::frameset::iterator&)’
          std::__iterator_category(__first));`

mattcorsaro1, 2019-02-15:
I fixed the error by installing librealsense-2.18.1 from source. I got the above error when using the latest debs.