realsenseai/realsense-ros #550: Compare to FlowNet calculated depth, which one is more accurate

jinfagang, 2018-12-21:
We are using realsense for accurate automous car outdoor navigation, so the accurate depth is very important. 

As far as we doing, the realsense camera produces points are very noisy....... If we calculate the depth with FlowNet etc, which one will be more accurate compare to the directly produced depth value?

BTW, how do filter the points generated by depth image?

RealSenseCustomerSupport, 2019-04-18:
----------------------------------------------

Hi Jinfagang,

This could give you some info on using filters off depth images:
https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md

Thanks!

RealSenseCustomerSupport, 2019-04-30:
----------------------------------------------

Hi Jinfagang,

Wonder if there is any update on this one?

Thanks!

jinfagang, 2019-05-01:
@RealSenseCustomerSupport  After we using filters, the result seems has no effect

RealSenseCustomerSupport, 2019-05-07:
----------------------------------------------

Hi jinfagang,

Can you please elaborate more details on what is not taking effect?
Some of the filter options may not change depth image dramatically but make some subtle changes at a time.
You may try to change between MIN and MAX value of a filter and see how much it can change.

On the other side, there a few whitepaper related to depth tuning. Please go through if not yet:
https://dev.intelrealsense.com/docs/whitepapers?_ga=2.225639123.853300450.1557166967-151375683.1557166967

Thanks!

RealSenseCustomerSupport, 2019-05-16:
----------------------------------------------

Hi jinfagang,

Wonder if there is any update on this one? Or do you still need help?
If nothing else is needed, this one will be closed.

Thanks!

closed 2019-05-29 by RealSenseCustomerSupport