realsenseai/realsense-ros #551: How to get 3D point from depth image

jinfagang, 2018-12-21:
Hi, I know there is some point cloud topic send out points, however, I need to calculate specific image coordinate into point cloud, here  is what I am currently do:

```
for (int j = 0; j < crt_img_color.rows; ++j) {
      for (int k = 0; k < crt_img_color.cols; ++k) {
        uint16_t d = depth_ptr[j * crt_img_color.rows + k];
        float real_dist = d * rs_real_distance_scale;
        if (real_dist <= 40) {
          // 2. with K, get the camera coordinate, (k, j) -> (X, Y, Z)
          float x = (k - cx) / fx;
          float y = (j - cy) / fy;

          float X = x * real_dist;
          float Y = y * real_dist;
          float Z = real_dist;
          // 3. with R, t, get the world coordinate, (X, Y, Z) -> (U, V, W)
          pcl::PointXYZRGB p;
          p.x = X;
          p.y = Y;
          p.z = Z;

          // segment color
          if (0 <= k && k <= 200) {
            p.r = 255;
            p.g = 255;
            p.b = 0;
          } else if (k >= 500){
            p.r = 0;
            p.g = 255;
            p.b = 0;
          } else {
            p.r = 255;
            p.g = 0;
            p.b = 127;
          }
          whole_cloud->points.push_back(p);
        }
      }
    }
```

This produces some abnormal points.......... I want to know how to iterate depth image and get the depth, map it to 3D point and orgnise into a point cloud, is there any snippets to do this?

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

jlysnantel, 2019-03-31:
Hi I would like to know if you are using the intresect informations and if so, how To find them. I am refferring to this documentation: 
https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0

The deprojection section.

RealSenseCustomerSupport, 2019-04-25:
----------------------------------------------

Hi jinfagang,

For converting depth and point cloud, there is a python example which could provide some ideas:

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/helper_functions.py
a) depth to point cloud https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/helper_functions.py
b) point cloud to depth https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/helper_functions.py#L191

Hope this can be helpful.

Thanks!

RealSenseCustomerSupport, 2019-05-07:
----------------------------------------------

Hi jinfagang,

Wonder if there is any update?
If nothing else is needed, this one will be closed.

Thanks!

jinfagang, 2019-05-07:
@RealSenseCustomerSupport thanks for your reply. I am actually using this one:

```
def convert_depth_pixel_to_metric_coordinate(depth, pixel_x, pixel_y, camera_intrinsics):
	"""
	Convert the depth and image point information to metric coordinates
	Parameters:
	-----------
	depth 	 	 	 : double
						   The depth value of the image point
	pixel_x 	  	 	 : double
						   The x value of the image coordinate
	pixel_y 	  	 	 : double
							The y value of the image coordinate
	camera_intrinsics : The intrinsic values of the imager in whose coordinate system the depth_frame is computed
	Return:
	----------
	X : double
		The x value in meters
	Y : double
		The y value in meters
	Z : double
		The z value in meters
	"""
	X = (pixel_x - camera_intrinsics.ppx)/camera_intrinsics.fx *depth
	Y = (pixel_y - camera_intrinsics.ppy)/camera_intrinsics.fy *depth
	return X, Y, depth
```

converting depth to point cloud at some specific pixel.

But, my calculation are not same with realsense ROS wrapper ( sometimes my point were offset a lot than that point cloud). Since realsense depth are not accurate always, how do I applying filters to get same result in ROS wrapper?(the ros wrapper can publish point cloud topic)

RealSenseCustomerSupport, 2019-05-13:
----------------------------------------------

Hi jinfagang,

For ROS wrapper, one way that you can do is adding filters to the "filters" field in a ROS launch file.
For example I am adding decimation and temporal filters to a ROS launch file
...
  <arg name="enable_pointcloud"  default="false"/>
  <arg name="enable_sync"        default="true"/>
  <arg name="align_depth"        default="true"/>

  <arg name="filters"            default="decimation,temporal"/>

  <group ns="$(arg camera)">
...

Hope this can be helpful.

Thanks!

jinfagang, 2019-05-14:
@RealSenseCustomerSupport  But My launch file does not set filters, the ROS wrapper point cloud can also get accurate result....

RealSenseCustomerSupport, 2019-05-14:
----------------------------------------------

Hi jinfagang,

For using filters in python code, you could refer to the example of box measurement under python wrapper:
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/realsense_device_manager.py#L49

Thanks!

RealSenseCustomerSupport, 2019-05-23:
----------------------------------------------

Hi jinfagang,

Any update on this one? Do you still need any help?

Thanks!

jinfagang, 2019-05-24:
Yes, I need help. Your answer does not solve problem yet. You let me set filters in launch file, but I does not  set filters your roswrapper get accurate point cloud too and I want did the same thing...

the ros wapper does not provide any useful code snippets do this. it needs librealsense but that I can not using it since the device already using by ros realsense wrapper (i can not open that device at the same time).

I don't know if you understand my situation..... but, simply, I want *a code snippet*  which converts `(px, py)` on RGB image to real point cloud. However, I am not stupid, the function I post above were slightly different than ros wrapper topic output. I want that calculation process.

RealSenseCustomerSupport, 2019-07-02:
----------------------------------------------

@jinfagang I'm taking over this ticket and would like to understand if your issue is that there's some offset from the point cloud created by ROS wrapper if you refer to https://github.com/IntelRealSense/librealsense/blob/v2.24.0/wrappers/python/examples/box_dimensioner_multicam/helper_functions.py#L121-L147 to create point cloud.  Could you please tell the offset details(on x, y or z) or provide some issue snapshot so that we can understand the issue more clearly? Thanks!

closed 2019-07-02 by jinfagang