realsenseai/realsense-ros #553: /camera/depth_registered/points err point cloud distance

woshisj, 2018-12-22:
when I run the command
$ roslaunch realsense_ros_camera rs_rgbd.launch align_depth:=true

view the /camera/depth_registered/points in rviz
and I measure the distance is 1.332m
[image]
while measure the same place  in  the  /camera/depth/color/points  topic 0.1596m
in face the last result correct
[image]

Thompson104, 2018-12-27:
I also have the same issue and the details can be found [here](https://answers.ros.org/question/311478/point-cloud-data-appears-larger-than-it-should-be-in-rviz/).

doronhi, 2019-01-27:
Hi @woshisj,
Sorry for the long delay. 
Can you confirm that the problem still exists with the latest release (2.1.4(ROS) and 2.18.0)?
I tried but could not reproduce.
I get the same correct measures on topic "/camera/depth_registered/points" when I run:
`roslaunch realsense2_camera rs_rgbd.launch align_depth:=true`

as when I run 
`roslaunch realsense2_camera rs_camera.launch filters:=pointcloud align_depth:=true`
which is what I assume you used for getting the topic "/camera/depth/color/points"

RealSenseCustomerSupport, 2019-04-04:
----------------------------------------------

@woshisj Did you get chance to try the latest release(2.2.2(ROS) and 2.19.2) to see if the issue still existing? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-04-08:
----------------------------------------------

@woshisj Any update? Thanks!

woshisj, 2019-04-09:
> @woshisj Any update? Thanks!

sorry， I have not used SR300 recently.
I will update it the next time I use it.

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

@woshisj Ok. If no other questions about the ticket, will close it accordingly. Please feel free to open new ticket if you still get the issue when use SR300 next time. Thanks!

closed 2019-05-03 by RealSenseCustomerSupport