realsenseai/realsense-ros #559: roslaunch realsense2_camera rs_camera.launch

lzy119, 2019-01-02:
The following error occurs when I type the above command.
I would appreciate it very much if you could help me.thank you.
[ WARN] [1546435342.373672375]: Unrecognized sr300 param (21)
[ WARN] [1546435342.373732382]: Unrecognized sr300 param (22)
[ WARN] [1546435342.373753224]: Unrecognized sr300 param (23)
[ WARN] [1546435342.373780888]: Unrecognized sr300 param (24)
[ WARN] [1546435342.373813235]: Unrecognized sr300 param (25)
[ WARN] [1546435342.373831495]: Unrecognized sr300 param (26)
[ WARN] [1546435342.373851137]: Unrecognized sr300 param (27)
[ WARN] [1546435342.373875972]: Unrecognized sr300 param (28)
[ WARN] [1546435342.373893820]: Unrecognized sr300 param (29)
[ WARN] [1546435342.373913359]: Unrecognized sr300 param (30)
[ WARN] [1546435342.373951892]: Unrecognized sr300 param (31)
[ WARN] [1546435342.373982814]: Unrecognized sr300 param (32)
[ WARN] [1546435342.374008295]: Unrecognized sr300 param (33)
[ WARN] [1546435342.374048348]: Unrecognized sr300 param (34)
[ WARN] [1546435342.374071746]: Unrecognized sr300 param (35)
[ WARN] [1546435342.374100028]: Unrecognized sr300 param (36)
[ WARN] [1546435342.374146802]: Unrecognized sr300 param (37)
[ WARN] [1546435342.374185249]: Unrecognized sr300 param (38)
[ WARN] [1546435342.374212312]: Unrecognized sr300 param (39)
[ WARN] [1546435342.374247557]: Unrecognized sr300 param (40)
[ WARN] [1546435342.374273381]: Unrecognized sr300 param (41)
[ WARN] [1546435342.374306714]: Unrecognized sr300 param (42)
[ WARN] [1546435342.374331770]: Unrecognized sr300 param (43)
[ WARN] [1546435342.374358502]: Unrecognized sr300 param (44)
[ WARN] [1546435342.374384938]: Unrecognized sr300 param (45)
[ WARN] [1546435342.374405938]: Unrecognized sr300 param (46)
[ WARN] [1546435342.374426241]: Unrecognized sr300 param (47)
[ WARN] [1546435342.374447490]: Unrecognized sr300 param (48)
[ WARN] [1546435342.374479007]: Unrecognized sr300 param (49)
[ WARN] [1546435342.374500521]: Unrecognized sr300 param (50)
[ WARN] [1546435342.374543986]: Unrecognized sr300 param (51)
[ WARN] [1546435342.374571189]: Unrecognized sr300 param (52)
[ WARN] [1546435342.374595798]: Unrecognized sr300 param (53)
[ WARN] [1546435342.374640690]: Unrecognized sr300 param (54)
[ WARN] [1546435342.374677173]: Unrecognized sr300 param (55)
[ WARN] [1546435342.374705767]: Unrecognized sr300 param (56)
[ WARN] [1546435342.374735149]: Unrecognized sr300 param (57)
[ WARN] [1546435342.374763160]: Unrecognized sr300 param (58)
[ WARN] [1546435342.374792351]: Unrecognized sr300 param (59)
[ WARN] [1546435342.374816990]: Unrecognized sr300 param (60)
[ WARN] [1546435342.374854529]: Unrecognized sr300 param (61)
[ WARN] [1546435342.374876777]: Unrecognized sr300 param (62)
[ WARN] [1546435342.374908729]: Unrecognized sr300 param (63)
[ WARN] [1546435342.374930918]: Unrecognized sr300 param (64)
[ WARN] [1546435342.374962375]: Unrecognized sr300 param (65)
[ WARN] [1546435342.374984663]: Unrecognized sr300 param (66)
[ WARN] [1546435342.375017475]: Unrecognized sr300 param (67)
[ WARN] [1546435342.375039638]: Unrecognized sr300 param (68)
[ WARN] [1546435342.375069580]: Unrecognized sr300 param (69)
[ WARN] [1546435342.375091312]: Unrecognized sr300 param (70)
[ WARN] [1546435342.375122586]: Unrecognized sr300 param (71)
[ WARN] [1546435342.375145301]: Unrecognized sr300 param (72)
[ WARN] [1546435342.375163543]: Unrecognized sr300 param (73)
[ WARN] [1546435342.375189193]: Unrecognized sr300 param (74)
[ WARN] [1546435342.375206930]: Unrecognized sr300 param (75)
[ WARN] [1546435342.375232762]: Unrecognized sr300 param (76)
[ WARN] [1546435342.375265971]: Unrecognized sr300 param (77)
[ WARN] [1546435342.375287019]: Unrecognized sr300 param (78)
[ WARN] [1546435342.375319835]: Unrecognized sr300 param (79)
[ WARN] [1546435342.375341874]: Unrecognized sr300 param (80)
[ WARN] [1546435342.375369984]: Unrecognized sr300 param (81)
[ WARN] [1546435342.375392323]: Unrecognized sr300 param (82)
[ WARN] [1546435342.375422700]: Unrecognized sr300 param (83)
[ WARN] [1546435342.375444518]: Unrecognized sr300 param (84)
[ WARN] [1546435342.375475476]: Unrecognized sr300 param (85)
[ WARN] [1546435342.375497541]: Unrecognized sr300 param (86)
[ WARN] [1546435342.375527260]: Unrecognized sr300 param (87)
[ WARN] [1546435342.375548860]: Unrecognized sr300 param (88)
[ WARN] [1546435342.375579681]: Unrecognized sr300 param (89)
[ WARN] [1546435342.375600898]: Unrecognized sr300 param (90)
[ WARN] [1546435342.375629701]: Unrecognized sr300 param (91)
[ WARN] [1546435342.375651852]: Unrecognized sr300 param (92)
[ WARN] [1546435342.375709727]: Unrecognized sr300 param (93)
[ WARN] [1546435342.375728081]: Unrecognized sr300 param (94)
[ WARN] [1546435342.375751594]: Unrecognized sr300 param (95)
[ WARN] [1546435342.375768157]: Unrecognized sr300 param (96)
[ WARN] [1546435342.375789459]: Unrecognized sr300 param (97)
[ WARN] [1546435342.375806466]: Unrecognized sr300 param (98)
[ WARN] [1546435342.375820861]: Unrecognized sr300 param (99)
[ WARN] [1546435342.375841496]: Unrecognized sr300 param (100)

doronhi, 2019-01-02:
What version are running? Could you provide the whole log?

lzy119, 2019-01-03:
Thank you for your reply.
Running with LibRealSense v2.17.0 
Here is another part of my run log.
PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 1000
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [13383]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 4e571888-0f0b-11e9-a78b-303a645f2481
process[rosout-1]: started with pid [13396]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [13399]
process[camera/realsense2_camera-3]: started with pid [13400]
[ INFO] [1546487689.879010185]: Initializing nodelet with 8 worker threads.
[ INFO] [1546487691.295919409]: RealSense ROS v2.1.2
[ INFO] [1546487691.295959054]: Running with LibRealSense v2.17.0
[ INFO] [1546487691.610549864]: getParameters...
[ INFO] [1546487691.652034764]: setupDevice...
[ INFO] [1546487691.652066207]: JSON file is not provided
[ INFO] [1546487691.652092689]: ROS Node Namespace: camera
[ INFO] [1546487691.652118140]: Device Name: Intel RealSense SR300
[ INFO] [1546487691.652133562]: Device Serial No: 612201005767
[ INFO] [1546487691.652147773]: Device FW version: 3.15.0.0
[ INFO] [1546487691.652161790]: Device Product ID: 0x0AA5
[ INFO] [1546487691.652176976]: Enable PointCloud: Off
[ INFO] [1546487691.652190552]: Align Depth: Off
[ INFO] [1546487691.652204024]: Sync Mode: Off
[ INFO] [1546487691.652284236]: Device Sensors: 
[ INFO] [1546487691.652315530]: Coded-Light Depth Sensor was found.
[ INFO] [1546487691.652334148]: RGB Camera was found.
[ INFO] [1546487691.652374840]: (Infrared, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1546487691.652391005]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1546487691.652406084]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1546487691.652420173]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1546487691.652440121]: setupPublishers...
[ INFO] [1546487691.687903616]: Expected frequency for depth = 30.00000
[ INFO] [1546487691.782109213]: Expected frequency for infra1 = 30.00000
[ INFO] [1546487691.804031864]: Expected frequency for color = 30.00000
[ INFO] [1546487691.826887309]: setupStreams...
 03/01 11:54:51,828 WARNING [139882169071488] (sensor.cpp:338) Unregistered Media formats : [ INZC PAIR YUYV ]; Supported: [ ]
[ INFO] [1546487691.874439582]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1546487691.874668169]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1546487691.879045095]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1546487691.889508440]: num_filters: 0
[ INFO] [1546487691.889562350]: publishStaticTransforms...
[ INFO] [1546487691.889730548]: RealSense Node Is Up!

lzy119, 2019-01-03:
> What version are running? Could you provide the whole log?

Thank you for your reply.
Running with LibRealSense v2.17.0
Here is another part of my run log.
PARAMETERS

/camera/realsense2_camera/accel_fps: 1000
/camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
/camera/realsense2_camera/align_depth: False
/camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
/camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
/camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
/camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
/camera/realsense2_camera/base_frame_id: camera_link
/camera/realsense2_camera/color_fps: 30
/camera/realsense2_camera/color_frame_id: camera_color_frame
/camera/realsense2_camera/color_height: 480
/camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
/camera/realsense2_camera/color_width: 640
/camera/realsense2_camera/depth_fps: 30
/camera/realsense2_camera/depth_frame_id: camera_depth_frame
/camera/realsense2_camera/depth_height: 480
/camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
/camera/realsense2_camera/depth_width: 640
/camera/realsense2_camera/enable_color: True
/camera/realsense2_camera/enable_depth: True
/camera/realsense2_camera/enable_fisheye: True
/camera/realsense2_camera/enable_imu: True
/camera/realsense2_camera/enable_infra1: True
/camera/realsense2_camera/enable_infra2: True
/camera/realsense2_camera/enable_pointcloud: False
/camera/realsense2_camera/enable_sync: False
/camera/realsense2_camera/filters:
/camera/realsense2_camera/fisheye_fps: 30
/camera/realsense2_camera/fisheye_height: 480
/camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
/camera/realsense2_camera/fisheye_width: 640
/camera/realsense2_camera/gyro_fps: 1000
/camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
/camera/realsense2_camera/infra1_fps: 30
/camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
/camera/realsense2_camera/infra1_height: 480
/camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
/camera/realsense2_camera/infra1_width: 640
/camera/realsense2_camera/infra2_fps: 30
/camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
/camera/realsense2_camera/infra2_height: 480
/camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
/camera/realsense2_camera/infra2_width: 640
/camera/realsense2_camera/json_file_path:
/camera/realsense2_camera/pointcloud_texture_index: 0
/camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
/camera/realsense2_camera/rosbag_filename:
/camera/realsense2_camera/serial_no:
/rosdistro: kinetic
/rosversion: 1.12.14
NODES
/camera/
realsense2_camera (nodelet/nodelet)
realsense2_camera_manager (nodelet/nodelet)
auto-starting new master
process[master]: started with pid [13383]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 4e571888-0f0b-11e9-a78b-303a645f2481
process[rosout-1]: started with pid [13396]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [13399]
process[camera/realsense2_camera-3]: started with pid [13400]
[ INFO] [1546487689.879010185]: Initializing nodelet with 8 worker threads.
[ INFO] [1546487691.295919409]: RealSense ROS v2.1.2
[ INFO] [1546487691.295959054]: Running with LibRealSense v2.17.0
[ INFO] [1546487691.610549864]: getParameters...
[ INFO] [1546487691.652034764]: setupDevice...
[ INFO] [1546487691.652066207]: JSON file is not provided
[ INFO] [1546487691.652092689]: ROS Node Namespace: camera
[ INFO] [1546487691.652118140]: Device Name: Intel RealSense SR300
[ INFO] [1546487691.652133562]: Device Serial No: 612201005767
[ INFO] [1546487691.652147773]: Device FW version: 3.15.0.0
[ INFO] [1546487691.652161790]: Device Product ID: 0x0AA5
[ INFO] [1546487691.652176976]: Enable PointCloud: Off
[ INFO] [1546487691.652190552]: Align Depth: Off
[ INFO] [1546487691.652204024]: Sync Mode: Off
[ INFO] [1546487691.652284236]: Device Sensors:
[ INFO] [1546487691.652315530]: Coded-Light Depth Sensor was found.
[ INFO] [1546487691.652334148]: RGB Camera was found.
[ INFO] [1546487691.652374840]: (Infrared, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1546487691.652391005]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1546487691.652406084]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1546487691.652420173]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1546487691.652440121]: setupPublishers...
[ INFO] [1546487691.687903616]: Expected frequency for depth = 30.00000
[ INFO] [1546487691.782109213]: Expected frequency for infra1 = 30.00000
[ INFO] [1546487691.804031864]: Expected frequency for color = 30.00000
[ INFO] [1546487691.826887309]: setupStreams...
03/01 11:54:51,828 WARNING [139882169071488] (sensor.cpp:338) Unregistered Media formats : [ INZC PAIR YUYV ]; Supported: [ ]
[ INFO] [1546487691.874439582]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1546487691.874668169]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1546487691.879045095]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1546487691.889508440]: num_filters: 0
[ INFO] [1546487691.889562350]: publishStaticTransforms...
[ INFO] [1546487691.889730548]: RealSense Node Is Up!

doronhi, 2019-01-03:
Can I see the whole log in one piece please?
I asked about the version of realsense2_camera. I think you may need to upgrade that one.

doronhi, 2019-01-03:
I’m sorry, I forgot that the latest release has a bug and still prints v2.1.2 instead of 2.1.3 so my question is are you using the latest release (released Tuesday)? These warnings should not be there.

lzy119, 2019-01-04:
> 对不起，我忘了最新版本有错误，仍然打印v2.1.2而不是2.1.3所以我的问题是你使用的是最新版本（周二发布）吗？这些警告不应该存在。
No, I installed it before Tuesday.

doronhi, 2019-01-04:
Sorry, can’t read the font..

lzy119, 2019-01-04:
No, I installed it before Tuesday.

doronhi, 2019-01-06:
As far as I can say, the messages "[ WARN] [1546435342.375651852]: Unrecognized sr300 param (xx)" are meaningless and have not effect on the performance. In any way, they were removed in the last release.

closed 2019-04-07 by doronhi