realsenseai/realsense-ros #561: realsense2_camera rs_rgbd.launch.error

gxxzs, 2019-01-07:
roslaunch realsense2_camera rs_rgbd.launch 
... logging to /home/zs/.ros/log/50a6f76e-1304-11e9-afe9-24050f12dd45/roslaunch-zs-32529.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Unknown substitution command [eval depth_registered_processing and sw_registered_processing]. Valid commands are ['find', 'env', 'optenv', 'anon', 'arg']
The traceback for the exception was written to the log file

doronhi, 2019-01-07:
Hello,
Sorry, but I didn't understand what the issue is.

stwirth, 2019-01-07:
Looks like the `eval` command in the launch file is not understood. You need ROS kinetic for this, see http://wiki.ros.org/roslaunch/XML.

gxxzs, 2019-01-09:
> Looks like the `eval` command in the launch file is not understood. You need ROS kinetic for this, see http://wiki.ros.org/roslaunch/XML.

but my ros is indigo in ubuntu 14.04. what should i do to make it work.

  
    <include if="$(arg rgb_processing)"
             file="$(find rgbd_launch)/launch/includes/rgb.launch.xml">
      <arg name="manager"                       value="$(arg manager)" />
      <arg name="respawn"                       value="$(arg respawn)" />
      <arg name="rgb"                           value="$(arg rgb)" />
      <arg name="debayer_processing"            value="$(arg debayer_processing)" />
    </include>

    <group if="$(eval depth_registered_processing and sw_registered_processing)">
      <node pkg="nodelet" type="nodelet" name="register_depth"
            args="load depth_image_proc/register $(arg manager) $(arg bond)" respawn="$(arg respawn)">
        <remap from="rgb/camera_info"             to="$(arg rgb)/camera_info" />
        <remap from="depth/camera_info"           to="$(arg depth)/camera_info" />
        <remap from="depth/image_rect"            to="$(arg depth)/image_rect_raw" />
        <remap from="depth_registered/image_rect" to="$(arg depth_registered)/sw_registered/image_rect_raw" />
      </node>

      
      <node pkg="nodelet" type="nodelet" name="points_xyzrgb_sw_registered"
            args="load depth_image_proc/point_cloud_xyzrgb $(arg manager) $(arg bond)" respawn="$(arg respawn)">
        <remap from="rgb/image_rect_color"        to="$(arg rgb)/image_rect_color" />
        <remap from="rgb/camera_info"             to="$(arg rgb)/camera_info" />
        <remap from="depth_registered/image_rect" to="$(arg depth_registered_filtered)/sw_registered/image_rect_raw" />
        <remap from="depth_registered/points"     to="$(arg depth_registered)/points" />
      </node>
    </group>

    <group if="$(eval depth_registered_processing and hw_registered_processing)">
      
      <node pkg="nodelet" type="nodelet" name="points_xyzrgb_hw_registered"
            args="load depth_image_proc/point_cloud_xyzrgb $(arg manager) $(arg bond)" respawn="$(arg respawn)">
        <remap from="rgb/image_rect_color"        to="$(arg rgb)/image_rect_color" />
        <remap from="rgb/camera_info"             to="$(arg rgb)/camera_info" />
        <remap from="depth_registered/image_rect" to="$(arg depth_registered)/image_raw" />
        <remap from="depth_registered/points"     to="$(arg depth_registered_pub)/points" />
      </node>
    </group>

manhha1402, 2019-01-13:
Hi, 
I had the same issue when launching rs_rgbd.launch. I want to use high-resolution RGB stream (1920x1080), but it seems the Realsense ROS package cannot align and up-sample depth stream 640x480 to 1920x1080 like it should do.
[image]
[image]
Does someone have the same problem?
Thank you

gxxzs, 2019-01-14:
deled the if="$(eval depth_registered_processing and sw_registered_processing)" it worked.

closed 2019-01-14 by gxxzs

doronhi, 2019-01-14:
@manhha1402, Your problem is a bug.
Fixed in PR#572 and hopefully will get in the upcoming release.