realsenseai/realsense-ros #562: Can't connect the D435I camera

AdamPengG, 2019-01-07:
I have install the realsense sdk librealsense-2.17.1 and ROS wrapper realsense-2.1.3.And I follow the https://github.com/intel-ros/realsense/wiki/SLAM-with-D435i and install all the components.
Finally I run the roslaunch realsense2_camera opensource_tracking.launch.But the terminal shows I did't connect the the D435i.

[ERROR] [1546838519.355105483]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists. [ERROR] [1546838519.355165653]: The error before refreshing the cache was: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists. [FATAL] [1546838519.355291096]: Failed to load nodelet '/camera/realsense2_camera of type realsense2_camera/RealSenseNodeFactory to manager `realsense2_camera_manager
[camera/realsense2_camera-3] process has died [pid 10090, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/peng/.ros/log/27cdb684-123c-11e9-8909-107b447e20dc/camera-realsense2_camera-3.log].
log file: /home/peng/.ros/log/27cdb684-123c-11e9-8909-107b447e20dc/camera-realsense2_camera-3*.log
[ WARN] [1546838525.358489472]: /rtabmap/rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set.
/rtabmap/rgbd_odometry subscribed to (approx sync):
/camera/color/image_raw,
/camera/aligned_depth_to_color/image_raw,
/camera/color/camera_info

'
`What should i do?
[image]
[image]

AdamPengG, 2019-01-07:
Yes I can run roslaunch realsense2_camera rs_camera.launch.
And the camera gets connected to USB3.But when I run lsusb.The D435i isn't on the usb list.
I wanna get the left and right camera raw picture topics and IMU data topics.Which launch can show those data.
[image]
[image]

AdamPengG, 2019-01-07:
is anybody here?

doronhi, 2019-01-07:
There is 1 RGB camera and 2 IR cameras.
The topics are /camera/color/image_raw, /camera/infra1/image_rect_raw, /camera/infra2/image_rect_raw.

closed 2019-01-07 by doronhi