realsenseai/realsense-ros #564: Transformation Matrix between Robot-Coordinatesystem and Camera

tragolf, 2019-01-07:
Hi,
I have the D435 attached to the endeffector of a robot. To grasp a object, i need to convert from the camera-coordinate system (in which the object is detected) to the robot-coordinate system. Therefore I need the relative transformation between the camera and the robot. Is there a tool or script I can use?
At the moment I designed the setup in CAD and measured the Translation there (assuming no rotation and only 90° angles). But this is not very accurate.
Any hints? Thanks

doronhi, 2019-01-27:
The urdf file for D435 was updated recently to contain the transformation between the bottom screw and the image base coordinate system (camera_link). Is it helpful?

tragolf, 2019-01-27:
I found a calibration package, which does what I need. (hand eye calibration)
https://github.com/jhu-lcsr/aruco_hand_eye

closed 2019-01-27 by tragolf