realsenseai/realsense-ros #569: catkin_make errors

ustyui, 2019-01-11:
I'm using the latest librealsense SDK. What cause this error?
```
In file included from /home/nvidia/catkin_ws/src/realsense-2.1.3/realsense2_camera/src/../include/base_realsense_node.h:7:0,
                 from /home/nvidia/catkin_ws/src/realsense-2.1.3/realsense2_camera/src/base_realsense_node.cpp:1:
/home/nvidia/catkin_ws/src/realsense-2.1.3/ddynamic_reconfigure/include/ddynamic_reconfigure/ddynamic_reconfigure.h:213:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
           #pragma clang diagnostic push // CLion suppressor
 
/home/nvidia/catkin_ws/src/realsense-2.1.3/ddynamic_reconfigure/include/ddynamic_reconfigure/ddynamic_reconfigure.h:214:0: warning: ignoring #pragma ide diagnostic [-Wunknown-pragmas]
           #pragma ide diagnostic ignored "OCUnusedGlobalDeclarationInspection"
 
/home/nvidia/catkin_ws/src/realsense-2.1.3/ddynamic_reconfigure/include/ddynamic_reconfigure/ddynamic_reconfigure.h:220:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
           #pragma clang diagnostic pop
 
In file included from /home/nvidia/catkin_ws/src/realsense-2.1.3/ddynamic_reconfigure/include/ddynamic_reconfigure/param/dd_all_params.h:11:0,
                 from /home/nvidia/catkin_ws/src/realsense-2.1.3/realsense2_camera/src/../include/base_realsense_node.h:8,
                 from /home/nvidia/catkin_ws/src/realsense-2.1.3/realsense2_camera/src/base_realsense_node.cpp:1:
/home/nvidia/catkin_ws/src/realsense-2.1.3/ddynamic_reconfigure/include/ddynamic_reconfigure/param/dd_enum_param.h:52:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
         #pragma clang diagnostic push
 
/home/nvidia/catkin_ws/src/realsense-2.1.3/ddynamic_reconfigure/include/ddynamic_reconfigure/param/dd_enum_param.h:53:0: warning: ignoring #pragma ide diagnostic [-Wunknown-pragmas]
         #pragma ide diagnostic ignored "OCUnusedGlobalDeclarationInspection"
 
/home/nvidia/catkin_ws/src/realsense-2.1.3/ddynamic_reconfigure/include/ddynamic_reconfigure/param/dd_enum_param.h:81:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
         #pragma clang diagnostic pop
 
/home/nvidia/catkin_ws/src/realsense-2.1.3/realsense2_camera/src/base_realsense_node.cpp:800:0: warning: ignoring #pragma omp parallel [-Wunknown-pragmas]
     #pragma omp parallel for schedule(dynamic) //Using OpenMP to try to parallelise the loop
 
In file included from /usr/include/c++/7/algorithm:62:0,
                 from /usr/include/boost/smart_ptr/shared_ptr.hpp:39,
                 from /usr/include/boost/shared_ptr.hpp:17,
                 from /opt/ros/melodic/include/class_loader/class_loader.hpp:36,
                 from /opt/ros/melodic/include/pluginlib/./class_list_macros.hpp:40,
                 from /opt/ros/melodic/include/pluginlib/class_list_macros.h:35,
                 from /home/nvidia/catkin_ws/src/realsense-2.1.3/realsense2_camera/src/../include/../include/realsense_node_factory.h:6,
                 from /home/nvidia/catkin_ws/src/realsense-2.1.3/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/nvidia/catkin_ws/src/realsense-2.1.3/realsense2_camera/src/base_realsense_node.cpp:1:
/usr/include/c++/7/bits/stl_algo.h: In instantiation of ‘_Iterator std::__find_if(_Iterator, _Iterator, _Predicate) [with _Iterator = rs2::frameset::iterator; _Predicate = __gnu_cxx::__ops::_Iter_pred<realsense2_camera::BaseRealSenseNode::publishPointCloud(rs2::points, const ros::Time&, const rs2::frameset&)::<lambda(rs2::frame)> >]’:
/usr/include/c++/7/bits/stl_algo.h:3932:28:   required from ‘_IIter std::find_if(_IIter, _IIter, _Predicate) [with _IIter = rs2::frameset::iterator; _Predicate = realsense2_camera::BaseRealSenseNode::publishPointCloud(rs2::points, const ros::Time&, const rs2::frameset&)::<lambda(rs2::frame)>]’
/home/nvidia/catkin_ws/src/realsense-2.1.3/realsense2_camera/src/base_realsense_node.cpp:1718:125:   required from here
/usr/include/c++/7/bits/stl_algo.h:162:34: error: no matching function for call to ‘__iterator_category(rs2::frameset::iterator&)’
          std::__iterator_category(__first));
          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
In file included from /usr/include/c++/7/bits/stl_algobase.h:65:0,
                 from /usr/include/c++/7/memory:62,
                 from /usr/include/boost/config/no_tr1/memory.hpp:21,
                 from /usr/include/boost/get_pointer.hpp:14,
                 from /usr/include/boost/bind/mem_fn.hpp:25,
                 from /usr/include/boost/mem_fn.hpp:22,
                 from /usr/include/boost/bind/bind.hpp:26,
                 from /usr/include/boost/bind.hpp:22,
                 from /opt/ros/melodic/include/class_loader/class_loader.hpp:35,
                 from /opt/ros/melodic/include/pluginlib/./class_list_macros.hpp:40,
                 from /opt/ros/melodic/include/pluginlib/class_list_macros.h:35,
                 from /home/nvidia/catkin_ws/src/realsense-2.1.3/realsense2_camera/src/../include/../include/realsense_node_factory.h:6,
                 from /home/nvidia/catkin_ws/src/realsense-2.1.3/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/nvidia/catkin_ws/src/realsense-2.1.3/realsense2_camera/src/base_realsense_node.cpp:1:
/usr/include/c++/7/bits/stl_iterator_base_types.h:205:5: note: candidate: template<class _Iter> constexpr typename std::iterator_traits<_Iterator>::iterator_category std::__iterator_category(const _Iter&)
     __iterator_category(const _Iter&)
     ^~~~~~~~~~~~~~~~~~~
/usr/include/c++/7/bits/stl_iterator_base_types.h:205:5: note:   template argument deduction/substitution failed:
/usr/include/c++/7/bits/stl_iterator_base_types.h: In substitution of ‘template<class _Iter> constexpr typename std::iterator_traits<_Iterator>::iterator_category std::__iterator_category(const _Iter&) [with _Iter = rs2::frameset::iterator]’:
/usr/include/c++/7/bits/stl_algo.h:162:34:   required from ‘_Iterator std::__find_if(_Iterator, _Iterator, _Predicate) [with _Iterator = rs2::frameset::iterator; _Predicate = __gnu_cxx::__ops::_Iter_pred<realsense2_camera::BaseRealSenseNode::publishPointCloud(rs2::points, const ros::Time&, const rs2::frameset&)::<lambda(rs2::frame)> >]’
/usr/include/c++/7/bits/stl_algo.h:3932:28:   required from ‘_IIter std::find_if(_IIter, _IIter, _Predicate) [with _IIter = rs2::frameset::iterator; _Predicate = realsense2_camera::BaseRealSenseNode::publishPointCloud(rs2::points, const ros::Time&, const rs2::frameset&)::<lambda(rs2::frame)>]’
/home/nvidia/catkin_ws/src/realsense-2.1.3/realsense2_camera/src/base_realsense_node.cpp:1718:125:   required from here
/usr/include/c++/7/bits/stl_iterator_base_types.h:205:5: error: no type named ‘iterator_category’ in ‘struct std::iterator_traits<rs2::frameset::iterator>’
realsense-2.1.3/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:86: recipe for target 'realsense-2.1.3/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o' failed
make[2]: *** [realsense-2.1.3/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
CMakeFiles/Makefile2:1982: recipe for target 'realsense-2.1.3/realsense2_camera/CMakeFiles/realsense2_camera.dir/all' failed
make[1]: *** [realsense-2.1.3/realsense2_camera/CMakeFiles/realsense2_camera.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```

doronhi, 2019-01-11:
You write that you use the latest version of librealsense sdk yet the error message seems just like compiling with an older version.
Could it be that there are a couple of versions of the file librealsense.so on your machine?

ustyui, 2019-01-12:
sorry I checked that I'm using librealsense 2.17.0.
How can I fix the error?