realsenseai/realsense-ros #573: what's the difference brtween depth raw and aligned depth?

jinfagang, 2019-01-15:
Hi, to start realsense camera there some launch files, just want to what is the difference between the demo_depath.launch and the aligned_depth.launch?

Should I just using aligned depth rather the raw? does they have big differences.

doronhi, 2019-01-16:
You can use rs_camera.launch as it the most updated.
You can alter any of the options in your command line.
Aligned depth is the depth image warped to match the color image or the other ir image.
To use it you need to set align_depth:=true, whether on the command line or in the launch file.
Warping takes some CPU, depends on resolution and your machine.
Whether you want to use aligned or raw depth image is really according to your needs.

jinfagang, 2019-01-17:
@doronhi  In my situation, I need the depth d match exactly the pixel because  I need do semantic segmentation on image, and need that depth to get a 3D point so that the road and grass can be seen in 3D space. In this side, should aligned depth be better then raw? To see from generate point cloud, it can not easy tell the difference cause it always noisy

doronhi, 2019-01-27:
It feels to me that indeed, aligned depth, matches your needs more as you can do the segmentation based on both color and depth.

Shashika007, 2019-02-14:
@doronhi I noticed that with new librealsense-2.18.1 and realsense-2.1.4 ros, camera/color/image_raw is aligned with camera/depth/image_raw_rect but not with camera/alignedalinged_depth_to_color/image_raw ? Do you have any idea about why ?

[image]

doronhi, 2019-02-14:
I assume the times in your graph show the published time and not the timestamp in the messages header.
realsense2_camera is built on top of librealsense. librealsense's syncer synchronizes the images from the depth sensor and the color sensor, therefor they are published on the same time. 
aligned_depth_to_color (or to any other sensor) requires some extra work at is published a bit later. It has the same time stamp in its header as the original images it is based on.
You can check out the code in base_realsense_node.cpp, function publishAlignedDepthToOthers.

Shashika007, 2019-02-15:
@doronhi Thank you.

closed 2019-02-18 by doronhi