realsenseai/realsense-ros #576: executing rm_camera.launch resulting error

pac3man, 2019-01-17:
Hi,
I am a student working with RealSence camera. I want to connect the camera through ROS. For that I have followed the instructions given in [here](https://github.com/intel-ros/realsense) and successfully installed all the required packages. I can see infrared output from the camera using realsense-viewer command. After that when I try to connect it using ROS by,

$ roslaunch realsense2_camera rs_camera.launch

I get error similar to the post mentioned in [here](https://github.com/intel-ros/realsense/issues/301) . I am copying the terminal prompts while I execute the roslaunch file at the end of this post. I am sure that I am connecting the camera to a USB 3 port and did not make any changes to rs_camera.launch . I have opened a question in ROS-Answers regarding this issue which you can find https://answers.ros.org/question/312744/failed-to-connect-realsense-camera-using-realsense2_camera-package/ . If you can help me solving this problem it would be great.

Thank you in advance.

`
$ roslaunch realsense2_camera rs_camera.launch

... logging to /home/nobug-ros/.ros/log/741bfb4a-19ee-11e9-a2a2-642737b8e683/roslaunch-nobug-18599.log

Checking log directory for disk usage. This may take awhile.

Press Ctrl-C to interrupt

Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nobug:39973/

summary

PARAMETERS * /camera/realsense2_camera/accel_fps: 250 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti... * /camera/realsense2_camera/align_depth: False * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de... * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de... * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de... * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de... * /camera/realsense2_camera/base_frame_id: camera_link * /camera/realsense2_camera/clip_distance: -2.0 * /camera/realsense2_camera/color_fps: 30 * /camera/realsense2_camera/color_frame_id: camera_color_frame * /camera/realsense2_camera/color_height: 480 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti... * /camera/realsense2_camera/color_width: 640 * /camera/realsense2_camera/depth_fps: 30 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame * /camera/realsense2_camera/depth_height: 480 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti... * /camera/realsense2_camera/depth_width: 640 * /camera/realsense2_camera/enable_color: True * /camera/realsense2_camera/enable_depth: True * /camera/realsense2_camera/enable_fisheye: True * /camera/realsense2_camera/enable_imu: True * /camera/realsense2_camera/enable_infra1: True * /camera/realsense2_camera/enable_infra2: True * /camera/realsense2_camera/enable_pointcloud: False * /camera/realsense2_camera/enable_sync: False * /camera/realsense2_camera/filters: * /camera/realsense2_camera/fisheye_fps: 30 * /camera/realsense2_camera/fisheye_height: 480 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op... * /camera/realsense2_camera/fisheye_width: 640 * /camera/realsense2_camera/gyro_fps: 400 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic... * /camera/realsense2_camera/infra1_fps: 30 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame * /camera/realsense2_camera/infra1_height: 480 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt... * /camera/realsense2_camera/infra1_width: 640 * /camera/realsense2_camera/infra2_fps: 30 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame * /camera/realsense2_camera/infra2_height: 480 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt... * /camera/realsense2_camera/infra2_width: 640 * /camera/realsense2_camera/initial_reset: False * /camera/realsense2_camera/json_file_path: * /camera/realsense2_camera/linear_accel_cov: 0.01 * /camera/realsense2_camera/pointcloud_texture_index: 0 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR * /camera/realsense2_camera/rosbag_filename: * /camera/realsense2_camera/serial_no: * /camera/realsense2_camera/unite_imu_method: * /rosdistro: kinetic * /rosversion: 1.12.14

NODES /camera/ realsense2_camera (nodelet/nodelet) realsense2_camera_manager (nodelet/nodelet)

auto-starting new master process[master]: started with pid [18609] ROS_MASTER_URI=http://localhost:11311

setting /run_id to 741bfb4a-19ee-11e9-a2a2-642737b8e683 process[rosout-1]: started with pid [18622] started core service [/rosout] process[camera/realsense2_camera_manager-2]: started with pid [18625] process[camera/realsense2_camera-3]: started with pid [18626] [camera/realsense2_camera_manager-2] process has died [pid 18625, exit code 1, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/nobug-ros/.ros/log/741bfb4a-19ee-11e9-a2a2-642737b8e683/camera-realsense2_camera_manager-2.log]. log file: /home/nobug-ros/.ros/log/741bfb4a-19ee-11e9-a2a2-642737b8e683/camera-realsense2_camera_manager-2.log [camera/realsense2_camera-3] process has died [pid 18626, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/nobug-ros/.ros/log/741bfb4a-19ee-11e9-a2a2-642737b8e683/camera-realsense2_camera-3.log]. log file: /home/nobug-ros/.ros/log/741bfb4a-19ee-11e9-a2a2-642737b8e683/camera-realsense2_camera-3.log
`

RealSenseCustomerSupport, 2019-04-19:
----------------------------------------------

Hi @pac3man,

Which camera do you use? Please try the latest ros wrapper with the correspondent librealsense version and see how it works.

RealSenseCustomerSupport, 2019-04-19:
----------------------------------------------

By the way, I also notice that your realsense camera is not detected based on below error message. Please run "rs-enumerate-devices" and "lsusb -v".
[ERROR] [1547684756.850043496]: No RealSense devices were found! Terminating RealSense Node...

RealSenseCustomerSupport, 2019-05-20:
----------------------------------------------

Hi @pac3man,

Do you still have any trouble for this issue? If not, I will close this.

closed 2019-08-06 by RealSenseCustomerSupport