realsenseai/realsense-ros #577: D435i IMU msgs have big gap frequently?

Shayens, 2019-01-17:
Just shown as the figure, it appears gaps frequently. Also, the frequency of IMU msgs are not stable, however in zr300, it's stable. 
Is this the bug hasn't fixed yet? Or I'm not using it correctly.
By the way, how to enable "hold_back_imu_for_frames", is it right as this `roslaunch realsense2_camera rs_camera.launch hold_back_imu_for_frames:=on`but I didn't see any difference...
D435i bag
[image]
[image]
ZR300 bag
[image]

Shashika007, 2019-02-13:
@Shayens  did you find a solution for this issue ? i noticed this too

Shashika007, 2019-02-14:
And another thing, i noticed that with new librealsense-2.18.1 and realsense-2.1.4 ros, camera/color/image_raw is aligned with camera/depth/image_raw_rect but not with camera/alignedalinged_depth_to_color/image_raw ? Do you have any idea about why ? 

[image]

doronhi, 2019-02-24:
@Shayens 
I used librealsense v2.18.1, realsense2_camera v2.2.0 and firmware: 05.10.13.00.
I used python to check the time stamps in the messages' header:
```
bag = rosbag.Bag(bag_filename)
tg = np.array([msg.header.stamp.to_sec() for topic, msg, t in bag.read_messages(topics=topic_str)])
dtg = tg[1:] - tg[:-1]
max_gap = print dtg.max() / np.median(dtg)
```
np.median(dtg) == 0.0025 (fps = 250)
dtg.max() == 0.005
I get max_gap = 2, which is not what we see in the rqt_bag graph.

When I take the rosbag times, however:
```
tg = np.array([t.to_sec() for topic, msg, t in bag.read_messages(topics='/camera/gyro/sample')])
```
I get max_gap = 8.98. In this case dtg.max() == 0.022.

In other words, what I see in the graph has to do with loads on the ROS environment and the bag record application. The timestamps in the messages themselves have no gaps.

doronhi, 2019-02-24:
@Shashika007 - duplicate question in issue#573

doronhi, 2019-02-24:
@Shayens 
regarding hold_back_imu_for_frames, it should be hold_back_imu_for_frames:=**true**
However, the argument is missing from rs_camera.launch and from nodelet.launch.xml. You need to add it there (copy the example of the argument `align_depth`)
In addition, this argument is only meaningful if `unite_imu_method` is also set to either **linear_interpolation** or **copy**.

Shayens, 2019-03-01:
Thanks for your test. And sorry for didn't reply to you @Shashika007. I also plot by package multiplot and verify, so it seems rqt_bag is using receipt time.
[image]

closed 2019-03-01 by Shayens