realsenseai/realsense-ros #578: Can I get the D435i's raw gray pictures without the IR

AdamPengG, 2019-01-17:
I only want the raw gray pics but the rostopic infra1 infra2 have many points on the pics.
How Can  I only get the gray pics?

doronhi, 2019-01-17:
Set the emitter off.

AdamPengG, 2019-01-18:
Can I set it in the ROS wrapper and where to set this value

doronhi, 2019-01-18:
1. Try rqt_reconfigure
2. These values look like integers and should be floating points. How did you extract them?
3. Not sure what you are going for. Getting radial velocity from a pure acceleration message? I feel like I miss something here.

AdamPengG, 2019-01-21:
1.Yes,it can turn off in rqt_reconfigure.But I clicked the "rgb_camera",which in the rqt_reconfigure,the rqt break.The terminal shows:
[image]
2.The extract the float message.
[image]
3.The full launch opensource-tracking running in my "CPU @ 1.80GHz × 8" laptop which is a bit stuck.
The light launch rs_camera.launch haven't the IMU message,with angular_velocity and linear_acceleration. 
I only want record the fluently IMU and two cameras message in my rosbag,which launch can I use?

doronhi, 2019-01-22:
Hi @AdamPengG 
1. It looks like a bug but I couldn't reproduce it. Could you add a screenshot of the rqt_reconfigure screen just before the error occurs, with a mark on the button you press for error to happen?
3. The error `AttributeError: '_realsense2_camera__IMUInfo' object has no attribute 'angular_velocity'` means that you are not reading the right message. It seems you are listening on topic `/camera/accel/imu_info` instead of `/camera/accel/sample`.
2. The big values don't seem reasonable. Maybe it has something to do with the transition to the graphic table? Can you try to build a minimal application that listens on a topic and print the values? Can you show the big numbers error with that app?

AdamPengG, 2019-01-23:
1.The error button is the RGB button
[image]
2.The topic,/camera/accel/sample, can extract the linear messages.Is there a topic,in rs_camera.launch, save the linear and angular messages.And the topic's ,/camera/accel/sample, timestamp seemed isn't  the Greenwich time.
[image]
3.I put out the big values with ros tool,rostopic echo,and the value are also big.
[image]

doronhi, 2019-01-24:
Not sure I understand what you mean in section 2.
1. When I push the RGB_camera everything seems fine. What version of realsense2_camera you use? Firmware? librealsense version?
3. The values as printed by rostopic echo seem normal and not as big as the ones in the table. Again, I don't feel I understand the problem. Sorry.

AdamPengG, 2019-01-24:
[image]
Emm,the linear_acceleration:
x:-22700883968.0
y:7567073280.0
z:7567114240.0
The x,y and z value is too big.The angular_velocity_covariance's values are as common.
Is the unit of linear acceleration velocity centimeter?

2.The topic /camera/accel/sample 's  timestamps are GreenWich time?

AdamPengG, 2019-01-29:
Did anyone face this problem?

closed 2019-01-29 by doronhi