realsenseai/realsense-ros #580: Can a windows10 ROS 2 RVIZ2 connect to ROS RealSense TOPIC?

tmichals, 2019-01-21:
Is there away to get RVIZ2 on my windows10 computer to connect to a Intell RealSense camera running on a nvidia TX2 ubuntu ROS(kinectic) computer?   I can get RVIZ to connect as long as I am running RVIZ on a ROS ubuntu computer, (however I need to use a  windows10 computer for RVIZ). Since only RVIZ2  will run on windows10 that is my only choice. 

I have tried getting RVIZ to work by installing ubuntu natively on windows10...but I was not able to get the video to stream into the Xming terminal. I could see the ROS Topics in Xming...just no video. 
Here is an error I would get on TX2 (lots of unknowns)...
________________
Could not initialize OpenGL for RasterGLSurface, reverting to RasterSurface.
QXcbConnection: XCB error: 170 (Unknown), sequence: 163, resource id: 90, major code: 146 (Unknown), minor code: 20
Segmentation fault (core dumped)
____________________________________________
I found no real help with this problem.  

So now I am trying to get RVIZ2 working in windows.

I am new to ROS2, but I was able to get the ROS2 Windows-Install-Binary working on Windows10.   I installed OpenSplice for Crystal.

I can start rviz2 from command prompt, and add a camera...but as I would expect the Topic has a status Error- Error Subscribing: Empty Topic name. There are no RealSense Topics to attach to.

I know from running rviz on ROS ubuntu computer there are some other prerequisites like.. exporting my ROS_IP and exporting to the ROS_MASTER_URI=http://xx.xx.xx.xx:11311 that have to be done before I can launch rviz. So how does ROS2 do this...do I need to do something in ROS2 before I start RVIZ2? Can ROS2 RVIZ2 get topics from ROS master and RealSense Node?!

Perhaps I have to install ROS 2 RealSense on my TX2?

atyshka, 2019-01-22:
@tmichals This package only supports ROS 1, and RViz 2 is incompatible with ROS 1. If you really want to visualize on Windows I suggest using a virtual machine and running ROS 1 in it.

tmichals, 2019-01-22:
Here are the links that I found running ROS1 on windows10:
                                   https://ms-iot.github.io/ROSOnWindows/
                                   https://ms-iot.github.io/ROSOnWindows/GettingStarted/Setup.html
_____________________
Here is a link explaining the wsl installs that I did on windows10...                               https://janbernloehr.de/2017/06/10/ros-windows

but there was no RealSense video displayed trying to use rvis. 
_________________________

    Do you know of a different camera interface for realsense other than using rviz?  That would work just to display realsense video?  Does not have to be ROS..just needs to be able to attach to camera on TX2 from windows10.

tmichals, 2019-01-23:
Thank you.   At least you understand the problem.  I hope if you have some pull that someone on would be able to bridge the ubuntu/windows ROS gap for RealSense.     I need RealSense to run on the nvidia TX2 ROS ubuntu and connect to a ROS RVIZ running on windows.   RVIZ will fire up in the windows10 wsl version which is ubuntu...but RVIZ has to run in a Xming type terminal which looses the video stream when RVIZ comes up there is a mapping problem...nothing shows up.   could be in the OpenGL to windows area.  The Topic selections in rviz are available ; that is specifically where someone would need to examine.   There are some settings for the Xming server that I have not tried perhaps that will work.    Second...when it comes to ROS2  the best understanding I have is ROS2 and ROS are not compatible...which is a big problem, and that is not a RealSense issue but perhaps someone is working to address that through a bridge connection.   Is this all correct?

tmichals, 2019-01-23:
Here is a video link for you to see my nvidia TX2 RealSense issue.. https://www.youtube.com/watch?v=Bd6TBd4Z8vk&t=15s

doronhi, 2019-01-24:
I'll quote @atyshka : "Sorry, but this really has nothing to do with the realsense ros". This is an issue with running RVIZ with ROS1 on windows.

closed 2019-01-24 by doronhi