realsenseai/realsense-ros #583: Decimation filter does not adapt CameraInfo

stwirth, 2019-01-22:
Thanks for adding the support for the decimation filter!
I have one issue using it:
When running
```
roslaunch realsense2_camera rs_camera.launch depth_width:=1280 depth_height:=720 filters:=decimation
```
I see the decimation filter parameters in dynamic reconfigure and can manipulate them. When I change the filter magnitude, I see that the image size changes accordingly:
```
$ rosrun dynamic_reconfigure dynparam set camera/decimation 'Filter Magnitude' 1
stwirth@x:~$ rostopic echo -n 1 /camera/depth/image_rect_raw --noarr
header: 
  seq: 21027
  stamp: 
    secs: 1548150346
    nsecs: 234946562
  frame_id: "camera_depth_optical_frame"
height: 720
width: 1280
encoding: "16UC1"
is_bigendian: 0
step: 2560
data: "<array type: uint8, length: 1843200>"
---
$ rosrun dynamic_reconfigure dynparam set camera/decimation 'Filter Magnitude' 4
$ rostopic echo -n 1 /camera/depth/image_rect_raw --noarr
header: 
  seq: 21271
  stamp: 
    secs: 1548150354
    nsecs: 365142743
  frame_id: "camera_depth_optical_frame"
height: 180
width: 320
encoding: "16UC1"
is_bigendian: 0
step: 640
data: "<array type: uint8, length: 115200>"
---
```
However, the CameraInfo message content does not change:
```
$ rosrun dynamic_reconfigure dynparam set camera/decimation 'Filter Magnitude' 1
$ rostopic echo -n 1 /camera/depth/camera_info
header: 
  seq: 26057
  stamp: 
    secs: 1548150513
    nsecs: 842895075
  frame_id: "camera_depth_optical_frame"
height: 720
width: 1280
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [933.902587890625, 0.0, 649.7789306640625, 0.0, 933.902587890625, 359.9447326660156, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [933.902587890625, 0.0, 649.7789306640625, 0.0, 0.0, 933.902587890625, 359.9447326660156, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
---
$ rosrun dynamic_reconfigure dynparam set camera/decimation 'Filter Magnitude' 4
$ rostopic echo -n 1 /camera/depth/camera_info
header: 
  seq: 26540
  stamp: 
    secs: 1548150529
    nsecs: 938007731
  frame_id: "camera_depth_optical_frame"
height: 720
width: 1280
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [933.902587890625, 0.0, 649.7789306640625, 0.0, 933.902587890625, 359.9447326660156, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [933.902587890625, 0.0, 649.7789306640625, 0.0, 0.0, 933.902587890625, 359.9447326660156, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
---
```
I would expect the `height` and `width` to change, matching the image dimensions and also the K and P matrices should change.

The [librealsense documentation](https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#decimation-filter) says:
> After the resulted frame is produced, the frame intrinsic parameters are recalculated to compensate for the resolution changes.

It seems that those recalculated parameters are not propagated to the camera info messages.

I'm using librealsense v2.17.1, RealSense ROS v.2.1.3, Device FW version 05.11.01.00.

doronhi, 2019-01-22:
Thanks for pointing that out.

closed 2019-01-27 by doronhi

zinan93, 2019-06-24:
Hi! I tried changing the decimation parameter using dynamic reconfigure, however got the following error:

> error updating parameters: don't know parameter: Filter Magnitude

I'm using realsense-ros 2.2.3 and d435 camera.

doronhi, 2019-06-25:
I checked with latest version 2.2.6 and can't reproduce.
The implementation of the ddynamic_reconfigure library was changed since 2.2.3. Could be related.
Can you upgrade and check again?

zinan93, 2019-06-26:
I tried updated to 2.2.6 realsense-ros, but it asks for SDK version 2.22.0, however I use 2.21.0. I would want to prevent from reinstalling the SDK if possible. I used apt to install the ddynamic_reconfigure library. Is there a previous version than 2.2.3 for realsense-ros to get it working?

doronhi, 2019-06-27:
The installed ddynamic_reconfigre library was changed, in its interface, relative to the copy realsense2_camera used to hold.
There is no realsense2_camera version that uses both the installed ddynamic_reconfigure library and a librealsense2 SDK with version < 2.22.0.
Is there a reason for not upgrading librealsense2 SDK?

zinan93, 2019-06-27:
ok it works now, the syntax is "filter_magnitude" rather than "Filter Magnitude", also just found out about using rqt_reconfigure.. thanks