realsenseai/realsense-ros #584: [Suggestion] Gyro and Accelerometer ROS Data types

marcelinomalmeidan, 2019-01-22:
Hi,

I have just receiver the D435i, and installed all the necessary drivers on my Jetson TX2.
I am currently on the development branch, and have been looking at the IMU data.
I can see that gyro and accelerometer info can be obtained from the topics:
```
/camera/accel/sample
/camera/gyro/sample
```

I guess I understand why they are not both in the same topic, since they are not synchronized. However, I find it weird that these topics are of the type ```sensor_msgs/Imu```,  since the message is no fully populated. Wouldn't it make more sense if the topics were of the type ```geometry_msgs/Vector3Stamped```?

This doesn't do much harm to anyone the way it is, but I just feel there is no need to populate a much bigger message type when only a few entries make sense. When it comes to use ros publishers/subscribers at 400Hz for WiFi communication, I think that the size of the message matters, specially if there are multiple robots with a realsense in each of them.

marcelinomalmeidan, 2019-01-22:
One more thing: the ```secs``` and ```nsecs``` fields are wrong in ```/camera/accel/sample``` and ```/camera/gyro/sample```. 
The field ```secs``` is displaying kilosseconds (it updates once every 1000 seconds), and ```nsecs``` is being updated with fractions of kilosseconds.

doronhi, 2019-01-23:
Your suggestion is interesting and needs to think about.
On the other thing, I can't see the problem you describe.
I run:
`roslaunch realsense2_camera rs_camera.launch`
and then:
`
rostopic echo -n 1 /camera/accel/sample --noarr | grep secs: && sleep 1 && rostopic echo -n 1 /camera/accel/sample | grep secs:
`
And I get:

secs: 1548249069
nsecs: 950823545
secs: 1548249071
nsecs: 713103533

secs seems to advance fine.

marcelinomalmeidan, 2019-01-30:
Sorry for taking so long to get back, but this is what I get. Whenever I run

```rostopic echo -n 1 /camera/accel/sample/header/stamp```

This is what I get:

```
secs: 0
nsecs: 826689482
---
```

RealSenseCustomerSupport, 2019-04-25:
----------------------------------------------

Hi marcelinomalmeidan,

Sorry take a while to come to this one.
I tried following tests off an Intel NUC, and have no issue (kernel 4.15).
I will try off a Nvidia TX2 later once I get it set up.

$ rostopic echo -n 1 /camera/accel/sample --noarr | grep secs: && sleep 1 && rostopic echo -n 1 /camera/accel/sample | grep secs:
    secs: 1556210363
    nsecs:  44781685
    secs: 1556210364
    nsecs: 619525671
$ rostopic echo -n 1 /camera/accel/sample/header/stamp

secs: 1556210529
nsecs: 329480648
---

On the other side, wonder if you have any update?

Thanks!

marcelinomalmeidan, 2019-04-29:
My kernel is 4.4.38-tegra (Ubuntu 16.04), which is different from yours (I presume you are using Ubuntu 18.04). 
I still have the same issue:
```
rostopic echo -n 3 /camera/accel/sample/header/stamp

secs: 0
nsecs: 312066555
---

secs: 0
nsecs: 312070370
---

secs: 0
nsecs: 312074423
```

Realsense would not install on the tx2 by default. I did have to do some kernel updates to install the realsense packages:
https://www.jetsonhacks.com/2018/07/10/librealsense-update-nvidia-jetson-tx-dev-kits/
https://www.jetsonhacks.com/2018/07/10/librealsense-update-nvidia-jetson-tx-dev-kits/

Would you have a different set of instructions for installing realsense on a TX2?

RealSenseCustomerSupport, 2019-04-30:
----------------------------------------------

Hi marcelinomalmeidan

Thaks for the update. There have been some discussion on Realsense installation off TX2.
And one of them could help: https://github.com/IntelRealSense/librealsense/issues/3543

Thanks!

RealSenseCustomerSupport, 2019-05-08:
----------------------------------------------

Hi marcelinomalmeidan

Any update on the TX2 installation?

Thanks!

marcelinomalmeidan, 2019-05-08:
Unfortunately, I don't. I don't want to risk my current (working) installation, specially since the guidelines for  installing on the TX2 are not entirely clear.

RealSenseCustomerSupport, 2019-05-10:
----------------------------------------------

HI marcelinomalmeidan,

Sorry for the inconvenience on installation off Nvidia TX2.
I will close this one for now. Please update back once you have a chance to try, and we will also update you on TX2 installation if any.

Thanks!

marcelinomalmeidan, 2019-05-10:
Thanks! It would be great to have an "official" TX2 Installation guide. Maybe this could be the subject of another issue?

marcelinomalmeidan, 2019-05-16:
I have just updated my SDK version to build 2.21.0 (I was using 2.17) before, and the issue is no longer there.

closed 2019-05-16 by marcelinomalmeidan