realsenseai/realsense-ros #589: E: Unable to locate package librealsense2-dkms

lz130dd, 2019-01-24:
When I follow the instructions in the file "travis.yml ", after I run the commend

$ sudo apt-get install librealsense2-dkms --allow-unauthenticated -y 

I got error like this,

$ sudo apt-get install librealsense2-dkms --allow-unauthenticated -y 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package librealsense2-dkms

I have tried to use " sudo apt-get update", but it doesn't work. Can anybody help me with this?

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

thanujadax, 2019-03-07:
Did you set the apt source as described in the installation guide for Ubuntu?

RealSenseCustomerSupport, 2019-04-22:
----------------------------------------------

Inactivity for 30+ days.
Close this issue. Please feel free to reopen if problem persists.

closed 2019-04-30 by RealSenseCustomerSupport

q8wwe, 2020-03-08:
i have same problem how to solve it ???

MarcoAndrade26, 2020-07-26:
Same problem here!

doronhi, 2020-07-26:
As @thanujadax commented, Did you set the apt source as described in the installation guide for Ubuntu?

MarcoAndrade26, 2020-07-26:
@doronhi Yes! Every command and step from installation guide