realsenseai/realsense-ros #590: Depth map becomes "darker" when lightling conditions change

MTDzi, 2019-01-25:
Hi,

I have the D435 camera, with librealsense v2.17.0, and the ROS wrapper in version v2.1.2, working on Jetson TX1, L4T r.28.2, Ubuntu 16.

I've mounted the camera on top of a radio-controlled car, and collected data at the office where I work, and I've experienced the following phenomenon:
https://youtu.be/-V7d-S5XUYg?t=13
i.e. the image becomes darker when moving from a darker into well-lit space. 

Maybe to make it a bit clearer without the need of running the youtube video, this is the depth map when the car is still in the darker space:
[image]
and this is what happens when I move into the well-lit space:
[image]

I have no experience working with cameras (in particular: stereo cameras) and very limited knowledge about optics.

**Question 1:** what is the source of this effect.

I don't care about the visuals, the image can become darker as long as I'm able to extract the correct distance of the objects from the camera. There's also [this issue](https://github.com/intel-ros/realsense/issues/436), and let's assume I'll do find a way of extracting the distance...

**Question 2:** will this "darkening" of the image impact the quality of the extracted distance?

I would very much appreciate any suggestions / help / leads / etc.

SteveMacenski, 2019-01-31:
Its essentially autoexposure for the IR images like you would have with your cell phone camera. If you're in direct lighting, change the exposure so the images don't get washed out. I would not expect this to change the results of your measurements in a tangible way.

skohlbr, 2019-02-02:
This looks a lot like the image visualization is just scaling the displayed gray value for depth images so the farthest point appears the most bright (i.e. white), while closer points are less bright. So as soon as points appear to be very far (either because they are, or because there is noise), most of the image appears darker due to everything close getting interpolated to a darker gray value.
Not sure what you used for visualization, but this is exactly the effect one sees when for instance selecting autocompute bounds in rviz.

MTDzi, 2019-02-02:
Hey,

Thanks for the hints!

@skohlbr This is indeed rviz. So I'll extract the distances later today and get back to you to report if this effect influences the distances.

MTDzi, 2019-02-02:
OK,

so on a different track (with more homogenous lighting conditions) I experienced the same effect.

But as @SteveMacenski and @skohlbr suggested, it seems to be an artifact of the normalization procedure. Indeed, this time I used `matplotlib.pyplot.imshow` (which, I know for sure, uses [colormap normalization](https://matplotlib.org/users/colormapnorms.html)) and I got these two results (for two almost consecutive frames):

[image]

and

[image]

But their corresponding histograms of depth values (pixel intensities) were:

[image]

and

[image]

and that's OK.

closed 2019-02-02 by MTDzi