realsenseai/realsense-ros #594: T265 Support Timeline?

atyshka, 2019-01-31:
Hi, this is just a general question as to when support for the T265 sensor will get built into this repo. ROS support was advertised on the product page and this is definitely something we're interested in, but a general timeline as to when these changes will be built in would greatly help!

SteveMacenski, 2019-01-31:
Especially for just tracking - with no loop closure / SLAM due to the fact it doesn't work in even moderately large spaces. Having something like this to fuse into my filters would be immensely helpful

atyshka, 2019-01-31:
@SteveMacenski You mean the T265 won't work in a larger space? Where did you hear that?

SteveMacenski, 2019-01-31:
@atyshka from the intel folks, they're branding it as SLAM in a box, but it can't maintain maps that are very large so in practice I just want it for tracking. This is not a depth camear.

atyshka, 2019-01-31:
@SteveMacenski I see what you're saying. While it does seem that it can't hold very large maps, they have promised the ability to save and load maps using librealsense. I believe that saving and loading maps could help with large areas. Without detail on how it works though, I guess we'll just have to wait for the release.

doronhi, 2019-02-04:
Hi, as a general time frame, I hope support for T265 will be available in a week.

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

immotri, 2019-04-16:
Does the T265 detect loop closures? If not, is there any way to achieve bundle adjustments after loop closures with this device? I can‘t find any more detailed information on the V-SLAM used by Intel. It appears to be more a V-odometry since Feature points are not recorded. Any info from Phillip Schmidt a co would be appreciated!

atyshka, 2019-04-16:
@immotri Yes, it does. I did some testing handheld with no wheel odometry, can’t speak for using it with wheel odometry. For small loops (25m or less) I got good results in a single loop. For larger loops I got better results taking two loops to refine the position a little more.

immotri, 2019-04-16:
I got good results running 3 loops in my flat. Not sure whether the loops adjusted after closure. I will take it into the garden tomorrow. Any info on how to integrate a wheel odometer would be great.

RealSenseCustomerSupport, 2019-04-18:
----------------------------------------------

Intel is continuing to improve upon our current offering and improve wheel odometry samples under the ROS repository for T265.  Continue to monitor the ROS release for updates.  Relocalization updates are coming in a future release.

closed 2019-05-13 by RealSenseCustomerSupport