realsenseai/realsense-ros #595: D435i No PointCloud2 and No Image in rviz

arthurlobo, 2019-02-01:
Hi:

My environment is:

RealSense ROS 2.1.4
Librealsense 2.18
D435i firmware 5.11.1
Platform Jetson TX2
OS Ubuntu 16.04

I am not getting a point cloud or image in rviz but realsense-viewer works fine.
Command in terminal 1: roslaunch realsense2_camera rs_rgbd.launch
Command in terminal 2: rviz

Does anyone have any suggestions?

Thanks
Art

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/unite_imu_method: none
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    color_rectify_color (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [5636]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to cccfe712-25d6-11e9-a4c0-00044b8cacff
process[rosout-1]: started with pid [5649]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [5666]
process[camera/realsense2_camera-3]: started with pid [5667]
process[camera/color_rectify_color-4]: started with pid [5668]
process[camera/points_xyzrgb_hw_registered-5]: started with pid [5669]
[ INFO] [1548994012.497521342]: Initializing nodelet with 6 worker threads.
[ INFO] [1548994012.571744371]: RealSense ROS v2.1.4
[ INFO] [1548994012.571825300]: Running with LibRealSense v2.18.0
 01/02 04:06:52,932 WARNING [547498426368] (types.cpp:57) hwmon command 0x4f failed. Error type: No data to return (-21).
[ INFO] [1548994012.978171440]: getParameters...
[ INFO] [1548994013.660937751]: setupDevice...
[ INFO] [1548994013.662004319]: JSON file is not provided
[ INFO] [1548994013.662411489]: ROS Node Namespace: camera
[ INFO] [1548994013.663289671]: Device Name: Intel RealSense D435I
[ INFO] [1548994013.663597257]: Device Serial No: 838212070478
[ INFO] [1548994013.663822283]: Device FW version: 05.11.01.00
255.255.255.255
[ INFO] [1548994013.664086125]: Device Product ID: 0x0B3A
[ INFO] [1548994013.664902098]: Enable PointCloud: Off
[ INFO] [1548994013.665148820]: Align Depth: On
[ INFO] [1548994013.665356757]: Sync Mode: On
[ INFO] [1548994013.665875961]: Device Sensors: 
[ INFO] [1548994013.666223803]: Stereo Module was found.
[ INFO] [1548994013.666434268]: RGB Camera was found.
[ INFO] [1548994013.666642334]: Motion Module was found.
[ INFO] [1548994013.667101761]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1548994013.668145160]: setupPublishers...
[ INFO] [1548994013.713467674]: Expected frequency for depth = 30.00000
[ INFO] [1548994013.754716369]: Expected frequency for infra1 = 30.00000
[ INFO] [1548994013.777837582]: Expected frequency for aligned_depth_to_infra1 = 30.00000
[ INFO] [1548994013.806267182]: Expected frequency for infra2 = 30.00000
[ INFO] [1548994013.836688219]: Expected frequency for aligned_depth_to_infra2 = 30.00000
[ INFO] [1548994013.870461952]: Expected frequency for color = 30.00000
[ INFO] [1548994013.904759304]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1548994014.017395553]: setupStreams...
 01/02 04:06:54,021 WARNING [547498426368] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ INFO] [1548994014.032656168]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1548994014.034611285]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1548994014.036477954]: infra2 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1548994014.054861726]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1548994014.085546222]: starting imu...
[ INFO] [1548994014.090687633]: gyro stream is enabled - fps: 400
[ INFO] [1548994014.091510934]: accel stream is enabled - fps: 250
[ INFO] [1548994014.092948288]: num_filters: 0
[ INFO] [1548994014.094070343]: RealSense Node Is Up!
[ INFO] [1548994014.094196424]: Setting Dynamic reconfig parameters.
[ INFO] [1548994014.256143247]: Done Setting Dynamic reconfig parameters.

kvigulis, 2019-02-06:
Having exactly the same console output with the same device/setup, except the firmware is: 05.10.13.00 and I'm testing on laptop. Also not getting anything in RViz, although I'm not sure if there is something that needs to be done from within RViz to access the output.

arthurlobo, 2019-02-06:
>not sure if there is something that needs to be done from within RViz to access the output

I followed the steps by icarpis in the following issue for my above post:
https://github.com/intel-ros/realsense/issues/302

arthurlobo, 2019-02-06:
I also did the following in rviz before the above:
Under "Global Options" select "camera_link" from "Fixed Frame" drop down menu

kvigulis, 2019-02-07:
Thanks for the instructions Arthur. I added the PointCloud2 topic, but it has an error in Transform, when selecting the "camera_link" in "Global Options".

[image]

Selecting "camera_color_optical_frame", gets rid of the Transform error and it shows that it is receiving messages, but still there's no visualisation of pointcloud/anything..

[image]

kvigulis, 2019-02-07:
Are you still getting this warning and it can simply be ignored? 

> 01/02 04:06:52,932 WARNING [547498426368] (types.cpp:57) hwmon command 0x4f failed. Error type: No data to return (-21).

arthurlobo, 2019-02-07:
I followed the following video link originally (rviz starts at 4:09):
https://www.jetsonhacks.com/2018/04/28/intel-realsense-package-ros-nvidia-jetson-tx/

About the warning I still get it.

kvigulis, 2019-02-07:
Seems like I have some other issue then. I can access the depth and RGB image streams from RViz, but can't get anything to show up for the PointCloud2.

Maybe it is something with my RViz. I'm getting the following console output at launch time:

```
[ INFO] [1549502444.282820852]: rviz version 1.12.16
[ INFO] [1549502444.282897777]: compiled against Qt version 5.5.1
[ INFO] [1549502444.282914509]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1549502445.055716882]: Stereo is NOT SUPPORTED
[ INFO] [1549502445.055820577]: OpenGl version: 4.5 (GLSL 4.5).
[ERROR] [1549502445.290600727]: Ignoring transform for child_frame_id "camera_color_frame" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 -0.500000 0.000000 0.000000)
[ERROR] [1549502445.290687420]: Ignoring transform for child_frame_id "camera_aligned_depth_to_color_frame" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 -0.500000 0.000000 0.000000)
```

Mango-kid, 2019-02-14:
I am also having the exact same output. I have tried this on two different machines with both Kinetic and Melodic (although I don't think that matters). The device works fine in the realsense-viewer but does not work in ROS. Some of the camera images can be seen (no color). Also tested with latest build and debian packages. 

Echoing any of the point cloud topics outside of rviz returns nothing at all. 

When this device was first released I had no issues with the point cloud and color camera. The realsense will not let me downgrade the firmware so I cannot test the old setup. Intel also has some notes on the USB device with comments like "unplug it and plug it back in quickly", I'm not sure what to make of that. 

Also not able to see a point cloud with the D200. 

Any thoughts?

arthurlobo, 2019-02-14:
The issue was solved by purging ros-kinetic and reinstalling ros-kinetic and RealSense ROS 2.1.4.

Mango-kid, 2019-02-14:
Okay. I will test this out and report back.

doronhi, 2019-02-14:
@kvigulis, about the transformations, what does tf_echo shows regarding camera_color_frame and camera_infra1_frame? What does rs-enumerate-device?
Using RealSense-viewer, make sure the device is connected via usb3 and not usb2.

Mango-kid, 2019-02-14:
See the output below. I am using the included cabling. I am interested in the recommend firmware section. Once I upgraded the firmware it wouldn't let me go back to the previous version. 

`
 Device info: 
    Name                          : 	Intel RealSense D410
    Serial Number                 : 	739112060258
    Firmware Version              : 	05.11.01.00
    Recommended Firmware Version  : 	05.10.03.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/video4linux/video0
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0AD2
    Usb Type Descriptor           : 	2.1
`
Looking at the color_frame tf. The result is the same for the infra as well.  

`
Failure at 1550179621.007469937
Exception thrown:"camera_color_frame" passed to lookupTransform argument target_frame does not exist. 
The current list of frames is:
Frame camera_depth_frame exists with parent camera_link.
Frame camera_depth_optical_frame exists with parent camera_depth_frame.
Frame camera_infra1_frame exists with parent camera_link.
Frame camera_infra1_optical_frame exists with parent camera_aligned_depth_to_infra1_frame.
Frame camera_aligned_depth_to_infra1_frame exists with parent camera_link.
`

doronhi, 2019-02-18:
I'm sorry, @Mango-kid, as I understand it, the current situation is:
realsense-viewer works fine. You can see infra red images 1,2, color frame and depth image.
With ROS, you can see each of the image streams on rviz but not the pointcloud.
when you use `rosrun tf view_frames` you don't see `camera_color_frame`. Can you post the frames.pdf file that was created?
or you do see `camera_color_frame` but `rosrun tf tf_echo camera_color_frame camera_link` failes?

It's known to happen that during firmware upgrade something went wrong with the extrinsics. Could you post the output of the command: `rs-enumerate-devices -c` (text please if possible) - it shows the extrinsics set in the camera.

Mango-kid, 2019-02-19:
Okay. I managed to get the firmware back to 5.10.06. Now the camera is working fine. It will not work with the 5.11.01 firmware.

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

mzahana, 2019-03-18:
@Mango-kid I am facing the same problem. I downgraded to 5.10.6 but now the D435i is not found
```
[ WARN] [1552893277.060069371]: No RealSense devices were found!
```

Do I also have to downgrade Realsense2 package and librealsense?

mzahana, 2019-03-18:
@doronhi I have D435 and D435i, both with firmware 5.10.13. I am now using librealsense 2.19.0. I am able to visualize the pointcloud when I use D435, but not with D435i. I am suspecting that the extrinsic parameters of D435i got corrupted somehow during FW changes. I don't know how to fix that.

Any suggestions?

zeeshancmu, 2019-03-24:
> @doronhi I have D435 and D435i, both with firmware 5.10.13. I am now using librealsense 2.19.0. I am able to visualize the pointcloud when I use D435, but not with D435i. I am suspecting that the [...]

Same problem with 5.10.13 and 5.11.1

mzahana, 2019-03-25:
Problem is solved using.

Intel.Realsense.CustomRW.exe -g

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_D400_Dyn_Calib_User_Guide.pdf

zeeshancmu, 2019-03-28:
> Problem is solved using,
> 
> Intel.Realsense.CustomRW.exe -g
> 
> https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_D400_Dyn_Calib_User_Guide.pdf

This worked for me. Thank you.

closed 2019-04-01 by doronhi

ruiqun, 2019-05-05:
My environment is :
RealSense ROS v2.2.3
LibRealSense v2.21.0
D435i firmware 5.11.1
OS Ubuntu 18.04
ROS melodic

@dorodnic I got the same issue as below :
[ERROR] [1557038813.747173116]: Ignoring transform for child_frame_id "camera_color_frame" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 -0.500000 0.000000 0.000000)
[ERROR] [1557038813.747221015]: Ignoring transform for child_frame_id "camera_aligned_depth_to_color_frame" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 -0.500000 0.000000 0.000000)

When viewing with rviz, there is no color image received, and certainly no point cloud received.
I've tried with reset hardware when initialized, but still not working.
Is there any way to solved this problem but not by downgrade the firmware?
Thanks!

FPSychotic, 2020-06-22:
> My environment is :
> RealSense ROS v2.2.3
> LibRealSense v2.21.0
> D435i firmware 5.11.6
> OS Ubuntu 18.04
> ROS melodic
> 
> @dorodnic I got the same issue as below :
> [ERROR] [1557038813.747173116]: Ignoring transform for child_frame_id "camera_color_frame" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 -0.500000 0.000000 0.000000)
> [ERROR] [1557038813.747221015]: Ignoring transform for child_frame_id "camera_aligned_depth_to_color_frame" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 -0.500000 0.000000 0.000000)
> 
> When viewing with rviz, there is no color image received, and certainly no point cloud received.
> I've tried with reset hardware when initialized, but still not working.
> Is there any way to solved this problem but not by downgrade the firmware?
> Thanks!

I have the exact same error, my camera has a new D4 module, and it stream color but it doesnt make pointcloud with color, in both realsense viewer or ros wrapper.
In ros send me this error[ERROR] [1592827024.440253337]: Ignoring transform for child_frame_id "camera_color_frame" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 -0.500000 0.000000 0.000000)

rs-enumerate-devices -c
Fx:         	217.160980224609
  Fy:         	217.160980224609
  Distortion: 	Brown Conrady
  Coeffs:     	0  	0  	0  	0  	0  
  FOV (deg):  	88.62 x 57.85

 Intrinsic of "Depth" / 480x270 / {Z16}
  Width:      	480
  Height:     	270
  PPX:        	239.203430175781
  PPY:        	132.693588256836
  Fx:         	245.842620849609
  Fy:         	245.842620849609
  Distortion: 	Brown Conrady
  Coeffs:     	0  	0  	0  	0  	0  
  FOV (deg):  	88.62 x 57.54

 Intrinsic of "Depth" / 640x360 / {Z16}
  Width:      	640
  Height:     	360
  PPX:        	318.937896728516
  PPY:        	176.924789428711
  Fx:         	327.790161132812
  Fy:         	327.790161132812
  Distortion: 	Brown Conrady
  Coeffs:     	0  	0  	0  	0  	0  
  FOV (deg):  	88.62 x 57.54

 Intrinsic of "Depth" / 640x480 / {Z16}
  Width:      	640
  Height:     	480
  PPX:        	318.725463867188
  PPY:        	236.309753417969
  Fx:         	393.348205566406
  Fy:         	393.348205566406
  Distortion: 	Brown Conrady
  Coeffs:     	0  	0  	0  	0  	0  
  FOV (deg):  	78.26 x 62.78

 Intrinsic of "Depth" / 848x100 / {Z16}
  Width:      	848
  Height:     	100
  PPX:        	422.592712402344
  PPY:        	45.9253387451172
  Fx:         	434.321960449219
  Fy:         	434.321960449219
  Distortion: 	Brown Conrady
  Coeffs:     	0  	0  	0  	0  	0  
  FOV (deg):  	88.62 x 13.13

[... 422 lines cut ...]
 Translation Vector: 0.0500477217137814  0  0  

Extrinsic from "Infrared 2"	  To	  "Color" :
 Rotation Matrix:
   0   0   0
   0   0   0
   0   0   0

 Translation Vector: 0  0  0  

Extrinsic from "Infrared 2"	  To	  "Infrared 1" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0.0500477217137814  0  0  

Extrinsic from "Infrared 2"	  To	  "Infrared 2" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0