realsenseai/realsense-ros #597: Instructions / CMake changes to allow building librealsense2 + realsense2_camera in the same workspace

SteveMacenski, 2019-02-04:
For various reasons we can't have librealsense2 installed in binaries on the computer, but needs to stay in the workspace. Librealsense1 could let us do this pretty easily, but I hit a wall of issues trying to do this with librealsense2 + realsense2_camera. 

This ticket https://github.com/intel-ros/realsense/issues/303 gave some assistance but it would be good if the master branch worked out of the box for this use-case as its common.

RealSenseCustomerSupport, 2019-04-28:
----------------------------------------------

@SteveMacenski Currently the installation steps are still installing the RealSense SDK 2.0 before installing the Intel ROS Wrapper.  Is this still problem for you?

SteveMacenski, 2019-04-28:
Yes. The issue being that I would like to not have to have a systemwide install before using intel ROS wrapper. Librealsense/ROS wrapper v1 would allow you to build the SDK in the same catkin repo and link fine, but since v2, it hasnt been. I

RealSenseCustomerSupport, 2019-05-05:
----------------------------------------------

@SteveMacenski Got your point! We'll evaluate it and update to you later. Thanks!

doronhi, 2020-06-25:
Sorry I just started looking at it.
Could you try the following:
- removed librealsense2 installation.
- put librealsense2 v2.35.2 in the "catkin_ws/src" directory, alongside "realsense" 
   You'll have the following directory:
      src/realsense/
      src/librealsense/
- from catkin_ws, run `catkin_make_isolated`
- added `source ~/catkin_ws/devel_isolated/setup.bash` to `~/.bashrc` file 

It runs for me.
Does this procedure help? If not, could you direct me to your scenario please?

SteveMacenski, 2020-06-25:
Hi @doronhi I've long since stopped working with realsense cameras for products, but this ticket describes more of the issue: https://github.com/IntelRealSense/realsense-ros/issues/303 (which was a real issue closed by your automated system again). 

I can't tell you if that specific version, but I can say doing all that didn't work in older versions I worked with, since that was our basic use (we never had it installed to remove and wanted to build them in the same workspace). 

I suppose if you had a clean environment and could show that you can do above (from fixes like those suggested in the ticket) and everything finds dependencies and links correctly, then we can close this.

doronhi, 2020-06-27:
Hi @SteveMacenski ,
I really appreciate your interest in this issue despite the fact that it's been so long and you've already moved on.
It's certainly possible that all kind of dependencies were resolved during this time and things now work that didn't used to before.
I tried the procedure I suggested, of course, and would try it on several clean environments before committing to anything.
I think though, that instead of converting to `catkin_make_isolated` we should move on to use `catkin make`. Do you have an opinion on the subject? Although you no longer have a direct interest in this, I would really value your opinion.

SteveMacenski, 2020-06-27:
`catkin build` would be a good reference, as that's what I think most pro-users use. `catkin_make_isolated` would be good as well, I suppose it should work either way in a properly setup librealsense2/realsense_ros build.

RealSenseSupport, 2020-06-30:
@SteveMacenski Doron added one wiki below to address this. Could you please check if it's ok? Thanks!
https://github.com/IntelRealSense/realsense-ros/wiki/Build-with-local-librealsense2

RealSenseSupport, 2020-07-01:
Wiki added https://github.com/IntelRealSense/realsense-ros/wiki/Build-with-local-librealsense2. Close this accordingly.

closed 2020-07-01 by RealSenseSupport