realsenseai/realsense-ros #599: D435i IMU messages don't carry any covariances

SteveMacenski, 2019-02-05:
Even if statically computed, inputting a symmetric all 0 covariance matrix into any algorithm is going to go unstable. 

Example message:
```

SteveMacenski, 2019-02-05:
Even placeholder good default values should be inputted into the message based on the noise seen in the IMU.

doronhi, 2019-02-05:
It is a bug - no initialization of the covariance values. I fixed  in PR#600.
However, there is also the question of what are the values for the covariance.
Until a better option is found, I added it as parameters `linear_accel_cov` and `linear_accel_cov` that could be added to the launch file. They are currently defaulted to 0.01 on the diagonal.

closed 2019-02-05 by doronhi

SteveMacenski, 2019-02-05:
What was 0.01 based off of?

doronhi, 2019-02-06:
The value is not based at all. The device currently provides no value at all. The ROS message contains the field so I filled it. The value should vary with the camera being calibrated or not and maybe on other parameters.
Hopefully I’ll learn and implement a better solution in the near future. Use the launch file parameter until then.

SteveMacenski, 2019-02-06:
Ok, I'll file a ticket for that