realsenseai/realsense-ros #603: Background/foreground segmentation using depth and color (depth and color not matching well)

marcelinomalmeidan, 2019-02-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 2.17.1 | 
| Operating System & Version | Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    | 4.4.38-tegra| 
| Platform                                 | NVIDIA Jetson Tegra  |
| SDK Version                            |  2.0 }                          | 
| Language                            |  C++ | 
| Segment			|  Robotics | 

### Issue Description
I am trying to do background/foreground segmentation using the realsense camera and the ROS wrapper (I am currently in origin/development).
I have enabled "align_depth" and "enable_sync". I am using the topics ```/camera/aligned_depth_to_color/image_raw``` and ```/camera/color/image_raw```. I can see that both topics have synchronized timestamps. They are both at 640x480.
I  made a ROS node that subscribe to both using ```message_filters::TimeSynchronizer```, meaning that I only receive both images if they are at the exact same timestamp, otherwise the image is rejected.

I made a rudimentary depth filter using OpenCV, in which I check pixel by pixel in the depth map. If the given pixel is beyond a certain threshold, the corresponding RGB pixel in the color image is set to black.
However, it seems that the alignment between the color image and the depth image are not always well aligned, specially when I move the camera. The video below shows this behavior.  
https://youtu.be/Il2muRqst5o

Is this expected behavior? Is this mismatch expected when in motion?

marcelinomalmeidan, 2019-02-07:
I have been able to improve the results by opening realsense-viewer and switching the preset to high accuracy. Still, there seems to be some temporal filtering happening at depth level. 
Is it possible to disable it?

RealSenseCustomerSupport, 2019-04-22:
----------------------------------------------

The post-processing filtering could be enabled. Please try to disable the temporal filter and enable the sync in the wrapper.
https://github.com/intel-ros/realsense#launch-parameters

closed 2019-04-30 by RealSenseCustomerSupport

marcelinomalmeidan, 2019-05-01:
Thanks for the answer. Still, I am confused as to how to disable the filter. 
Checking for the parameter in runtime, this is what I get (empty result):

```
rosparam get /camera/realsense2_camera/filters
''
```

BTW, my sync parameter is enabled:

```
rosparam get /camera/realsense2_camera/enable_sync
true
```

Looking at the point cloud, there I can also clearly see spatial filtering, which I would desire to disable. Are these features available in ```Firmware Version | 2.17.1``` and realsense ros version 2.1.3?

marcelinomalmeidan, 2019-05-01:
BTW, the behavior can also be observed by running the RGBD Point Cloud example described [here](https://github.com/intel-ros/realsense#launch-parameters):

```
roslaunch realsense2_camera rs_camera.launch filters:=pointcloud
rosrun rviz rviz -d ~/catkin_ws/src/realsense/realsense2_camera/rviz/pointcloud.rviz
```