realsenseai/realsense-ros #604: testdd_int_test

Achllle, 2019-02-07:
I'm getting the following test failure from `test_dd_int.cpp` when running tests on Travis since I updated to librealsense2=2.18.0 and realsense ros 2.1.4.

```
rostest log file is in /home/travis/.ros/log/rostest-travis-job-bd7b62e8-8fdb-4aa4-aed8-0d7a911f012a-11018.log
[Testcase: testdd_int_test] ... FAILURE!
FAILURE: test [dd_int_test] did not generate test results
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/runner.py", line 164, in fn
    self.assert_(os.path.isfile(test_file), "test [%s] did not generate test results"%test_name)
  File "/usr/lib/python2.7/unittest/case.py", line 422, in assertTrue
    raise self.failureException(msg)
--------------------------------------------------------------------------------
[ROSTEST]-----------------------------------------------------------------------
[testdd_int_test][failed]
SUMMARY
 * RESULT: FAIL
 * TESTS: 0
 * ERRORS: 0
 * FAILURES: 1
ERROR: The following tests failed to run:
 * testdd_int_test
```

Oddly enough, this consistently fails on Travis but not on my machine. This might be unrelated to realsense even though it's a realsense ros test but I couldn't find much about this so trying to figure out if anyone has a clue.

Achllle, 2019-02-07:
Some of the tests will fail on a different file:
```

[ROSTEST]-----------------------------------------------------------------------
[ddynamic_reconfigure.rosunit-dd_full_scale_test/doTest][FAILURE]---------------
/home/travis/catkin_ws/src/realsense/ddynamic_reconfigure/test/dd_full_scale/dd_client.cpp:53
Value of: dynamics.call(srv)
  Actual: false
Expected: true
--------------------------------------------------------------------------------
SUMMARY
 * RESULT: FAIL
 * TESTS: 1
 * ERRORS: 0
 * FAILURES: 1
rostest log file is in /home/travis/.ros/log/rostest-travis-job-f447ef16-b10a-4aa0-9583-d502e1782489-10803.log
Unhandled exception in thread started by 
sys.excepthook is missing
lost sys.stderr
```

RealSenseCustomerSupport, 2019-06-12:
----------------------------------------------

Hi @Achllle ,

Does the failure still happen on the latest build?

Achllle, 2019-06-12:
It doesn't! closing

closed 2019-06-12 by Achllle