realsenseai/realsense-ros #606: Using Image Transport CameraPublisher to get DepthCloud tools

SteveMacenski, 2019-02-08:
Right now, the info and camera images are not published through an image transport CameraPublisher, but just publishers, which do not allow for the image_proc pipelines to work properly - especially as in relation to rviz's depthcloud utility. They require a CameraPublisher (which matches camera info and the images) so it can project it into a pointcloud. 

My fork has this implemented [here](https://github.com/intel-ros/realsense/compare/development...SteveMacenski:2.0.3-stable#diff-15761e8f7813869ae417a96398deda78R321), basically 3 line changes 
```
_image_publishers[stream] = {image_transport.advertiseCamera(image_raw.str(), 1), frequency_diagnostics};
```

Remove the info publishers, no longer required. Then change the image publisher map to 

```
 typedef std::pair<image_transport::CameraPublisher, std::shared_ptr<FrequencyDiagnostics>> ImagePublisherWithFrequencyDiagnostics;
```

and publish via

```
image_publisher.first.publish(*img, cam_info);
```

RealSenseCustomerSupport, 2019-06-24:
----------------------------------------------

@SteveMacenski You suggestion should be implemented in latest release. Could you please double check that? Thanks!

RealSenseCustomerSupport, 2019-06-27:
----------------------------------------------

@SteveMacensk Did you get chance to check that? Looking forward to your update. Thanks!

closed 2019-07-02 by RealSenseCustomerSupport