realsenseai/realsense-ros #607: Dynamic reconfigure GUI crashes when loading RGB_Module

SteveMacenski, 2019-02-08:
```
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/node_selector_widget.py", line 275, in _selection_changed_slot
    self._selection_selected(index_current, rosnode_name_selected)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/node_selector_widget.py", line 225, in _selection_selected
    item_widget = item_child.get_dynreconf_widget()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/treenode_qstditem.py", line 154, in get_dynreconf_widget
    self._param_name_raw)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/dynreconf_client_widget.py", line 67, in __init__
    group_desc, node_name)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/param_groups.py", line 152, in __init__
    self._create_node_widgets(config)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/param_groups.py", line 174, in _create_node_widgets
    widget = EnumEditor(self.updater, param)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/param_editors.py", line 423, in __init__
    self._combobox.setCurrentIndex(self.values.index(config['default']))
ValueError: 3 is not in list
```

doronhi, 2019-02-10:
I can't reproduce. I remember an issue regarding the same problem, maybe a month ago, but I can't find it now and I remember I couldn't reproduce it back then either.
Can you add the launch command line for both rqt_reconfigure and realsense2_camera? Do you use the original launch files included?  Can you add the log file up to the crash?
Any help in reproducing the crash would be appreciated.

SteveMacenski, 2019-02-11:
Sure,
It was run with rosrun rqt_reconfigure rqt_reconfigure, and selecting the RGB Module, then crashes. If just the depth module, no issue. The realsense2_camera launch file was tested with `rs_camera.launch` both default and with some custom params (didn't seem to effect).

This is running the latest release code. There's no other output leading up to it in dynamic reconfigure, I run the node and hit the RGB_module and immediately happens with the print out given above.

doronhi, 2019-02-13:
Its could be a dkms problem.
What do you see when you run:
`modinfo uvcvideo | grep version`?

If it's not `version:        1.1.2.realsense-1.3.4`
You should run:
`
sudo apt-get install librealsense2-dkms
`
You might need to reboot after installing dkms.

SteveMacenski, 2019-02-13:
I see 
```
version:        1.1.2.realsense-1.3.4
srcversion:     C4FB5E9B212E9C39ED6CD7C
```

All seems to be in place

doronhi, 2019-02-17:
I can't manage to reproduce it, so I really appreciate the help.
In the latest release from today I included a warning when an option in the device has a value that exceeds the range for that option. If you upgrade I believe we'll see it when the node starts.
However, to avoid upgrading if it doesn't suit you right now I would appreciate if you could copy lines 262-273 from the new base_realsense_node.cpp into your version, in line 268 (just before we add the new enum option). You will also need to replace `option_name` with `rs2_option_to_string(option)`.
This, I believe, will at least let us know which parameter is causing the crash and hopefully will help track down the origin of the problem.

doronhi, 2019-02-28:
I'm closing the issue for now. If you manage to reproduce with more information in the logs I would happily reopen.

Karleno, 2019-03-12:
Hi, I am experiencing the same problem where selecting the RGB module in rqt_reconfigure immediately crashes the program, yielding the same error as @SteveMacenski SteveMacenski got. The dkms is up to date and the log outputted from the launch scipt is pasted below, is that what you meant @doronhi ?

`PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 60
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 60
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 60
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 60
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/unite_imu_method: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

[... 41 lines cut ...]
[ INFO] [1552382003.492890200]: Expected frequency for infra2 = 60.00000
[ INFO] [1552382003.618930676]: Expected frequency for color = 60.00000
[ INFO] [1552382003.774198948]: setupStreams...
 12/03 10:13:23,778 WARNING [140139916461952] (sensor.cpp:366) Unregistered Media formats : [ UYVY Y8   ]; Supported: [ ]
[ INFO] [1552382003.781665008]: depth stream is enabled - width: 640, height: 480, fps: 60
[ INFO] [1552382003.782665311]: infra1 stream is enabled - width: 640, height: 480, fps: 60
[ INFO] [1552382003.783402976]: infra2 stream is enabled - width: 640, height: 480, fps: 60
 12/03 10:13:23,784 WARNING [140139916461952] (backend-v4l2.cpp:1248) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 12/03 10:13:23,785 WARNING [140139916461952] (sensor.cpp:366) Unregistered Media formats : [ RW16 ]; Supported: [ ]
[ INFO] [1552382003.791581340]: color stream is enabled - width: 640, height: 480, fps: 60
[ INFO] [1552382003.795721907]: insert Depth to Stereo Module
[ INFO] [1552382003.796397550]: insert Color to RGB Camera
[ INFO] [1552382003.796573817]: insert Infrared to Stereo Module
[ INFO] [1552382003.796680099]: insert Infrared to Stereo Module
[ INFO] [1552382003.796775507]: insert Gyro to Motion Module
[ INFO] [1552382003.796862478]: insert Accel to Motion Module
[... cut ...]
[ WARN] [1552382003.916846940]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 12/03 10:13:23,926 WARNING [140139028256512] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
 12/03 10:13:23,959 WARNING [140138667566848] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ INFO] [1552382003.992911350]: Done Setting Dynamic reconfig parameters.`

blubbi321, 2019-04-25:
Same issue here, same error. Should mention that this is compiled from source on a Jetson TX2.

doronhi, 2019-06-10:
As I couldn't reproduce, I will appreciate to know if the problem still exists in the new version with the ddynamic_reconfigure 0.2.0 debian package.
Thanks.

atyshka, 2019-06-27:
@doronhi I also have the issue/ Regarding the debian package, I don't think it's available for melodic yet, so I am building from source.

Zafrire, 2019-10-30:
I got the same issue using D435. 
The reason for this crash / ValueError is the default Value in RGB-Camera for "Power Line Frequency"
which is automatically loaded / set, when the camera is plugged in. Normally you just have Values between 0 - 2. But the camera is loading by default the value "**3**"

When you save the parameter with "realsense-viewer" into a JSON-File, you can detect the wrong Parameter:
"controls-color-power-line-frequency": "3",

I'm trying to load a JSON.File while launching the camera..but roslaunch isn't setting the parameters properly.. (is there a fix for this?)

`PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: /home/pokini/Sett...
 * /camera/realsense2_camera/linear_accel_cov: 0.01
[... 69 lines cut ...]
[ INFO] [1572433987.177712984]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
 30/10 12:13:07,177 WARNING [140125245794048] (backend-v4l2.cpp:1148) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[ INFO] [1572433987.179717913]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1572433987.181521538]: setupPublishers...
[ INFO] [1572433987.183681761]: Expected frequency for depth = 30.00000
[ INFO] [1572433987.188702186]: Expected frequency for infra1 = 30.00000
[ INFO] [1572433987.190360411]: Expected frequency for aligned_depth_to_infra1 = 30.00000
[ INFO] [1572433987.191867428]: Expected frequency for infra2 = 30.00000
[ INFO] [1572433987.193421862]: Expected frequency for color = 30.00000
[ INFO] [1572433987.194731444]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1572433987.197501283]: setupStreams...
[ INFO] [1572433987.198087050]: insert Depth to Stereo Module
[ INFO] [1572433987.198126564]: insert Color to RGB Camera
[ INFO] [1572433987.198149937]: insert Infrared to Stereo Module
[ INFO] [1572433987.198178625]: insert Infrared to Stereo Module
[ INFO] [1572433987.211116433]: SELECTED BASE:Depth, 0
[ INFO] [1572433987.236569205]: RealSense Node Is Up!
`

I found a little workaround for this crash, but it's pretty annoying to this every time you plugin the camera...
**Workaround:**
1. Start "realsense-Viewer" and select  under "RGB-Camera - Controls - Power Line Frequency" one of the three options. You don't need to start the RGB-Stream.
[image]

2. Now u can launch your camera in ros and use rqt_reconfigure with rgb-camera settings without the instant crash.

doronhi, 2019-10-30:
The issue is caused by using an unpatched kernel.
Try patching according to [installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md), section 4.

atyshka, 2019-10-30:
Yes, it’s caused by not having the kernel patches, but I would argue that this ros package should work without the kernel patches. Patches are not available for all kernels and the benefits may be minimal in certain use cases.

RealSenseCustomerSupport, 2020-01-06:
----------------------------------------------

@atyshka Sorry for the trouble. Did you still have issue with the latest SDK and ROS wrapper? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@atyshka Could you please update? Thanks!

atyshka, 2020-02-03:
I haven’t been using the realsense lately, but kernel patches seemed to have fixed it. I still think there should be some error handling since not all users can easily apply kernel patches, i.e. Jetson users. But aside from that feature request I’m fine closing this

RealSenseCustomerSupport, 2020-02-05:
----------------------------------------------

@atyshka Thanks for your suggestion! We'll evaluate it. Close this issue at this point. Thanks again!

closed 2020-02-05 by RealSenseCustomerSupport

jara001, 2020-02-19:
Hint for other truth-seekers (and developers afraid to patch the kernel):
Luckily, it can be 'patched' quite easily by adding few lines to the node `base_realsense_node.cpp:BaseRealSenseNode::registerDynamicOption()` just after skipping unsupported parameters.

For older 2.2.0 I have used:
`
if (option_name == "Power_Line_Frequency") {
    ddynrec->add(new DDInt(option_name, i, sensor.get_option_description(option), sensor.get_option(option), 0, 3));
    continue;
}
`

This is NOT ideal solution, but it works.
J.

doronhi, 2020-10-21:
PR #1461 removes parameters with values out of range from list of dynamic reconfigure parameters.
I don't think it's up to the node to decide which value to assign instead in such conditions so I settled for issuing a warning.