realsenseai/realsense-ros #610: Problem when running roslaunch realsense2_camera rs_camera.launch on odroid

florianspy, 2019-02-11:
I get the following exception:
[ERROR] [1549883789.649319258]: An exception has been thrown: hwmon command 0x7d failed. Error type: Invalid Command (-1).

doronhi, 2019-02-27:
Is realsense-viewer working ok?

florianspy, 2019-03-03:
no its not working

doronhi, 2019-03-04:
What librealsense version do you use? 
check [librealsense issue#3137](https://github.com/IntelRealSense/librealsense/issues/3137)

florianspy, 2019-03-05:
i am using version 2.18.0

RealSenseCustomerSupport, 2019-05-10:
----------------------------------------------

Hi florianspy,

To better understand and support, can you share more info of the issue?
a) This is running Ubuntu (16.04 or 18.04) off Odroid XU or something else?
b) As you said RS Viewer 2.18.0 is not working neither, has any RS releases worked off your platform?
c) Did you install from packages or build from source? Any issue along the installation/build?

Thanks!

RealSenseCustomerSupport, 2019-05-22:
----------------------------------------------

Hi florianspy,

Wonder if there is any update?

Thanks!

closed 2019-06-04 by RealSenseCustomerSupport