realsenseai/realsense-ros #611: rs_camera.launch fails when running with D435i

gopikaajaykumar, 2019-02-11:
When I run "roslaunch realsense2_camera rs_camera.launch," I get the following error:

[ERROR] [1549905227.776789670]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1549905227.776843167]: The error before refreshing the cache was: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[FATAL] [1549905227.777080030]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera-3] process has died [pid 15209, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/gopika/.ros/log/6485670c-2e20-11e9-96f0-54bf64800d73/camera-realsense2_camera-3.log].
log file: /home/gopika/.ros/log/6485670c-2e20-11e9-96f0-54bf64800d73/camera-realsense2_camera-3*.log

The device I am running is the **D435i**. I am running on **librealsense version 2.18.0, firmware 5.10.13, 4.4.0-142-generic kernel, and version 2.1.4 of realsense**. (I also tried running with a more recent version of librealsense, but had the same issue) _realsense-viewer_ is working correctly. Everything was working just a few days ago, so I suspect something got updated/uninstalled but I haven't been able to figure out what.
 
Thanks.

SteveMacenski, 2019-02-11:
This is a pluginlib / nodelet issue not registering, did you correctly source your workspace?

gopikaajaykumar, 2019-02-11:
Yes, I sourced the workspace prior to running the realsense launch file.

SteveMacenski, 2019-02-11:
I've seen those errors a bunch with pluginlib and nodeletlib, its 95% of the time has nothing to do with the code and with your environment - if you have the versions of the code listed, I have the same setup at my desk working. Try a `catkin config --install && catkin clean -y && catkin build` for me, sourcing the install space and trying again with `roslaunch realsense2_camera rs_camera.launch`. 

With your current setup what's the output of rospack find `realsense2_camera`

gopikaajaykumar, 2019-02-11:
Originally, rospack find was able to find realsense2_camera and listed it as being in my current workspace. After running `catkin config --install && catkin clean -y && catkin build` and then sourcing the workspace, rospack find is no longer able to find the package (output: "[rospack] Error: package 'realsense2_camera' not found")

SteveMacenski, 2019-02-11:
Did you source your install space or devel space?

gopikaajaykumar, 2019-02-11:
I had sourced the devel space, forgot about the install space. The launch file is working now. Thanks for your help!

closed 2019-02-11 by gopikaajaykumar

SteveMacenski, 2019-02-11:
No problem, glad to help

Afandi30, 2020-06-29:
> I had sourced the devel space, forgot about the install space. The launch file is working now. Thanks for your help!

Can you elaborate more on that.