realsenseai/realsense-ros #612: Pointcloud and RGB Image alignment

aitorgutierrez, 2019-02-12:
I am working on a dog detection system using deep learning (Tensorflow object detection) and Real Sense D425 camera. I am using the Intel(R) RealSense(TM) ROS Wrapper in order to get images from the camera.

I am executing "roslaunch rs_rgbd.launch" and my Python code is subscribed to "/camera/color/image_raw" topic in order to get the RGB image. Using this image and object detection library, I am able to infer (20 fps) the location of a dog in a image as a box level  (xmin,xmax,ymin,ymax)

I will like to crop the PointCloud information with the object detection information (xmin,xmax,ymin,ymax)
and determine if the dog is far away or near the camera. I will like to use the aligned information pixel by pixel between the RGB image and the pointcloud.

 How can I do it? Is there any topic for that?

Thanks in advance

mfkenson, 2019-02-24:
You could list the topics. Try those with "aligned_depth_to_color"
```
#terminal 1
~/catkin_ws$ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud  align_depth:=true
```

```
#terminal 2
~/catkin_ws$ rostopic list
/camera/Coded_Light_Depth_Sensor/parameter_descriptions
/camera/Coded_Light_Depth_Sensor/parameter_updates
/camera/RGB_Camera/parameter_descriptions
/camera/RGB_Camera/parameter_updates
/camera/aligned_depth_to_color/camera_info
/camera/aligned_depth_to_color/image_raw
/camera/aligned_depth_to_color/image_raw/compressed
/camera/aligned_depth_to_color/image_raw/compressed/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/compressed/parameter_updates
/camera/aligned_depth_to_color/image_raw/compressedDepth
/camera/aligned_depth_to_color/image_raw/compressedDepth/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/compressedDepth/parameter_updates
/camera/aligned_depth_to_color/image_raw/theora
/camera/aligned_depth_to_color/image_raw/theora/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/theora/parameter_updates
/camera/aligned_depth_to_infra1/camera_info
/camera/aligned_depth_to_infra1/image_raw
/camera/aligned_depth_to_infra1/image_raw/compressed
/camera/aligned_depth_to_infra1/image_raw/compressed/parameter_descriptions
/camera/aligned_depth_to_infra1/image_raw/compressed/parameter_updates
/camera/aligned_depth_to_infra1/image_raw/compressedDepth
/camera/aligned_depth_to_infra1/image_raw/compressedDepth/parameter_descriptions
/camera/aligned_depth_to_infra1/image_raw/compressedDepth/parameter_updates
/camera/aligned_depth_to_infra1/image_raw/theora
/camera/aligned_depth_to_infra1/image_raw/theora/parameter_descriptions
/camera/aligned_depth_to_infra1/image_raw/theora/parameter_updates
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/depth/camera_info
/camera/depth/color/points
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_infra1
/camera/infra1/camera_info
/camera/infra1/image_rect_raw
/camera/infra1/image_rect_raw/compressed
/camera/infra1/image_rect_raw/compressed/parameter_descriptions
/camera/infra1/image_rect_raw/compressed/parameter_updates
/camera/infra1/image_rect_raw/compressedDepth
/camera/infra1/image_rect_raw/compressedDepth/parameter_descriptions
/camera/infra1/image_rect_raw/compressedDepth/parameter_updates
/camera/infra1/image_rect_raw/theora
/camera/infra1/image_rect_raw/theora/parameter_descriptions
/camera/infra1/image_rect_raw/theora/parameter_updates
/camera/pointcloud/parameter_descriptions
/camera/pointcloud/parameter_updates
/camera/realsense2_camera_manager/bond

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

mzahana, 2019-03-25:
@aitorgutierrez I am trying to do almost the same thing. I basically would like to get the pointcloud portion that corresponds to a region of interest indicated by a bounding box which is drawn by the user on RGB frame. Have you found a way to solve your problem?
Any hints are appreciated ?
Thanks

mzahana, 2019-03-26:
@doronhi @icarpis Any hints on this?

RealSenseCustomerSupport, 2019-05-13:
----------------------------------------------

@aitorgutierrez Did you get it through? Any other questions about this ticket? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@mzahana  Sorry for late reply. Did you still need help on this? Thanks!

mzahana, 2019-06-13:
@RealSenseCustomerSupport thanks. My problem is solved. I construct the pointcloud portion corresponding to a bounding box in the “aligned” RGB image by performing some calculation based on the corresponding depth image.

RealSenseCustomerSupport, 2019-06-17:
----------------------------------------------

@mzahana That's great! Thanks for the update.

closed 2019-06-17 by RealSenseCustomerSupport