realsenseai/realsense-ros #613: Publish projector and asic temperatures from camera driver

tulku, 2019-02-13:
It would be nice to have the driver publish the projector and ASIC temperatures like they are displayed on the RealSense Viewer software.

As the cameras have temperature protection features that might reduce the power or even turn off the projector, it would be good to be able to monitor the temperatures to be sure that the camera is operating safely.

These temperatures could be publish in diagnostics or using a new custom topics.

SteveMacenski, 2019-02-13:
`diagnostics` would be best, that's a great suggestion! I would also make great use of that

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

doronhi, 2019-08-06:
added in #878.

doronhi, 2019-08-08:
merged into [release 2.2.8](https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.8)

closed 2019-08-08 by doronhi